freifunk-gluon/gluon #1623: gl.inet-gl-ar750 autoupdate fails on model

mmalte, 2019-01-18:
The Autoupdate of gl.inet-gl-ar750 fails with:

```
root@ffac-Bergdriesch-GL-AR750:/tmp# autoupdater 
Retrieving manifest from http://updates.freifunk-aachen.de/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model gl.inet-gl-ar750)
Retrieving manifest from http://updates.ffac.rocks/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model gl.inet-gl-ar750)
Retrieving manifest from http://updates.aachen.freifunk.net/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model gl.inet-gl-ar750)
autoupdater: error: no usable mirror found
```
there is an inconsistency between cpuinfo and board_name:

```
root@ffac-Bergdriesch-GL-AR750:~# cat /tmp/sysinfo/board_name
gl-ar750
```

```
root@ffac-Bergdriesch-GL-AR750:~# cat /proc/cpuinfo
system type		: Qualcomm Atheros QCA9533 ver 2 rev 0
machine			: GL.iNet GL-AR750
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 432.53
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available
```
The manifest offers updates for:

`gl-ar750 2018.2-1~exp20190117`

I'm happy to provide additional information if needed.

mweinelt, 2019-01-18:
> The first parameter defines the Gluon profile name, which is used to refer to the device and is part of the generated image name. The profile name must be same as the output of the following command (on the target device), so the autoupdater can work:

```
lua -e 'print(require("platform_info").get_image_name())'
```

Run the above and then - can you provide a PR to fix that?

mmalte, 2019-01-18:
```
root@ffac-Bergdriesch-GL-AR750:~# lua -e 'print(require("platform_info").get_image_name())'
gl.inet-gl-ar750
```
This is consistent to `/tmp/sysinfo/board_name`

After extending the manifest with additional lines for gl.inet-gl-ar750 the autoupdate was successful, but I think this is not the way you want things to be.
You want the output of the platform_info to be just gl-ar750?

mweinelt, 2019-01-18:
The image file name is provided in the targets file at `targets/ar71xx-generic`, the first parameter probably needs a fix.

achterin, 2019-01-18:
@mweinelt i've created the original pull request based on the feedback provided by @Commifreak as i only have the AR750S. He should be able to confirm/deny that this is broken.

closed 2019-01-18 by mweinelt