freifunk-gluon/gluon #1628: squashfs: rethink memory savings

CodeFetch, 2019-01-22:
ar71xx-tiny uses the small_flash flag, but not the low_memory_footprint flag.
This commit https://github.com/openwrt/openwrt/commit/11d6547455e848efb4fac0b983a07b11be5caf25 increases the squashfs block size for such devices. That might impair the load issue https://github.com/freifunk-gluon/gluon/issues/1243 as I've found out that it is likely a thrashing issue linked to the computation-intensive decompression of squashfs blocks on cache misses. Unfortunately it is not easy to say that in- or decreasing the block size makes the situation better or worse. It depends on the size of the cached data and in which block it is located and how the kernel handles freeing up pages on memory pressure and finally I doubt, that we have a squashfs cache at all as we use CONFIG_SQUASHFS_FILE_DIRECT and not CONFIG_SQUASHFS_FILE_CACHE, but I have to inspect the code to be sure.
Indeed it might be possible that if we have a cache, a bigger block size leads to scarcer decompression, but it introduces randomness depending on the squashfs layout and makes sysupgrade harder to perform. And on the other hand, if we don't have a cache at all, it definitely does not make sense to use such a big block size on low RAM devices (actually we don't want a squashfs cache as it introduces redundancy).
If we decide to use the low_memory_footprint flag for tiny devices, we might remove the support for DIR-615-C1 as I'm the only owner of this device and it has the smallest flash memory partition (I don't have in mind how much space that saves, but we are desperate, aren't we?).

christf, 2019-02-27:
there is a talk on  https://www.youtube.com/watch?v=M3yUjHKvde8&feature=youtu.be&t=916
 addressing both.

CodeFetch, 2019-03-02:
This issue is only about squashfs memory savings, which the talk did not cover. With the introduction of the low_memory_footprint flag other methods could be applied as well like those mentioned in the talk, but to make use of it we have to split the ar71xx-generic and ar71xx-tiny even more into ar71xx-generic-lowmem and ar71xx-tiny-lowmem upstream. I've started doing it, but my time is so sparse at the moment. BTW NAT426 is still on my ToDo, but also getting my WireGuard Layer 2 patch merged, adding support for pepe2k's u-boot for WR940N v3,4, writing a tutorial for replacing flash chips and RAM on 4MB/32MB devices, porting some devices to ath79 and so on. I always think we need much more man-power.

CodeFetch, 2020-03-03:
@mweinelt Why did you close it? It's not resolved...

CodeFetch, 2020-03-03:
Okay, I assume you closed it because of the switch to ath79... I think the situation in ath79 should also be looked at.

NeoRaider, 2020-05-29:
7e8af99cf504ca1dc389f282a0c94f4a911571be addresses this issue, so I guess we can close this now.

closed 2020-05-29 by NeoRaider

CodeFetch, 2020-06-02:
Thank you for your tests. This looks promising.