freifunk-gluon/gluon #1629: wrong primary MAC with UniFi AP AC Pro

rotanid, 2019-01-22:
during the discussions at #1610 it appeared to us, that there seems to be a related issue with the primary MAC of the Ubiquiti UniFi AP AC Pro device (UAP-AC-Pro).

it looks like the wrong MAC address is taken from eeprom as its primary MAC, because the one printed on the device is not the one used as primary.

i checked one device where the printed mac address can be found twice in the eeprom while the curent primary can be found only once:

right one twice:
> 0000a020  07 77 78 8a 20 8c d5 16  7a 8a 20 8c d5 16 00 c2  |.wx. ...z. .....|
wrong one once (together with another one)
> 0000a0c0  8a 20 8d d5 16 78 8a 20  8e d5 16 24 11 42 02 00  |. ...x. ...$.B..|

blocktrron, 2019-01-23:
Okay so the primary mac [is retrieved from phy0](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/luasrc/lib/gluon/upgrade/010-primary-mac#L23) which in case of the QCA9563 corresponds to the 5GHz radio.

The MAC address we use for the primary mac is [stored in the caldata](https://github.com/openwrt/openwrt/blob/master/target/linux/ar71xx/base-files/etc/hotplug.d/firmware/11-ath10k-caldata#L134) at offset 0x5006 in the eeprom partition.

mweinelt, 2019-01-23:
@rotanid Per your dump that would result in

00005000  44 08 3c 36 02 0d **78 8a  20 8e d5 16** 00 00 3f 00  |D.<6..x. .....?.|

mweinelt, 2019-01-23:
@rotanid Can you confirm that the correct MAC address was assigned to `eth0`? What is the value in `/tmp/sysinfo/board_name`?

mweinelt, 2019-01-23:
I created #1631 with the assumption that board_name is `unifiac-pro` and that the primary mac is on `eth0`. Please test and report back.

rotanid, 2019-01-24:
@mweinelt both correct, just checked it, sorry for the delay. have made photos of device and packaging if necessary

h03e, 2019-01-24:
@blocktrron send you the eeprom partition by mail sorry for the delay.

mweinelt, 2019-01-24:
@h03e Please test #1631, it should fix that issue.

closed 2019-01-25 by mweinelt

h03e, 2019-01-25:
@mweinelt Yea, works on my device.

- Hardware model Ubiquiti UniFi-AC-MESH-PRO
- Gluon version v2018.2-13-g9153b381
- MAC address 78:8a:20:b0:ec:d9
Matches the lable 
I'm impressed that this fix is not changing the MAC for devices already registered with fastd.

mweinelt, 2019-01-25:
These are scripts that are usually only run on the nodes initialization, so that is not surprising.

PR #1631 (2019-01-25): gluon-core: fix primary mac selection on unifiac devices
also sort by target and board_name

Tested on
- [x] Unifi AP AC Lite/Mesh
- [x] Unifi AP AC Pro/Mesh Pro

```sh
root@64283-cccda-w:~# cd /lib/gluon/core/sysconfig
root@64283-cccda-w:/lib/gluon/core/sysconfi# cat primary_mac
f0:9f:c2:de:c4:c5
root@64283-cccda-w:/lib/gluon/core/sysconfig# cp primary_mac primary_mac.bak
root@64283-cccda-w:/lib/gluon/core/sysconfig# rm primary_mac
root@64283-cccda-w:/lib/gluon/core/sysconfig# /lib/gluon/upgrade/010-primary-mac 
root@64283-cccda-w:/lib/gluon/core/sysconfig# cat primary_mac
f0:9f:c2:dc:c4:c5
r [... cut]
commit 9153b381 (2019-01-25): gluon-core: fix unifiac-{lite,pro} primary mac selection

fixes #1629
commit 35b34e2d (2019-02-04): gluon-core: fix unifiac-{lite,pro} primary mac selection

fixes #1629