freifunk-gluon/gluon #1630: Potential Bug in 2018.2 (cause unclear, perhaps routing)

sistein, 2019-01-23:
Hi all,

we just got aware that there might be a bug in 2018.2 . I found non matching issue so I startet one. 

We were unable to track it down by now und just want to notify in case you want to roll it out. 

The Problem is that (in some) meshing only cases it looks like there is a issue with using IPv4. Using IPv6 seems to work flawlessly. On node with a uplink everything works as expected. 

Since I'm quite busy I need to leave it like this for now. Perhaps someone else saw the same problem. 

Cheers

rotanid, 2019-01-23:
please also:

- add a link to your site.conf & site.mk
- name exact commits or tags you built from
- name the last commit or tag with which you didn't have your problem
- use a precise issue title, your current title is rather arbitrary

sistein, 2019-01-28:
Hi all,

after several testings using different Firmware and constellations I was unable to reproduce the bug we were informed about. The person issuing the bug stayed somehow silent so 'til now we don't see a 2018.2-breaker anymore. 

Sorry for this false alarm. Just wanted to make sure that no one is rolling out a potential bug like we were informed about (including routing of client traffic). We decided to roll out 2018.2 to our beta branch. 

If I get new informations I will keep you upto date. But until this happens I will close this issue. 

Cheers,

closed 2019-01-28 by sistein