freifunk-gluon/gluon #1636: Add support for encrypted 802.11s connections (SAE)

mweinelt, 2019-01-30:
Building on #1586, where we achieve more secure station connections, enduser traffic may still be routed via 802.11s connections, which at this point are entirely unprotected, and can be passively sniffed to recover previously encrypted traffic.

Therefore we should allow to enable SAE (Simultaenous Authentication of Equals) on 802.11s mesh connections. They would need to use a well-known PSK, and therefore offer little to no mitigation of active attacks, but since routers would generate a session key from the PSK and their MAC addresses we would protect the traffic after a secure connection was made from a fairly easy mountable passive attacks.

skorpy2009, 2019-02-12:
If I see it right, tiny devices couldn't stay in the mesh then, right?

mweinelt, 2019-02-12:
That's very unlikely seeing as we'd need roughly 600 kB of additional space to add required packages.

```bash
# opkg info wpa-supplicant-mesh-wolfssl
Package: wpa-supplicant-mesh-wolfssl
Version: 2018-05-21-62566bc2-5
Depends: libc, libnl-tiny, libwolfssl
Provides: wpa-supplicant, wpa-supplicant-mesh
Conflicts: wpad, wpad-mesh-openssl, wpad-mesh-wolfssl, wpad-mini, wpad-openssl, wpad-wolfssl, wpa-supplicant, wpa-supplicant-mesh-openssl
Status: unknown ok not-installed
Section: net
Architecture: mips_24kc
Size: 332817
Filename: wpa-supplicant-mesh-wolfssl_2018-05-21-62566bc2-5_mips_24kc.ipk
Description: WPA Supplicant (with 802.11s and SAE)

# opkg info libwolfssl
Package: libwolfssl
Version: 3.15.3-stable-1
Depends: libc
Provides: libcyassl
Status: unknown ok not-installed
Section: libs
Architecture: mips_24kc
Size: 278291
Filename: libwolfssl_3.15.3-stable-1_mips_24kc.ipk
Description: wolfSSL (formerly CyaSSL) is an SSL library optimized for small
 footprint, both on disk and for memory use.
```

txt-file, 2019-02-13:
Is it possible to use SAE on capable links but still have the fallback option to unsecured mesh links?
What is the upgrade path from unsecured to SAE secured connections?

mweinelt, 2019-02-14:
> Is it possible to use SAE on capable links but still have the fallback
> option to unsecured mesh links?

I don' think so.

> What is the upgrade path from unsecured to SAE secured connections?

As with every breaking change there is now the possibility to use the
gluon-scheduled-domain switch package.

https://gluon.readthedocs.io/en/latest/package/gluon-scheduled-domain-switch.html

CodeFetch, 2019-02-15:
Requirements from authsae:

> libcrypto from OpenSSL, version 0.9.8 or later. Basically, all you need is EC support that has the pre-defined groups.

Maybe one could try to build libwolfssl only with WOLFSSL_HAS_ECC?

For #1586 we may also check which dependencies are really needed for OWE.
Currently hostapd selects:
> 	select WOLFSSL_HAS_AES_CCM
> 	select WOLFSSL_HAS_AES_GCM
> 	select WOLFSSL_HAS_ARC4
> 	select WOLFSSL_HAS_DES3
> 	select WOLFSSL_HAS_DH
> 	select WOLFSSL_HAS_ECC
> 	select WOLFSSL_HAS_OCSP
> 	select WOLFSSL_HAS_PSK
> 	select WOLFSSL_HAS_SESSION_TICKET
>      select WOLFSSL_HAS_WPAS

Still 4 MB might not be enough.

mweinelt, 2019-02-15:
We can certainly use a reduced wolfssl build, I'm fairly certain we could drop at least `ARC4` and `DES3`, but that's a job for openwrt/packages as far as I'm concerned. I'm not keen on dropping much more than that because that would drastically reduce it's usefulness, for example for HTTPS connections.

I selected wolfssl (271kB) because we can use it for wpa-supplicant (SAE, 326 kB), hostapd (OWE, 243 kB) and even libustream, which would get us an HTTPS client implementation for (at least) wget for just another 4 kB. OpenSSL is a lot bigger (705 kB) and mbedtls (164 kB) lacks EC support.

Regarding authsae, it is a dead end and has been removed from OpenWrt in https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=8061c62f5d598fcbc7e08a8cc8728be55d236eab, but I believe you only mentioned it for its crypto requirements and didn't actually think of using it.

J0WI, 2019-02-23:
> mbedtls (164 kB) lacks EC support

They aim to support it:
https://tls.mbed.org/core-features

txt-file, 2019-10-20:
I did setup a 802.11s mesh at home in the last 24 hours. Problems occurred:
  * I got no working config with openwrt-19.07 branch
  * wpad-wolfssl provided me an kernel stack trace on the joining device
  * With wpad-openssl I currently have a WPA3/sae secured mesh running.
It might be that wpad-wolfssl (+dependencies) is much smaller but this doesn't help if it is currently broken.

Used devices are been Netgear wndr3800 and SamKnows WhiteBox8.

mweinelt, 2019-10-20:
Yep, we did a workshop where we used wpa-supplicant-mesh-openssl as well. Worked great.

txt-file, 2019-10-29:
For future reference: https://bugs.openwrt.org/index.php?do=details&task_id=2571

Me need to either fix this bug or go with wpad-openssl. I prefer first option but don't have time (and knowledge) to do it.

txt-file, 2019-10-30:
For future reference: https://bugs.openwrt.org/index.php?do=details&task_id=2571

mweinelt, 2019-10-30:
The bugreport is against OpenWrt master with the 5.4-rc2 mac80211 stack, so I wouldn't worry too much about it today.

txt-file, 2019-11-01:
The bug report is against OpenWrt master cause I am unable to setup WPA3-SAE secured mesh between two VoCore2 with 19.07. All I get is
```
Thu Oct 31 08:25:14 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Thu Oct 31 08:25:15 2019 daemon.notice netifd: radio0 (1062): Usage:    iw [options] dev <devname> set channel <channel> [NOHT|HT20|HT40+|HT40-|5MHz|10MHz|80MHz]
Thu Oct 31 08:25:15 2019 daemon.notice netifd: radio0 (1062): Options:
Thu Oct 31 08:25:15 2019 daemon.notice netifd: radio0 (1062):   --debug         enable netlink debugging
Thu Oct 31 08:25:15 2019 user.notice mac80211: Failed command: iw dev wlan0 set channel 1 HT40
Thu Oct 31 08:25:15 2019 kern.info kernel: [   27.175379] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Thu Oct 31 08:25:16 2019 daemon.notice wpa_supplicant[1293]: Successfully initialized wpa_supplicant
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Line 7: too large mode (value=5 max_value=4)
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Line 7: failed to parse mode '5'.
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Line 8: unknown network field 'mesh_fwding'.
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Line 9: unknown network field 'mesh_rssi_threshold'.
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Line 16: failed to parse network block.
Thu Oct 31 08:25:16 2019 daemon.err wpa_supplicant[1293]: Failed to read or parse configuration '/var/run/wpa_supplicant-wlan0.conf'.
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): cat: can't open '/var/run/wpa_supplicant-wlan0.pid': No such file or directory
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): Command failed: Invalid argument
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): Interface 0 setup failed: WPA_SUPPLICANT_FAILED
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): command failed: Link has been severed (-67)
Thu Oct 31 08:25:16 2019 user.notice mac80211: Failed command: iw dev wlan0 set mesh_param mesh_rssi_threshold 0
Thu Oct 31 08:25:16 2019 daemon.notice netifd: radio0 (1062): command failed: Link has been severed (-67)
Thu Oct 31 08:25:16 2019 user.notice mac80211: Failed command: iw dev wlan0 set mesh_param mesh_fwding 1
```
with /etc/config/wireless on both sides
```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option channel '1'
        option htmode 'HT40'
        option country 'DE'
        option legacy_rates '0'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'mesh'
        option mesh_id 'foobar'
        option mesh_fwding '1'
        option mesh_rssi_threshold '0'
        option encryption 'sae'
        option key '12345678'
```
Further most projects prefer bug reports against the development branch instead of some stable branch. Furthermore I gave up to wait for a real 19.07 release and concentrate therefore on the master branch.

@mweinelt Against which branch do you suggest that I test?

txt-file, 2019-11-02:
As far as I understand we need wpad-* for WPA3-SAE secured mesh. hostapd-* or wpa-supplicant-* is not enough. Correct me if I am wrong.

Test based on https://github.com/freifunk-gluon/gluon/commit/9dba2ca58023368e47c9b59948d1dafce3871fb1 with patch
```
diff --git a/targets/ramips-mt76x8 b/targets/ramips-mt76x8
index 4c2f8fed..c4035cae 100644
--- a/targets/ramips-mt76x8
+++ b/targets/ramips-mt76x8
@@ -48,4 +48,5 @@ device('tp-link-tl-wr841n-v13', 'tl-wr841n-v13', {
 
 device('vocore2', 'vocore2', {
        factory = false,
+        packages = {'-hostapd-mini', 'wpad-wolfssl'},
 })
```
and manually setup WPA3-SAE
```
wireless.mesh_radio0.encryption='sae'
wireless.mesh_radio0.key='12345678'
```
resulted in logread
```
daemon.notice netifd: Interface 'mesh_vpn' is now up
daemon.notice wpa_supplicant[2195]: Successfully initialized wpa_supplicant
daemon.err wpa_supplicant[2195]: Line 7: too large mode (value=5 max_value=4)
daemon.err wpa_supplicant[2195]: Line 7: failed to parse mode '5'.
daemon.err wpa_supplicant[2195]: Line 8: unknown network field 'mesh_fwding'.
daemon.err wpa_supplicant[2195]: Line 16: failed to parse network block.
daemon.err wpa_supplicant[2195]: Failed to read or parse configuration '/var/run/wpa_supplicant-mesh0.conf'.
daemon.notice netifd: radio0 (1392): cat: can't open '/var/run/wpa_supplicant-mesh0.pid': No such file or directory
daemon.notice netifd: radio0 (1392): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
daemon.notice netifd: radio0 (1392): Command failed: Invalid argument
daemon.notice netifd: radio0 (1392): Interface 0 setup failed: WPA_SUPPLICANT_FAILED
daemon.notice netifd: radio0 (1392): command failed: Link has been severed (-67)
user.notice firewall: Reloading firewall due to ifup of mesh_vpn (mesh-vpn)
user.notice mac80211: Failed command: iw dev mesh0 set mesh_param mesh_fwding 0
daemon.notice netifd: Network device 'client0' link is up
```
and iwinfo
```
mesh0     ESSID: unknown
          Access Point: B2:94:75:42:2F:51
          Mode: Mesh Point  Channel: unknown (unknown)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7628 14C3:0004 [MediaTek MT76x8]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
That the ESSID is not set is a very clear sign for me that setting up the interface failed.

With ```wpad-openssl``` the output of ```iwinfo``` looks better. Sadly it shows ```encryption: none``` but at least the link quality changes when I move one of the nodes around. pinging did **not** work.

mweinelt, 2019-11-02:
Please don't confuse SAE with WPA3. While WPA3 uses SAE, SAE on 802.11s does not equal WPA3.

Gluon will likely be using OpenWrt 19.07 (or whatever it's name) for the year to come, so if you want to test SAE do it on 19.07.
We've had great success using `wpa-supplicant-mesh-openssl` and setting just `encryption` and `key` on 19.07.
If you set that up watch `logread` for connections, because I guess `iwinfo` does not peek into wpa_supplicant, who manages the the encryption.

PR #1658 (2020-03-23): gluon-mesh-wireless-sae: Add support for SAE on 802.11s mesh
Setting a pre-shared key on the wireless mesh will enable encryption
using the SAE key agreement scheme and require encryption for all mesh
links between any two wireless mesh nodes.

<s>Disables the ar71xx-tiny target and other devices with 4 MB of flash
when the GLUON_WLAN_MESH_SAE variable is set to 1 as they don't have
enough free space for the required packages.</s>

Fixes #1636
commit 461d9040 (2020-03-23): Add gluon-mesh-wireless-sae package

This package adds support for SAE on 802.11s mesh connections.

Enabling this package will require all 802.11s mesh connections
to be encrypted using the SAE key agreement scheme. The security
of SAE relies upon the authentication through a shared secret.

In the context of public mesh networks a shared secret is an
obvious oxymoron. Still this functionality provides an improvement
over unencrypted mesh connections in that it protects against a
passive attacker who did not observe the key agreement. In addition
Management Frame Protection (802.11w) gets aut [... cut]
closed 2020-03-23 by blocktrron