freifunk-gluon/gluon #1652: GL Innovations GL-AR300M LEDs

rotanid, 2019-02-17:
the LEDs of the GL Innovations GL-AR300M and its variants (16, -Lite) don't work completely.

e.g. the wifi led isn't used to indicate wifi activity.
that's probably the case because in `target/linux/ar71xx/base-files/etc/board.d/01_leds` it is referring to "red:wlan" while there is no such thing in `target/linux/ar71xx/files/arch/mips/ath79/mach-gl-ar300m.c`

also, there are 4 LEDs specified in the mach file, all of them named green, although i can see only 3 LEDs on the device, two of them being green (and constantly shining after boot), the third one being red.
the middle LED is blinking during boot.

i welcome suggestions on how to fix it, i can test things.

rotanid, 2019-03-09:
has been merged upstream 2019-02-26 but isn't backported yet.
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=53c46b504c630e4c9c5a00bf644a69919bfc0c03
@NeoRaider maybe you can do the backport to openwrt-18.06 ? i have tested it and it applies and works.

NeoRaider, 2019-04-22:
@rotanid done

rotanid, 2019-06-06:
as the backport is now in gluon master, closing this issue.

closed 2019-06-06 by rotanid