freifunk-gluon/gluon #1653: gluon-config-mode-geo-location-osm: Providing a TileLayer URL

yanosz, 2019-02-18:
### Bug report

**What is the problem?**
There is no configuration option for a TileLayer URL in gluon-config-mode-geo-location-osm.
Roughly following https://wiki.openstreetmap.org/wiki/Tile_usage_policy using custom TileLayers (e.g. from Freifunk Aachen appears to be a good idea)

**What is the expected behaviour?**
Having a setting like tilelayer_url to specifiy the tile-layer.

**Gluon Version:**
2018.2

yanosz, 2019-02-18:
imho this is a minor enhancement

mweinelt, 2019-02-20:
The passing of the XYZ tile url would need to happen in `gluon-config-mode-geo-location-osm` but we would also need to patch `gluon-web-osm` here (https://github.com/freifunk-gluon/gluon/blob/0d4188d40b4d406feb0f9224b39acb93c50e052b/package/gluon-web-osm/javascript/gluon-web-osm.js#L50) and in the function signature.

PR #2072 (2020-07-05): gluon-config-mode-geo-location-osm, gluon-web-osm: add support for custom tile layers
Allow replacing the default OSM layer with a custom XYZ layer in site.conf.

Closes: #1653

I wonder if the "OSM" in the package names etc. even still makes sense like this... but I guess it can be adjusted in a separate PR if necessary.
closed 2020-07-05 by blocktrron