freifunk-gluon/gluon #1659: Node-local DNS resolver often responds extremely slowly

RalfJung, 2019-02-27:
### Bug report

**What is the problem?**

DNS queries to the DNS resolver on the node are often taking a long while to be answered.

I am repeatedly executing the following on a client connected to the node:
```
dig @fd4e:f2d7:88d2:ffff::1 ralfj.de
```
where `fd4e:f2d7:88d2:ffff::1 ` is the "next node" anycast IP. This command sometimes comes back immediately, sometimes it comes back after several seconds, and sometimes it times out after several seconds (all of these cases seem very roughly equally likely).
After the initial query, even when the answer takes several seconds, there is no DNS traffic on the tunnel between the node and the gateway. The link between the client and the node is a Linux bridge; both of these are just VMs on the same host and hence there should be 0 packet loss. I see the same behavior when talking to a real node over a WiFi link.

When I do `nslookup ralfj.de` on the node locally, the answer always comes back immediately. Also when I do `dig @fd4e:f2d7:88d2:ffff::101 ralfj.de`, which sends the query to one of our gateways, the answer comes back immediately -- so the network connection(s) all seem fine.

**What is the expected behaviour?**

The DNS reply should come back with decent latency (<50ms, say).

**Gluon Version:**

v2018.2

**Site Configuration:**

https://git.hacksaar.de/FreifunkSaar/gluon-site/blob/e2a66b1a461f95a80ee9483c9dfca96fe6fb2b36/site.conf

**Custom patches:**

None.

mweinelt, 2019-02-28:
I don't seem to be able to reproduce this, but I'm testing against a TL-WR4900v1, so my milage may vary. What devices have you been experiencing this on?

RalfJung, 2019-02-28:
I have experienced this on a TL-WDR4300 v1, as well as an x86 VM running on my laptop.

RalfJung, 2019-02-28:
It might be reproducible for you with this VM image? <https://mgmt.saar.freifunk.net/firmware/stable/factory/gluon-ffsaar-1.7.2-x86-64.img.gz>

RalfJung, 2019-02-28:
FWIW, I installed tcpdump on my virtual node, and I can see the DNS packets arriving locally -- and no answer being sent.

Is it possible that there is some kind of rate limiting in effect? The first query pretty much always works, so it feels a bit like very aggressive rate limiting.

mweinelt, 2019-03-02:
Just tried and I can reproduce the issue with your image.

rubo77, 2019-03-02:
@RalfJung on which gluonSource your firmware is built? can you build an ffsaar firmware on master and reproduce it there?

RalfJung, 2019-03-02:
That firmware is built with Gluon v2018.2. I tried master, but the problem remains.

rubo77, 2019-03-02:
So it must be related to some packages you use in ffsaar. Which is the exact site Configuration you used?

NeoRaider, 2019-03-02:
@RalfJung Do all 4 upstream servers work fine? In the description you only mention the first one.

mweinelt, 2019-03-02:
> @RalfJung Do all 4 upstream servers work fine? In the description you only mention the first one.

I checked all upstream nameservers multiple times and they looked fine.

RalfJung, 2019-03-03:
Also from what I see in the Wireshark log, the node will actually send out queries to all 4 upstream name servers in parallel and then use whatever answer comes back first.

RalfJung, 2019-03-03:
I tried disabling a bunch of packages, down to:
```
GLUON_FEATURES := \
	autoupdater \
	mesh-batman-adv-15 \
	mesh-vpn-tunneldigger \
	radvd
```
That does not fix the problem. I feel the rate of successful queries increased a bit, but there's certainly still tons of failing queries.

csamsel, 2019-03-05:
> Also from what I see in the Wireshark log, the node will actually send out queries to all 4 upstream name servers in parallel and then use whatever answer comes back first.

this behavior is activated by --all-servers (or all-servers in dnsmasq.conf) which isnt present on my v2018.2 node.

RalfJung, 2019-03-17:
> this behavior is activated by --all-servers (or all-servers in dnsmasq.conf) which isnt present on my v2018.2 node.

Where would I check this?

RalfJung, 2019-03-17:
Interestingly, when I do `tcpdump -ni local-node ip udp port 53`, I only see queries that actually get a successful response. On `br-client`, I also see the queries that get no response.

Seems like some kind of filtering blocks the other queries?

RalfJung, 2019-03-17:
I tried basically randomly removing filtering rules from both `iptables` and `ebtables`, and it turns out that the following makes a huge difference:
```
ebtables-tiny -D FORWARD -i bat0 --logical-out br-client -o local-port -j DROP
```
Of course that makes no sense, since the traffic is not coming from `bat0`, but, uh, it helps. There's still an occasional gap in DNS replies, but the vast majority of them works with this. When I add the rule back, after ~10s the drops start happening again.

What...?!?

RalfJung, 2019-03-17:
Is there a way to get full ebtables, not just ebtables-tiny, onto a Gluon device? Then at least I could check which packets are matching this rule.

RalfJung, 2019-03-17:
I did this experiment with some old firmware versions, and it turns out that version 1.4.1 worked fine but 1.5.1 has this issue (1.5.0 was a bad version from our side).  That corresponds to [this diff in our site config](https://git.hacksaar.de/FreifunkSaar/gluon-site/compare/1.4.1...1.5.1).

Other than build system changes, this marks the update from Gluon v2016.2.6 to v2017.1.3, and the addition of `gluon-ebtables-source-filter` (but removing that package on a current build with `-gluon-ebtables-source-filter` changed nothing).

Comparing the ebtables for these releases, indeed the following are present on 1.5.1 but not on 1.4.1:
```
-i bat0 --logical-out br-client -o local-port -j DROP
-i local-port --logical-out br-client -o bat0 -j DROP
```

bobcanthelpyou, 2019-03-17:
these rules are part of the `gluon-mesh-batman-adv` package.

```
gluon-mesh-batman-adv/files/lib/gluon/ebtables/250-next-node:rule('FORWARD --logical-out br-client -i bat0 -o local-port -j DROP')
gluon-mesh-batman-adv/files/lib/gluon/ebtables/250-next-node:rule('FORWARD --logical-out br-client -i local-port -o bat0 -j DROP')
```

RalfJung, 2019-03-18:
@bobcanthelpyou 	ah so this is my own bugreport biting me. ;)

Another interesting piece of data: the issue also occurs with `dig @fd4e:f2d7:88d2:ffff::1 my.ffsaar`, which is our next-node DNS name. And the issue stops occurring when I unplug the node from the internet. That also explains why other communities are not seeing this. There is something about the traffic in our network that leads to to these packet drops. But somehow dropping some of that traffic makes it worse... that just doesn't make any sense to me.

I wonder if it is the sheer size? We have ~550 nodes in a single batman domain -- that's large, but to my knowledge there are bigger networks out there. (Yes, we are working towards a domain split.) Also, my laptop CPU is barely used when the VM runs, so I wouldn't know why it'd do any load-related drops.

RalfJung, 2019-03-23:
When I ping fd4e:f2d7:88d2:ffff::1, our next-node anycast IPv6 address, I get around 80% packet loss. When I drop the two rules from `gluon-mesh-batman-adv` mentioned above, the drop rate goes down to around 2%. These 2% remain even when I add an unconditional `-j ACCEPT` as the first rule in the `INPUT`, `FORWARD` and `OUTPUT` ebtables. When I ping a gateway inside our network, 0% are dropped. This is consistent with what I observed using `watch dig` before.

I noticed that we have a bunch of ICMP6 neighbor solicitations for fd4e:f2d7:88d2:ffff::1. Indeed those are not being filtered by the ebtables rules. But they are coming all the time so I suppose this is not some particularity of our network.

---

I also think I found the first real clue:

Interestingly, when I build a firmware where I use a different MAC for the next-node anycast (but the same IPv4/IPv6), the problem goes away. It seems like somehow, a node connected to our batman network is tainted such that the particular MAC address our network uses does not work any more.

So I searched by Wireshark traces for this MAC address, and indeed there is some traffic! There are some router advertisments and some multicast listener reports being sent through our network from this MAC address and IPv6 source address fe80::303d:f9ff:fe8f:152 (the link-local address matching our anycast MAC).  They appear about every 3 seconds. I suppose this should not happen?

RalfJung, 2019-03-23:
Indeed blocking the node from which the traffic with this MAC originates solves the problem. Phew.

Now I am just left wondering what is going on with that node...

rotanid, 2019-03-25:
as it now looks like it's no gluon bug i changed the labels.

RalfJung, 2019-03-26:
Yeah. Though it also seems like whatever ebtables filters gluon has locally to drop incoming packets with the anycast MAC are insufficient -- they are dropped too later, after the kernel already registered this MAC as existing on some bridge port.

NeoRaider, 2019-04-22:
We currently don't drop incoming packets with the anycast MAC address at all, only outgoing packets. I'll look into improving this.

PR #1714 (2019-04-28): gluon-mesh-batman-adv: drop packets from mesh with next-node source or destination address
Fixes #1659

@RalfJung Could you check if this is working as expected?
commit 09916f6e (2019-04-28): gluon-mesh-batman-adv: drop packets from mesh with next-node source or destination address (#1714)

Fixes #1659
commit c5042399 (2019-06-08): gluon-mesh-batman-adv: drop packets from mesh with next-node source or destination address (#1714)

Fixes #1659

(cherry picked from commit 09916f6e359114fdc6f9594f60b76a27e7352f1b)
closed 2019-04-28 by NeoRaider