freifunk-gluon/gluon #1660: 802.1x authentication for open infrastructure 802.11s networks

CodeFetch, 2019-02-28:
Using SAE with a pre-shared key does allow encryption, but not authentication of nodes if the PSK is publicly known. 
For open infrastructure networks a possibility to ensure at least trust on first use would be to initiate a connection between nodes using a publicly known PSK the first time, then generate a random PSK using that secure session and save it to flash on the peering nodes. A custom RADIUS server would be needed for that.

mweinelt, 2019-02-28:
What would happen if either side lost its state due to a reset?

CodeFetch, 2019-02-28:
They need to handle out new credentials. Actually I don't even know whether it is implemented open-source, yet. The goal of having something like that is to implement a trust model. If they lose their credentials, they lose their accumulated trust.

mweinelt, 2019-03-02:
How would you distinguish lost state versus an attack by someone malicious?

CodeFetch, 2019-03-02:
Distinguishing between a lost state and an attack is easy as the ID of the node should not rely on a MAC address, but a randomly initiated value. You could also use public key crypto for that. Trust models are the hard problem. They are ongoing research e.g. by the military. There are many papers about that. Some propose using neural networks other simple evidence based rating algorithms. For ordinary commercial networks you either use a shared PSK (which is prone to physical hacking) or a centralized trust agent which is a single point of failure and needs to be reachable all the time. Without having authenticated mesh connections, we have no possibility to implement trust models at all. Trust on first use would be a first step.

mweinelt, 2022-03-26:
Closing as stale and unlikely to be implemented.

closed 2022-03-26 by mweinelt