freifunk-gluon/gluon #1664: Supported devices table?

CodeFetch, 2019-03-09:
As far as I know, we don't have one.
It would be nice if it would have a similar format like the OpenWrt tables.

One table for each target with:
- Vendor, Model, Version
- CPU, RAM, Flash
- Number of Ports/Gigabit
- WLAN1,2,3

bobcanthelpyou, 2019-03-09:
There is an Gluon table (view) available as an OpenWrt table 
https://openwrt.org/toh/views/toh_gluon_supported

Is that up2date? And how is it generated?

CodeFetch, 2019-03-09:
WTF? :grin:

rotanid, 2019-03-09:
@rubo77 is "in charge" of the provided link, this can be used.
apart from that, feel free to create one in the wiki, everyone has access there :)

closed 2019-03-09 by rotanid

CodeFetch, 2019-03-09:
@mweinelt @blocktrron Are you shocked of my phrasing? The OpenWrt wiki is really not the place where I expect a supported hardware table for Gluon...

bobcanthelpyou, 2019-03-09:
Thx @rubo77 :)

mweinelt, 2019-03-09:
> @mweinelt @blocktrron Are you shocked of my phrasing? The OpenWrt wiki is really not the place where I expect a supported hardware table for Gluon...

The way you communicate occasionally is quite astonishing. I would kindly ask you to refrain from replying with purely emotional comments in this developer community, it's simply not the helpful.

CodeFetch, 2019-03-09:
@mweinelt Okay. Emotions seem to be hard to me mediate over the internet even if one uses emoticons. Next time I just write "That is astonishing. Why is the supported hardware table of this project on an external website?" even if it is long and boring and this is a free-time project for most of us.

bobcanthelpyou, 2019-03-09:
@CodeFetch At least the device "database" is available on the external website already. Gluon is OpenWRT based. A lot of Gluon devices issues are fixed upstream. There is a connection between them. It would not generate a lot of fun to duplicate and maintain such a device "database".

bobcanthelpyou, 2019-03-09:
@CodeFetch thx for #1665

CodeFetch, 2019-03-09:
@bobcanthelpyou I understand that. As device specifications don't change I'd have created a table in the Gluon docs right from the beginning, but it's okay for me to have this on the OpenWrt Wiki with a link in the Gluon docs.

rubo77, 2019-03-09:
There are some newly adopted devices still missing there.
Someone has to add the gluon support line in the openWRT database entries