freifunk-gluon/gluon #1674: 2018.2 sysupgrade to self looses hostname

smoe, 2019-03-15:
**What is the problem?**
- What is not working as expected?
When (by accident) sysupgraded a router that was on 2018.2 already, that machine then lost its hostname.
- How is it misbehaving?
Only the hostname seems lost. Other modified data like the channel and ssh keys were still intact.
- When did the problem first start showing up?
I am installing a few routers with 2018.2, coming from 2018.1.4. This has worked just fine for many (~30) devices.
- What were you doing when you first noticed the problem?
Only when going from 2018.2 to "itself", this happenes reproducibly.
- On which devices (vendor, model and revision) is it misbehaving?
Confirmed on three instances for CPE210 v1.1 and one WDR4300 v1.

**What is the expected behaviour?**
Idempotency. The device should basically not have ended up no different from how it was before. 

**Gluon Version:**
2018.2

**Site Configuration:**
github.com/ffoh/site

**Custom patches:**
none

bobcanthelpyou, 2019-03-15:
https://github.com/ffoh/site-ffoh

smoe, 2019-03-15:
> https://github.com/ffoh/site-ffoh

Thanks - corrected.

rubo77, 2019-03-17:
Is this reproducible?

How exactly did you upgrade? On the command line or via config mode?

Maybe you did use sysupgrade -n by accident

bobcanthelpyou, 2019-03-17:
@rubo77 it seems that `sysupgrade -c` was used on the command line.

> ```
> scp gluon-2018.2.sysupgrade.bin root@[ip]:/tmp
> ssh root@ip
> cd /tmp
> sysupgrade -c gluon-2018.2.sysupgrade.bin
> ```

rubo77, 2019-03-17:
Why the -c?

     -c attempt to preserve all changed files in /etc/ 

Maybe the option doesn't work correctly.

Try without option!

NeoRaider, 2019-03-17:
I'm not sure how `-c` could cause this issue, as all information regarding the hostname is stored in /etc/config, which is always preserved on upgrades even without `-c`.

Still, `-c` is not supported in any way, weird things may happen when you pass is.

bobcanthelpyou, 2019-03-17:
Referring to https://openwrt.org/docs/techref/sysupgrade and https://openwrt.org/docs/guide-user/installation/sysupgrade.cli 
> The list of files preserved is by default in `/lib/upgrade/keep.d`. Additionally files returned by `opkg list-changed-conffiles` are also preserved.

`-c` results in `export SAVE_OVERLAY=1` 

```
if [ $SAVE_OVERLAY = 1 ]; then
        [ ! -d /overlay/upper/etc ] && {
                echo "Cannot find '/overlay/upper/etc', required for '-c'"
                exit 1
        }
        sysupgrade_init_conffiles="add_overlayfiles"
else
        sysupgrade_init_conffiles="add_uci_conffiles"
fi
```

```
add_uci_conffiles() {
        local file="$1"
        ( find $(sed -ne '/^[[:space:]]*$/d; /^#/d; p' \
                /etc/sysupgrade.conf /lib/upgrade/keep.d/* 2>/dev/null) \
                -type f -o -type l 2>/dev/null;
          list_changed_conffiles ) | sort -u > "$file"
        return 0
}

add_overlayfiles() {
        local file="$1"
        find /overlay/upper/etc/ -type f -o -type l | sed \
                -e 's,^/overlay\/upper/,/,' \
                -e '\,/META_[a-zA-Z0-9]*$,d' \
                -e '\,/functions.sh$,d' \
                -e '\,/[^/]*-opkg$,d' \
        > "$file"
        return 0
}
```

smoe, 2019-03-17:
Yes, I used the -c option - and happily report that it worked so far. Not at my own computer at the moment, will report later today about the behaviour  without -c and if problem persists with 2018.2.1.

NeoRaider, 2019-03-17:
The issue is indeed `-c`: Instead of backing up `/etc` in addition to the regular list of config files, it backs up `/etc` **instead of** the regular list. This is very broken, and was always broken even with older Gluon versions (although symptoms might differ). Don't ever use `-c`. It will also reset parts of your wireless config, and possibly other settings.

As you are using uci to set the hostname in your issue description, I'd also like to note the following:

Two different concepts of hostnames exist in Gluon: The internal (technical) hostname, and the pretty (human-readable) hostname. The internal hostname must be a valid DNS name; it is usually derived from the pretty hostname automatically by removing invalid characters (everything outside of `[A-Za-z0-9-]`). The pretty hostname can be any UTF-8 string.

Internally, we use two different UCI fields for the hostnames:
- `system.@system[0].hostname` for the internal hostname
- Optionally: `system.@system[0].pretty_hostname` for the pretty hostname. This field is only set when the technical and pretty hostnames differ. The field `system.@system[0].hostname` must never contain an invalid DNS name.

In your example you directly set `system.@system[0].hostname` to an invalid value. The fix is simple: Instead of calling uci manually, always set the hostname using
`pretty-hostname 'FOH-HafeninfoNiendorf-4300_K1-44'`, which will update both hostname fields as needed.

closed 2019-03-17 by NeoRaider

smoe, 2019-03-17:
Thank you all.

rotanid, 2019-03-18:
@NeoRaider maybe we could patch sysupgrade for glupn to not contain the "-c" switch if it doesnt work?