freifunk-gluon/gluon #1680: Add a CI pipeline for pulll request

bobcanthelpyou, 2019-03-21:
Some Gluon Developers like to have a kind of CI pipeline for pull requests.

1) What should such a pipeline check for, could help with?
* Syntax of scripts, code, docs, ...
* Linting of scripts, code, docs, ... 
* Need something to be updated, language files, target missing in docs, ...
* ...
 
2) And what kind of software should be used?
* TravisCI
* self hosted
  * Jenkins
  * ...
* ...

3) What rules should apply? What should happen when pipeline fails?
* just warning/suggestions
* block merge
* ...

4) What else should we think about?
* only changes and depending stuff should be checked vs. check everything all the time
* ...

CodeFetch, 2019-03-26:
I have a powerful server where I could offer a KVM for CI. @lemoer has written a test setup where Gluon nodes could be virtualized using the x86 images a while ago. I don't know the state of it, but it would be cool if we'd support proper unit tests.

lemoer, 2019-03-26:
I would not consider it as stable, but it is working: https://github.com/lemoer/pynet ([besides a bug in python3.7 and python3.8 itself](https://github.com/lemoer/pynet/issues/2))

Feel free to test it.

christf, 2019-05-25:
I think standardizing the build in some tool would be interesting - then in a second step talking about capacity would be interesting. Building a jenkinsfile that builds gluon inside a prepared docker environment could be of merit. It certainly would stop each community building their own build tool.

phaus, 2019-08-08:
I kind of did a first test with Gitlab: 
https://git-lab.de/gluon/build/-/jobs/2561

I would like to get some feedback about that.
What kind of architecture/versions/patchsets would make sense?

Due to the Gitlab Runner Architecture, it is quite easy to add new Build Nodes.

Also unclear - at least to me - is where to store the final Binaries.
Is there a central file repository somewhere for distribution?

mweinelt, 2019-11-04:
Thank everyone for contributing to the initial continuous integration getting set up.

CI has been live for a few weeks on the master and v2019.1.x branches as well as for all pull requests that were rebased accordingly.

https://build.ffh.zone/blue/organizations/jenkins/gluon-ci/branches/

I think this issue is resolved and additional work necessary should be discussed in new issues. Especially the automated testing part (which pynet is part of) should be discussed in #1734.

closed 2019-11-04 by mweinelt