freifunk-gluon/gluon #1681: Status page - 500 Internal Server Error page does not show up cause of HTML issue

bobcanthelpyou, 2019-03-23:
**What is the problem?**
I was trying to access the status page of a node affected by high CPU load (maybe #1243) it was not possible to access the status page at all or a 500 error is shown. But the HTML code seems to be messed up. `<!DOCTYPE html>` and `<head>`, .. are repeated.

```
<!DOCTYPE html>
<html xmlns="http://www.w3.org/1999/xhtml" lang="" xml:lang="">
	<head>
		<meta charset="utf-8" />
		<meta name="viewport" content="width=device-width, user-scalable=no" />

		<title><!DOCTYPE html>
<html xmlns="http://www.w3.org/1999/xhtml" lang="" xml:lang="">
	<head>
		<meta charset="utf-8" />
		<meta name="viewport" content="width=device-width, user-scalable=no" />

		<title>Error</title>

		<link rel="stylesheet" href="/static/status-page.css" type="text/css" />
	</head>
	<body>
		<header>
			<h1>Error</h1>
		</header>
		<div class="container">
			<div class="frame">
				<h2 name="content">500 Internal Server Error</h2>
<p>Sorry, the server encountered an unexpected error.</p>
<pre class="error500">Failed to execute dispatcher target for entry &#39;/&#39;.
The called action terminated with an exception:
/usr/lib/lua/gluon/web/template.lua:43: Failed to execute template &#39;status-page&#39;.
A runtime error occured: [string &#34;/lib/gluon/status-page/view/status-page.htm...&#34;]:79: attempt to index global &#39;nodeinfo&#39; (a nil value)</pre>

			</div>
		</div>
	</body>
</html>
```

**What is the expected behaviour?**
```
<!DOCTYPE html>
<html xmlns="http://www.w3.org/1999/xhtml" lang="" xml:lang="">
	<head>
		<meta charset="utf-8" />
		<meta name="viewport" content="width=device-width, user-scalable=no" />

		<title>Error</title>

		<link rel="stylesheet" href="/static/status-page.css" type="text/css" />
	</head>
	<body>
		<header>
			<h1>Error</h1>
		</header>
		<div class="container">
			<div class="frame">
				<h2 name="content">500 Internal Server Error</h2>
<p>Sorry, the server encountered an unexpected error.</p>
<pre class="error500">Failed to execute dispatcher target for entry &#39;/&#39;.
The called action terminated with an exception:
/usr/lib/lua/gluon/web/template.lua:43: Failed to execute template &#39;status-page&#39;.
A runtime error occured: [string &#34;/lib/gluon/status-page/view/status-page.htm...&#34;]:79: attempt to index global &#39;nodeinfo&#39; (a nil value)</pre>

			</div>
		</div>
	</body>
</html>
```

**Gluon Version:**
`gluon-v2018.1.4` 

https://github.com/freifunk-gluon/gluon/releases/tag/v2018.1.4

**Site Configuration:**
https://github.com/freifunk-ilmenau/site-ffil/tree/v2018.1.4

**Custom patches:**
none

bobcanthelpyou, 2019-03-23:
That issue is reproducible with Gluon v2018.2.1 replacing `nodeinfo` with a non existing variable in `status-page.html` e.g.
```
diff --git a/package/gluon-status-page/files/lib/gluon/status-page/view/status-page.html b/package/gluon-status-page/files/lib/gluon/status-page/view/status-page.html
index 94c1d433..47cefbdf 100644
--- a/package/gluon-status-page/files/lib/gluon/status-page/view/status-page.html
+++ b/package/gluon-status-page/files/lib/gluon/status-page/view/status-page.html
@@ -83,7 +83,7 @@
                <meta charset="utf-8" />
                <meta name="viewport" content="width=device-width, user-scalable=no" />
 
-               <title><%| nodeinfo.hostname %> - <%:Status%></title>
+               <title><%| nonodeinfo.hostname %> - <%:Status%></title>
 
                <link rel="stylesheet" href="/static/status-page.css" type="text/css" />
        </head>
```

bobcanthelpyou, 2019-03-23:
`<%|` does not escape the HTML code as expected. `<%|` and `<%=` result in the same output. (e.g. replacing `nodeinfo.hostname` with `print("<html>")`)

`<%|` is handled as `T_TYPE_EXPR` and `<%=` as `T_TYPE_EXPR_RAW` in `template_parser.c`
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web/src/template_parser.c#L80

Also the docs missing an entry for the tag `<%|` 
https://gluon.readthedocs.io/en/latest/dev/web/view.html

NeoRaider, 2019-03-23:
We cannot really do anything useful when an error occurs after the template parsing has started, as there is no caching at all - all template constructs write their output directly to the browser to avoid wasting memory.

When an error occurs, we have no idea what state the template parser was in - in particular, whether our layout header has already been sent. The current solution is a best effort: We send the error to the browser, so we can at least find the message in the HTML source - getting out anything valid just doesn't have priority (and would make the parser, and possibly its usage, much more complex).

Two remarks regarding your last comment:
- *print* is the low-level I/O primitive, no escaping can happen when you call it directly (which you should never do in templates)
-  `<%| print("<html>") %>` is plain weird. The correct way to use the template tags is to pass string into them, for example `<%| "<html>" %>, which should escape the string correctly

bobcanthelpyou, 2019-03-23:
Ah i thought something like the output of lua code is escaped. Then there will be just a fix for the docs i guess #1683

closed 2019-03-23 by bobcanthelpyou