freifunk-gluon/gluon #1688: tunneldigger client update

RalfJung, 2019-03-31:
https://github.com/wlanslovenija/tunneldigger/pull/88 hast just landed in tunneldigger upstream, fixing https://github.com/wlanslovenija/tunneldigger/issues/87. It would be great to also get this update in Gluon.

bobcanthelpyou, 2019-03-31:
Does https://github.com/freifunk-gluon/packages/pull/214 fix that issue?

RalfJung, 2019-03-31:
Well, that applies a local patch instead of updating upstream. I assume usually updating upstream is preferred?

bobcanthelpyou, 2019-03-31:
You mean updating the `PKG_SOURCE_VERSION` (and maybe `PKG_SOURCE_DATE`) in the tunneldigger `Makefile`?
https://github.com/freifunk-gluon/packages/blob/master/net/tunneldigger/Makefile#L9

RalfJung, 2019-03-31:
Yes.

bobcanthelpyou, 2019-03-31:
Then maybe https://github.com/freifunk-gluon/packages/pull/215 will update the tunneldigger package.

rotanid, 2019-04-02:
new PR is #216 but it still needs some love

kaechele, 2019-04-20:
Can be closed as freifunk-gluon/packages#216 is merged and https://github.com/freifunk-gluon/gluon/commit/01fd25de429e3a92d93ab551520f0500210daa1d happened.

closed 2019-04-20 by NeoRaider