freifunk-gluon/gluon #1689: Make Gluon packages available via opkg

bobcanthelpyou, 2019-03-31:
Several packages¹² of Gluon are available during build via `site.mk`, only. And can't be installed via `opkg`, later.

Could it be useful to make some of these packages available via `opkg` on a site mirror? I know some packages require an additional configuration, these should be skipped.

What is needed to build and provide additional packages where no site configuration is required? Or do i miss something and it is already possible and these packages should be available?

¹ https://github.com/freifunk-gluon/packages
² https://github.com/freifunk-gluon/gluon/tree/master/package

NeoRaider, 2019-04-22:
While Gluon packages are generally designed to make separate installation possible in theory, there are various problems in practice:
- Many packages require specific site.conf settings
- Many packages define general firmware features that should be configured uniformly across a whole mesh
- Installed packages don't survive upgrades; in some cases, this also leads to loss of configuration, as packages usually define which config files to preserve on upgrades themselves

For these reasons, we didn't make the effort to support generating opkg repos from Gluon packages.

mweinelt, 2019-04-22:
The benefit is not really obvious to me either, it would be more plausible to simply rebuild Gluon with the required packages and/or discuss the package selection in your community.

closed 2019-04-22 by mweinelt

SvenRoederer, 2019-10-27:
I found this issue based on #1846 

> * Many packages require specific site.conf settings

This might indeed require some rework of the current code / system.
> 
> * Many packages define general firmware features that should be configured uniformly across a whole mesh
> 

Such issues needs to be handeled by convention. Even with the current layout of the core-packages a user is free to build it's own firmware with incomplete / incompatible features of the current mesh.

> * Installed packages don't survive upgrades; in some cases, this also leads to loss of configuration, as packages usually define which config files to preserve on upgrades themselves

For the Freifunk-Berlin Firmware we use the imagebuilder to embed the relevant packages in the firmware-image and ensure this way to packages are present.