freifunk-gluon/gluon #1691: TX-Powerloss since 2018.2 - 2018.1.x was fine

biggesee, 2019-04-04:
Hi,

since updating to 2018.2 (2018.2.1) the transmit power is defnitly lower than it was before (like 2018.1).

Example: tp-link 940v3 (941v6)
Mode: Mesh Point Channel: 13 (2.472 GHz)
Tx-Power: 14 dBm Link Quality: 35/70

Sadly it's not possible to gain the tx-power - set uci wireless.radio0.txpower to "20" has defnitly no effekt (uci commit/wifi/reboot), the device is still on (low) TX-Power 14...

Could anybody confirm this behavior?
(Meanwhile some nodes were downgrades to 2018.1.3 - see also https://forum.freifunk.net/t/tx-powerloss-gluon-2018-2-2018-2-1/20356/1

Cheers
Frank

bobcanthelpyou, 2019-04-04:
I could also confirm a txpower drop with a TP-Link TL-WR740N/ND v4 switching from v2018.1.4 to v2018.2.1 (at least the output of `iw` was different before/after the upgrade, i don't know if the real dBm usage is different now).

rotanid, 2019-04-04:
i'm not really happy with the title of this report, especially the "was fine" part.
maybe it was not fine, but too high before?
your report is missing information like:
- is more than one type of device affected
- is more than one wifi driver affected
- how about trying plain openwrt 17.01.x and 18.06.x ?
if the latter is also "affected", it has to be discussed upstream - and this is very likely as gluon usually doesn't change things in this area.

biggesee, 2019-04-05:
> your report is missing information like:

1.) is more than one type of device affected
2.) is more than one wifi driver affected
3.) how about trying plain openwrt 17.01.x and 18.06.x ?
> 

1.) => yes, serveral devices (tp-link: wr801v2, wr841v9, wr841v11, wr940v3, wr1043,v2) 
2.) => wifi chipsets: Qualcomm Atheros QCA9558, QCA9533, QCA9561 
3.) => have to check this later

rotanid, 2019-04-05:
regarding wifi chipsets: these are all ath9k chipsets, are there any non-ath9k chips affected?

btw., neither us nor freifunk darmstadt noticed such problems with 2018.2.... i havent measured it though.
you don't seem to use custom patches or packages in your 2018.1-based release which would affect txpower?

(noticed also, you ignored our bug report template when creating this issue :-P )

biggesee, 2019-04-05:
> how about trying plain openwrt 17.01.x and 18.06.x ?

Dammed, UR right - this is an openwrt - not a gluon - thing...

`BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06.2, r7676-cddd7b4c77
 -----------------------------------------------------
**root@OpenWrt:~# iwinfo**
wlan0     ESSID: "mesh-my-tiny mesh-11n"
          Access Point: F8:1A:67:90:55:42
          Mode: Mesh Point  Channel: 13 (2.472 GHz)
          **Tx-Power: 16 dBm**  Link Quality: 31/70
          Signal: -79 dBm  Noise: -95 dBm
          Bit Rate: 13.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

**root@OpenWrt:~# uci show wireless**
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/ar934x_wmac'
wireless.radio0.htmode='HT20'
wireless.radio0.legacy_rates='1'
wireless.radio0.country='DE'
wireless.radio0.channel='13'
**wireless.radio0.txpower='20'**
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.encryption='none'
wireless.default_radio0.mode='mesh'
wireless.default_radio0.mesh_id='mesh-ln-11n'
wireless.default_radio0.mesh_fwding='1'
root@OpenWrt:~#
`

> (noticed also, you ignored our bug report template when creating this issue :-P )
sorry - did not know there is thomthing like that

Anyway - it's abaout time to close this "issue"

closed 2019-04-05 by biggesee

rubo77, 2019-04-05:
@biggesee , so is the problem solved? or did you re-post it as an OpenWRT issue somewhere?

rotanid, 2019-04-05:
we dont have to close it necessarily, just need to know that nothing will/can be fixed here.
@rubo77 let me repeat, that it's not clear yet if it really is a problem - or if it was wrong before and is correct now ;-)

biggesee, 2019-04-08:
>  , so is the problem solved? 
no - I closed dtis case becouse its not Gluon related

> or did you re-post it as an OpenWRT issue somewhere?
No - most of "the others outside" do not seem to be affected.

Still struggling with with my own "field research" - it looks like an Archer C50 (V3) sending 100mW  TX-Power (Device with ramips mt76x8 chipset)

rotanid, 2019-04-08:
not affected, well, that we don't know about problems so far does not mean there aren't any. maybe we just don't rely on strong mesh connections or we haven't checked as good as you...
can this be reproduced in a  "lab environment" (and again, not by reading the values which the router says, but by some TQ value or similar) ?

biggesee, 2019-04-09:
Hello Andreas,

just to avoid missunderstandings...

> can this be reproduced in a "lab environment" (and again, **not by reading the values which the router says, but by some TQ value or similar)** ?

... plaes tell me exacly what you want me to (if possible a copy / translation in geman language wopuld be extremly helpful 4 me)

rubo77, 2019-04-09:
Kann dies reproduziert werden im Labor durch auslesen der tq Werte oder ähnlichen? (Die Werte, die der Router intern anzeigt sind nicht aussagekräftig)

Maybe reopen this issue meanwhile?

biggesee, 2019-04-09:
Um eich hier nicht mit unbrauchbaren Ergebnissen aus falschen Messreihen zu langweilen faage ich nich mal explizit nach: WAS soll ich WO und WIE "messen" wenn die im Router angezeigten Werte NICHT aussagekräftig sind - eine Anleitung für DOOFE wäre hilfreich ;-)

rotanid, 2019-04-09:
i'm no professional in wifi measurements at all.
maybe you could set up two nodes with the old firmware in places so they have a connection, but no good one, and to speed measurements through this connection.
update and repeat the exact same measurements afterwards (and be sure to not do one measurement at night and one during the day, which also impacts the results)