freifunk-gluon/gluon #1694: Using make and GLUON_DEBUG=1 is unsuccessful for several devices

bobcanthelpyou, 2019-04-04:
**What is the problem?**

I tried to use `export GLUON_DEBUG=1` together with `make GLUON_TARGET=ar71xx-generic`. But the build got stuck

```
make[1]: Leaving directory '/home/gluon/gluon-2018.2.x-experimental/openwrt'
cp: cannot stat 'openwrt/bin/targets/ar71xx/generic/openwrt-ar71xx-generic-hornet-ub-squashfs-factory.bin': No such file or directory
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 1
```

The result is
```
/home/gluon/gluon-2018.2.x-experimental/output
└── [4.0K]  images
    ├── [4.0K]  factory
    ├── [4.0K]  other
    └── [4.0K]  sysupgrade
        ├── [ 12M]  gluon-ffil-2018.2.1.0-ffil~20190403-8devices-carambola2-board-sysupgrade.bin
        └── [ 12M]  gluon-ffil-2018.2.1.0-ffil~20190403-alfa-network-ap121f-sysupgrade.bin
```

What are the requirements for a Debug build at the moment?

**What is the expected behaviour?**

Build all debug firmware images. If not possible ignore unsuccessful builds of devices if Debug is enabled.

**Gluon Version:**

https://github.com/freifunk-gluon/gluon/releases/tag/v2018.2.1

**Site Configuration:**

https://github.com/freifunk-ilmenau/site-ffil/tree/v2018.2.1

NeoRaider, 2019-04-04:
GLUON_DEBUG significantly increases the rootfs size; this causes many image builds to fail due to size constraints. Building only a single device, preferably one with 16MB of flash or more, is recommended.

Also, on the more constrained targets (basically everything that is based on MIPS), a full debug build rarely helps. I often add debug flags only to individual package Makefiles (modifying TARGET_CFLAGS and overriding STRIP to `:` (for example used in `package/firmware/layerscape/ppfe-firmware/Makefile`) might be enough - it's been a while since I last used this).

Basically, GLUON_DEBUG is only expected to do anything useful for x86 and similar targets...

rotanid, 2019-04-04:
hm, so closing as "wontfix"/"cantfix"

closed 2019-04-04 by rotanid

bobcanthelpyou, 2019-04-06:
Would you allow something like the following (it is just a draft) to skip missing images (for whatever reason) and continue copying?
```
--- a/scripts/copy_output.sh
+++ b/scripts/copy_output.sh
@@ -64,12 +64,16 @@ do_copy() {
        local file="$(get_file "$dir" "$out_suffix" "$ext" "$output")"
 
        do_clean "$dir" "$out_suffix" "$ext" "$output"
+       if [ -e "openwrt/bin/targets/${OPENWRT_BINDIR}/openwrt-${OPENWRT_TARGET}${profile}${in_suffix}${ext}" ]; then
                cp "openwrt/bin/targets/${OPENWRT_BINDIR}/openwrt-${OPENWRT_TARGET}${profile}${in_suffix}${ext}" "$file"
 
                for alias in $aliases; do
                        do_clean "$dir" "$out_suffix" "$ext" "$alias"
                        ln -s "$(basename "$file")" "$(get_file "$dir" "$out_suffix" "$ext" "$alias")"
                done
+       else
+               echo "IMAGE MISSING: ${OPENWRT_TARGET}${profile}${in_suffix}${ext}"
+       fi
 }
```

NeoRaider, 2019-04-06:
No, a missing image always means that something has gone wrong. Aborting the build is the correct behaviour in this case.