freifunk-gluon/gluon #1701: How to prevent a respondd connection between different sites

bobcanthelpyou, 2019-04-11:
What i have:

* TP-Link TL-MR3020 v1
  * Gluon-v2018.2.1 (https://github.com/freifunk-ilmenau/site-ffil)
  * Freifunk Ilmenau (sub site of Freifunk Erfurt)
  * https://map.erfurt.freifunk.net/#/en/map/f4f26d6cfa4e
* TP-Link TL-WDR4300 v1
  * Gluon-v2018.1.4 (https://github.com/freifunkhamburg/site-ffhh)
  * Freifunk Hamburg (domain Hamburg-Sued)
  * https://map.hamburg.freifunk.net/#!v:m;n:30b5c23839e3

What i did:

The WDR4300 has an uplink connection via VPN. I connected a LAN port of the WDR4300 with the WAN port of the MR3020. The MR3020 has its own VPN connection.

What happened:

The MR3020 appeared on the map of Freifunk Hamburg
https://map.hamburg.freifunk.net/#!v:m;n:f4f26d6cfa4e

What i checked:

Running tcpdump on the LAN interface of the WDR4300 and checking for port 1001 (respondd). There is some traffic from an ipv6 address (high port) to the multicast address (port 1001) and traffic from the br-wan address (port 1001) of the MR3020 to an ipv6 address (high port).

What i need to do:

@mweinelt suggested to check the respondd traffic on the br-wan interface of the MR3020. But i didn't setup that tiny MR3020 device with `tcpdump` nor did i checked `batctl`, yet.

Is it somehow possible to prevent that from happening? Or is my setup just to wired/weird?

mweinelt, 2019-04-11:
From what I understand the WAN zone has an INPUT REJECT policy and only DHCP, ICMP, IGMP, DHCPv6, MLD, ICMPv6 and VXLAN are allowed, Not sure how this is happening as its udp/1001.

Respondd certainly does a wildcard bind, so if the packet passes through the firewall it will be answered.

```
udp        0      0 :::1001                 :::*                                3391/respondd
```

bobcanthelpyou, 2019-04-16:
tcpdump on the MR3020 (more or less the same i see on the WDR4300 side)
```
tcpdump -i br-wan -nn port 1001
listening on br-wan, link-type EN10MB (Ethernet), capture size 262144 bytes
22:11:02.338853 IP6 fe80::dcad:[..].45123 > ff02::1.1001: UDP, length 12
22:11:02.382468 IP6 fe80::b83b:[..].1001 > fe80::dcad:[..].45123: UDP, length 408
22:11:02.546855 IP6 fe80::dcad:[..].45123 > ff02::1.1001: UDP, length 14
22:11:02.548517 IP6 fe80::dcad:[..].45123 > ff02::1.1001: UDP, length 14
22:11:02.560630 IP6 fe80::b83b:[..].1001 > fe80::dcad:[..].45123: UDP, length 117
22:11:02.587319 IP6 fe80::b83b:[..].1001 > fe80::dcad:[..].45123: UDP, length 469
```
`fe80::b83b:[..]` is the address of the br-wan of MR3020
`fe80::dcad:[..]` seems to originate from connected gateway of WDR4300

`iptables -L` as attachment: [iptables.txt](https://github.com/freifunk-gluon/gluon/files/3086781/iptables.txt)

NeoRaider, 2019-04-16:
We currently explicitly allow respondd queries on WAN. Before VXLAN, this was necessary for respondd to work for Mesh-on-WAN...

The removal of the legacy wired meshing without VXLAN should fix this as well, but I'm not sure what we can do in the mean time. There might be some improvements possible, for example:
- Only allow respondd queries on WAN when Mesh-on-WAN is enabled
- Do not respond to mesh-wide respondd queries on non-mesh interfaces

The second option seems like a good idea - the mesh-wide respondd address ff05::2:1001 was added in Gluon v2017.1, deprecating the old ff02::2:1 for mesh-wide queries, so hopefully all query tools have migrated by now (I'm sure some haven't...).

rotanid, 2019-04-16:
vote for 2nd option

closed 2019-04-21 by NeoRaider

bobcanthelpyou, 2019-04-23:
Thank you very much, updated MR3020 to Gluon master and it seems promising :)

commit 59a44274 (2019-04-21): gluon-respondd: do not join link-local multicast group on br-client

Fixes #1701