freifunk-gluon/gluon #1711: Network rewrite

CodeFetch, 2019-04-20:
This is an important milestone https://github.com/freifunk-gluon/gluon/milestone/10.
Without it, it is hard to modularize packages and introduce more network configuration options.

As far as I remember @NeoRaider suggested to use network namespaces.
Can you please explain the steps to be taken in detail?

We'll likely get funding for special software development projects at the end of the year. I'd like to submit this as a proposal. If someone is willing and capable of doing this, please tell me how much this would cost roughly.

NeoRaider, 2019-04-22:
While network namespaces are one possible part of the puzzle, the core of my idea for a more flexible network configuration is to get rid of UCI (at least for network-related config), netifd and some other components.

For most of this, there is no clear approach yet - I think the right way to go forward is to start with some prototypes and try out what "feels good", from a developer perspective.

Some material I've produced during the last months:
- https://md.darmstadt.ccc.de/pluggable-network-configuration
- https://github.com/NeoRaider/network-examples
- https://git.universe-factory.net/neco/

None of this is usable in its current state, the implementation of neco is a very early prototype with essential features missing even for basic network configuration, but it might give an overview of what I want to achieve.

Regarding " Without it, it is hard to modularize packages and introduce more network configuration options." ... I'm not sure where to go with this. What concrete config options do you have in mind? Where do you want to improve modularization? Too many options and too much modularization also lead to high complexity...

CodeFetch, 2019-04-22:
This looks very promising. If getting rid of UCI is the way to go, then it should be done. 
Most features are described in https://github.com/freifunk-gluon/gluon/issues/284 and in the NeCo description you've provided.

Personally I think of a new UI for network configuration and easier exchangeability of used mesh protocols without having to modify the core network configuration.

Assume we start without a default/(defined in site.conf) network configuration without any WLAN VIFs and the LAN/WAN ports not assigned to anything. The config-mode has a default network configuration equivalent to the one used at the moment of course.
We should be able to create VIFs in the WLAN section and assign them a SSID, Key etc. This creates new ports/VIF interfaces which can then be assigned a function in the network section depending on their type. E.g. Uplink-bridge/(replacement of existing Private WLAN without DHCP), NAT-Bridge/(additional Private WLAN with DHCP), Client-WLAN/(br-client), Mesh-WLAN/(which can of course only be used if the VIF is a mesh interface). These functions should be provided by packages and not be predefined by the generic Gluon network configuration. That is what I want to say with "being modular". For a concrete planning of the functions, I'd need to sit down a week or two to think about it.

NeCo seems to tackle most of this. Thus I'd love to see NeCo becoming reality.

Yes, as you say the network configuration in the expert mode will be a little more complex with a higher modularity, but I assume that actually the complexity of the network configurations in other (e.g. mesh, VPN, Private WLAN) packages will decrease with a higher modularity and it would be easier to support new meshing protocols like BMX or whatever special things people want to do with the nodes like setting up an independent web-server on a Gluon node as e.g. a map, monitoring or update server.

With Babel, NAT426, NAT64, my N2N/N2S/S2N/S2S approach etc. there won't be a real difference between a node and a supernode anymore when the packages have been modularized. T-X works on similar things using Batman. Personally I'd like to be able to run Gluon in a KVM as a supernode and provide additional packages for e.g. using Mullvad/GRE/WireGuard or whatever for the exit of traffic. At the moment the network configuration for this is a pain and needs patching in the generic network configuration. It should be much easier for communities whether they are in Spain, France, Germany, South America or Africa to setup supernodes. I also want to be able to run a node as a ordinary node and a supernode at the same time in the case where I have a uplink that can be shared without legal problems. That are only some of my personal expectations of what the network rewrite could allow us to do.

CodeFetch, 2019-05-26:
@NeoRaider I've sent you an email regarding funding.