freifunk-gluon/gluon #1717: respondd crashes with Arithmetic exception

Sprinterfreak, 2019-04-30:
### Bug report

**What is the problem?**

Today I built freifunk-gluon with site-ffrgb/hybrid and few local changes to the official repository:
- use openwrt instead of lede
- removed basic_rate configuration
- force gluon-repo commit GLUON_GIT_REF=ad1a1906ab2aa620417406cc8d112d7549b56640

**What is the expected behaviour?**
After flashing firmware to the Node, Hopglass says it's offline but shows uptime and has active mesh connections. Visiting the local status page worked a few times. Than just an error appears:
```
This page contains the following errors:
error on line 7 at column 10: internal error: detected an error in element content
```
So I took a lok onto the node via ssh. respondd was gone. `/etc/init.d/respondd start` started it and the status page works again. But after few reloads respondd finally crashed again. So I started respondd manually on the cli and reloaded the status page. This is what happend:
```
~# /usr/bin/respondd -d /usr/lib/respondd -p 1001 -g ff02::2:1001 -i ibss0 -i ibss1 -i mesh0 -i mesh-vpn
Arithmetic exception
```
So I narrowed down the affected URL and figured out that respondd instantly crashes with Arithmetic exception if `/cgi-bin/dyn/neighbours-nodeinfo?ibss0` is requested. Other mesh interfaces do not crash respondd but don't have nodes connected to them either. That request wich crashes respondd is served fine and gives normal event-json as expected but after that respondd is gone.
```
event: neighbour
data: {"software":{"autoupdater":{"branch":"stable","enabled":true},"batman-adv":{"version":"2018.1","compat":15},"fastd":{"version":"v18","enabled":false},"firmware":{"base":"gluon-v2018.1","release":"v2018.1.3"}},"network":{"addresses":["2a00:9d80:6000:102:6a::abc","fdef:f20f:1337:cafe::abc","fe80::6a::abc"],"mesh":{"bat0":{"interfaces":{"wireless":["ea:bc:aa:aa:aa:aa"],"other":["ea:bc:aa:aa:aa:aa","ea:bc:aa:aa:aa:aa"]}}},"mac":"68:72:aa:aa:aa:aa"},"location":{"latitude":49.02800000,"longitude":12.21000000,"altitude":6.00000000},"owner":{"contact":"hidden@email.com"},"system":{"site_code":"ffrgb","domain_code":"ffrgb_uml"},"node_id":"68aaaaaaaaaa","hostname":"hidden-name","hardware":{"model":"Ubiquiti NanoStation M2","nproc":1}}

event: eot
data: null
```

**Gluon Version:**
v2018.2-104-gad1a1906

**Site Configuration:**
github.com/ffrgb/site-ffrgb.git (3cac72d)

**Patches**
```
From 2760947711d51f881f2ca59f21263f1d68b2588a Mon Sep 17 00:00:00 2001
From: =?UTF-8?q?Jan-Jonas=20S=C3=A4mann?= <sprinterfreak@localhorst>
Date: Mon, 29 Apr 2019 22:38:38 +0200
Subject: [PATCH] Make site config comply with gluon v2018.2.1

* remove obsolete basic_rate parameter
* switch back from lede to openwrt
---
 .gitlab-ci.yml         |  6 ++++++
 Makefile               | 20 +++++++++++++-------
 domains/ffrgb_cty.conf |  3 ---
 domains/ffrgb_tst.conf |  3 ---
 domains/ffrgb_uml.conf |  3 ---
 site.conf              |  6 ------
 6 files changed, 19 insertions(+), 22 deletions(-)
 create mode 100644 .gitlab-ci.yml

diff --git a/.gitlab-ci.yml b/.gitlab-ci.yml
new file mode 100644
index 0000000..8fb25b3
--- /dev/null
+++ b/.gitlab-ci.yml
@@ -0,0 +1,6 @@
+before_script:
+  - sudo apt-get update -qq && sudo apt-get install -y -qq build-essential gawk unzip libncurses-dev libz-dev libssl-dev wget git subversion python-pip python3-pip
+
+fullbuild:
+  script:
+    - make V=s FORCE_UNSAFE_CONFIGURE=1 GLUON_BRANCH=experimental GLUON_RELEASE=v2018.2.1 GLUON_GIT_REF=ad1a1906ab2aa620417406cc8d112d7549b56640 GLUON_TARGETS="ar71xx-generic ar71xx-nand ipq40xx"
diff --git a/Makefile b/Makefile
index 8f5af04..24248cd 100644
--- a/Makefile
+++ b/Makefile
@@ -1,23 +1,29 @@
-GLUON_BUILD_DIR := gluon-build
-GLUON_GIT_URL := https://github.com/freifunk-gluon/gluon
-GLUON_GIT_REF := v2018.1
+GLUON_BUILD_DIR ?= gluon-build
+GLUON_GIT_URL ?= https://github.com/freifunk-gluon/gluon
+GLUON_GIT_REF ?= 2ce07ae2fe251bcf8a424dc39aba8da1df7f164d
 
 SECRET_KEY_FILE ?= ${HOME}/.gluon-secret-key
 
 GLUON_TARGETS ?= \
 	ar71xx-generic \
-	x86-64 \
-	ar71xx-tiny \
+	ar71xx-mikrotik \
 	ar71xx-nand \
+	ar71xx-tiny \
 	brcm2708-bcm2708 \
 	brcm2708-bcm2709 \
+	brcm2708-bcm2710 \
+	ipq40xx \
+	ipq806x \
 	mpc85xx-generic \
+	mvebu-cortexa9 \
+	ramips-mt7620 \
 	ramips-mt7621 \
+	ramips-mt76x8 \
 	ramips-rt305x \
+	sunxi-cortexa7 \
 	x86-64 \
 	x86-generic \
-	x86-geode \
-	sunxi
+	x86-geode
 
 
 GLUON_RELEASE := $(shell git describe --tags 2>/dev/null)
diff --git a/domains/ffrgb_cty.conf b/domains/ffrgb_cty.conf
index 5524e4e..6b0c8d6 100644
--- a/domains/ffrgb_cty.conf
+++ b/domains/ffrgb_cty.conf
@@ -3,10 +3,7 @@
   domain_seed = '1124de9b0fe2dde424c8a4c481816b5899e6d1b4b4cf72de42136c4f1f197c7a',
 
   opkg = {
-    --[[ old
     openwrt = 'http://downloads.openwrt.org/%n/%v/%S/packages',
-    ]]
-    lede = 'http://downloads.lede-project.org/snapshots/packages/%A',
   },
 
   prefix4 = '10.90.32.0/19',
diff --git a/domains/ffrgb_tst.conf b/domains/ffrgb_tst.conf
index 05416fe..36ab080 100644
--- a/domains/ffrgb_tst.conf
+++ b/domains/ffrgb_tst.conf
@@ -4,10 +4,7 @@
   hide_domain = true,
 
   opkg = {
-    --[[ old
     openwrt = 'http://downloads.openwrt.org/%n/%v/%S/packages',
-    ]]
-    lede = 'http://downloads.lede-project.org/snapshots/packages/%A',
   },
 
   prefix4 = '10.90.96.0/19',
diff --git a/domains/ffrgb_uml.conf b/domains/ffrgb_uml.conf
index 1718e28..42cd336 100644
--- a/domains/ffrgb_uml.conf
+++ b/domains/ffrgb_uml.conf
@@ -3,10 +3,7 @@
   domain_seed = 'f3d54e3dfc4108dd3356664bbab4baef3b2920ad943c27edce4b284a68e776fd',
 
   opkg = {
-    --[[ old
     openwrt = 'http://downloads.openwrt.org/%n/%v/%S/packages',
-    ]]
-    lede = 'http://downloads.lede-project.org/snapshots/packages/%A',
   },
 
   prefix4 = '10.90.64.0/19',
diff --git a/site.conf b/site.conf
index b45cf8c..9905dd3 100644
--- a/site.conf
+++ b/site.conf
@@ -5,10 +5,7 @@
   default_domain = 'ffrgb_cty',
 
   opkg = {
-    --[[ old
     openwrt = 'http://downloads.openwrt.org/%n/%v/%S/packages',
-    ]]
-    lede = 'http://downloads.lede-project.org/snapshots/packages/%A',
   },
 
   timezone = 'CET-1CEST,M3.5.0,M10.5.0/3',
@@ -19,9 +16,6 @@
   regdom = 'DE',
 
   wifi24 = {
-    -- disable 802.11b support:
-    supported_rates = {6000, 9000, 12000, 18000, 24000, 36000, 48000, 54000}, -- List of supported wifi rates (optional)
-    basic_rate = {6000, 9000, 18000, 36000, 54000}, -- List of basic wifi rates (optional, required if supported_rates is set)
     mesh = {
       mcast_rate = 12000,
     },
-- 
2.17.1

```

NeoRaider, 2019-05-01:
If this is easily reproducible, please try the following:
- Build an image with `gdb` in its GLUON_SITE_PACKAGES and GLUON_DEBUG=1
- Run respondd in gdb from the command line: `gdb --args /usr/bin/respondd -d /usr/lib/respondd -p 1001 -g ff02::2:1001 -i ibss0 -i ibss1 -i mesh0 -i mesh-vpn`
gdb should then interrupt respondd when the bug occurs. Use the `bt` command to print a backtrace and post it.

Sprinterfreak, 2019-05-01:
Allready fiddled around with gdb yesterday. Today I built x86-64 too. Same thing but kernel logs a message:
```
[   95.626788] traps: respondd[3482] trap divide error ip:401ec5 sp:7ffdf540b940 error:0 in respondd[400000+6000]
```
Build with GLUON_DEBUG is in progress...

Sprinterfreak, 2019-05-01:
```
   ┌──/home/gitlab-runner/builds/veVATRzi/0/sprinterfreak/site-ffrgb/gluon-build/openwrt/build_dir/target-x86_64_musl/respondd-1/respondd.c─────────────────────────────────┐
   │601                     break;                                                                                                                                          │
   │602             }                                                                                                                                                       │
   │603                                                                                                                                                                     │
   │604             input[input_bytes] = 0;                                                                                                                                 │
   │605                                                                                                                                                                     │
   │606             const struct interface_info *iface = NULL;                                                                                                              │
   │607             if (IN6_IS_ADDR_MULTICAST(&destaddr)) {                                                                                                                 │
   │608                     iface = find_multicast_interface(groups, ifindex, &destaddr);                                                                                   │
   │609                     // this should not happen                                                                                                                       │
   │610                     if (!iface)                                                                                                                                     │
   │611                             return;                                                                                                                                 │
   │612             }                                                                                                                                                       │
   │613                                                                                                                                                                     │
   │614             struct request_task *new_task = malloc(sizeof(*new_task));                                                                                              │
   │615             // input_bytes cannot be greater than REQUEST_MAXLEN-1                                                                                                  │
   │616             memcpy(new_task->request, input, input_bytes + 1);                                                                                                      │
   │617             new_task->scheduled_time = 0;                                                                                                                           │
   │618             new_task->client_addr = addr;                                                                                                                           │
   │619                                                                                                                                                                     │
   │620             bool is_scheduled;                                                                                                                                      │
   │621             if (iface) {                                                                                                                                            │
   │622                     // scheduling could fail because the schedule is full                                                                                           │
  >│623                     new_task->scheduled_time = now + rand() % iface->max_multicast_delay;                                                                           │
   │624                     is_scheduled = schedule_push_request(schedule, new_task);                                                                                       │
   │625             } else {                                                                                                                                                │
   │626                     // unicast packets are always sent directly                                                                                                     │
   │627                     is_scheduled = false;                                                                                                                           │
   │628             }                                                                                                                                                       │
   │629                                                                                                                                                                     │
   │630             if (!is_scheduled) {                                                                                                                                    │
[... cut ...]
   └────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
native process 3912 In: main                                                                                                                              L623  PC: 0x401ec5 
(gdb) p iface->max_multicast_delay
$1 = 0
(gdb) 606       in /home/gitlab-runner/builds/veVATRzi/0/sprinterfreak/site-ffrgb/gluon-build/openwrt/build_dir/target-x86_64_musl/respondd-1/respondd.c
```

commit 7f0299b5 (2019-05-01): modules: update Gluon packages

9a364fb8d0a4 respondd: fix crash caused by division by zero
67177de99afc libs/lua-math-polygon: add add math lib for gauss jordan Ponit in Polygon algorithm

Fixes #1717
closed 2019-05-01 by NeoRaider