freifunk-gluon/gluon #1718: [DOC] GLUON_TARGET=ipq40xx needs BROKEN=1, but it is listed as supported

yanosz, 2019-05-02:
### Bug report

**What is the problem?**
* Is listed as supported https://gluon.readthedocs.io/en/v2018.2.x/index.html?highlight=ipq40xx
* Building ipq40xx fails.
```
make GLUON_TARGET=ipq40xx
Please set GLUON_TARGET to a valid target. Gluon supports the following targets:
```
* Using `BROKEN=1` works.

**What is the expected behaviour?**
* The docs reflect the status of ipq40xx (i.e. needs `BROKEN=1`)
* Issues requiring `BROKEN=1` are listed in the corresponding sections in chapter "Supported Devices & Architectures"

**Gluon Version:**

v2018.2.1-6-g6229c599

CodeFetch, 2019-05-02:
Have a look at the footnote:

> (1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24) Device does not support IBSS; images are built by default unless GLUON_WLAN_MESH is explicitly set to something other than 11s

I assume you're trying to build it with IBSS mesh. Unfortunately the IBSS ath10k driver (candelatech) is a bit buggy at the moment, but Ben Greear is currently trying to fix it.

yanosz, 2019-05-02:
hmm.. thanks for your feedback. ... hard one. 

Maybe you're right, I rember the 11s built was failing, too, but the logs are purged after the sucessfull one. I guess, that the situation is somewhat confusing. I read the docs multiple times, but I didn't notices, that theses things might be related.

The note appears to be applicable to other GLUON_TARGET than ipq40xx, but the situation "Please set GLUON_TARGET to a valid target" shows no relation to the select mesh type. I guess, that this is due to all ipq40xx devices require 802.1s - in contrast others - e.g. ramips-mt7620. That feels a little bit confusing and inconsistent.

I guess, it'll be better, to explain this situation more explicit: Either in the error message or in the docs or both.

rotanid, 2019-05-02:
the targets you compare behave the same?
https://github.com/freifunk-gluon/gluon/blob/master/targets/targets.mk#L16-L17

some targets don't support IBSS, so they are not built if you choose IBSS unless you also set BROKEN=1

so its 11s **OR** BROKEN for those (not XOR, not AND)

yanosz, 2019-05-03:
@rotanid : You're correct about describing the code. I was able to reproduce this.

I'm not able to get this from the docs. The docs suggest, that firmware images are not built. IMHO this differs from: The whole architecture doesn't build.

CodeFetch, 2019-05-03:
It does not necessarily correlate with the architecture. It would be false information if we say, that e.g. the platform for ar71xx does not work, because only ath10k devices are affected. It's the WLAN chip that is affected. Ben said he is going to fix at least wave 2 chips. If he doesn't fix wave 1 chips, it's getting even more complicated which devices are supported and which not. Thus it is a good idea to add the footnote on a per device basis as it already is.

yanosz, 2019-05-03:
@CodeFetch  That's not the point. It's fine to have a per device footnote.

The point is: I don't expect a complete architecture to be missing from GLUON_TARGET after reading the foodnotes.

CodeFetch, 2019-05-03:
What about adding the footnote to the architecture name, too?
As in the case of ipq40xx all devices come with an integrated ath10k chip.

yanosz, 2019-05-03:
@CodeFetch Good idea!

Maybe: Add a comment: To build ipq40xx, `GLUON_WLAN_MESH=11s` must be set in `site.mk`. If not, the target cannot be selected and `make GLUON_TARGET=ipq40xx` will fail.

CodeFetch, 2019-05-03:
You can build it with IBSS and broken, but it might be buggy. Maybe we should add a note right at the beginning before all targets:

> Currently the ath10k driver for meshing using IBSS is unstable. The affected devices and targets are marked with [2]. Images for these devices can still be build if 802.11s is being used (GLUON_WLAN_MESH set to 11s). For experimental purposes the BROKEN flag can be set to 1, to allow building images for these devices with IBSS.

yanosz, 2019-05-04:
@CodeFetch can I really built with IBSS ipq40xx? I tried v2018.2.x and GLUON_TARGET=ipq40xx was invalid. That caused the initial report ...

@rotanid Could you swap the invalid label with a doc label please? I guess, that fits better.

rotanid, 2019-05-04:
a ibss+ipq40xx *build* might work, but you need to set BROKEN=1 as ibss probably wont work, thats the reason for this after all

rotanid, 2019-06-29:
@yanosz maybe you want to improve the docs on this now that you know how it's meant?

yanosz, 2019-07-11:
Nope - I'm not interested in creating new pull-requests for gluon at the moment.

CodeFetch, 2019-07-11:
As IBSS support will be dropped, I'm not sure if we need this anymore.

yanosz, 2019-07-11:
@CodeFetch : I think, that broken (definition, impact, etc.) is not mentioned in the docs as of today ( or I'ven't found /o\). 
No matter what criteria is needed - the docs could be more specific / useful (c.f. #1687) here.

NeoRaider, 2019-07-20:
See #1788

commit 0a230761 (2019-07-20): docs: improve note on targets without IBSS support

Fixes #1718
PR #1788 (2019-07-20): docs: improve note on targets without IBSS support
Fixes #1718
commit df5bf20b (2019-07-20): docs: improve note on targets without IBSS support (#1788)

Fixes #1718
commit 76988870 (2019-08-26): docs: improve note on targets without IBSS support (#1788)

Fixes #1718
closed 2019-07-20 by NeoRaider