freifunk-gluon/gluon #1723: User-friendly units in traffic shaper

mweinelt, 2019-05-13:
The traffic shaper currently expects their limit to be given in kbit/s. Endusers buy their internet uplinks specified in Mbit/s and mostly think in those scales when trying to limit the bandwidth in relation to their uplink.

Specifying the limit in kbit/s gives an unnecessary level of detail, as most links actually need at least some Mbit/s to actually work.

We recently had a user who, by error, configured 100/40 and didn't notice the unit was off.

rotanid, 2019-05-13:
i agree this change might make sense, but there needs to be either some kind of migration - or we still save it as kbit/s and what the user enters is transformed into kbit/s ?

mweinelt, 2019-05-13:
Yep, save in kbit/s, as that's what we need to pass to tc.

rotanid, 2019-06-29:
fixed w ada723a0df106894bb1c25e063ee7d231d363b5a

closed 2019-06-29 by rotanid