freifunk-gluon/gluon #1726: status-page does not show neighbours with BATMAN_V

krombel, 2019-05-20:
### Bug report

**What is the expected behaviour?**
Mesh partners are shown.

**What is the problem?**
Mesh-Partners are not shown - neither for mesh-vpn nor other.

**Gluon Version:**
v2018.2.1

**Site Configuration:**
Site: ffmuc
Domain: ffmuc_welt
Link: https://github.com/freifunkMUC/site-ffm/blob/0c84212/domains/ffmuc_welt.conf

**Custom patches:**
None

Some hints which we (hexa- and I) to while debugging this:
- We have the strong feeling that the issue here is that BATMAN_V is not supported yet.
- The "wrong" response came from /cgi-bin/neighbours-batadv which is implemented [here](https://github.com/freifunk-gluon/gluon/blob/v2018.2.1/package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c)
- One node where this is failing currently is [here](http://[2001:608:a01:109:12fe:edff:fe57:f6a6]/cgi-bin/status)

ecsv, 2019-05-21:
@ambassador86  your change 582d09615bdcd9d9b6cb5ee74173ab78af3d846d is missing some follow up changes to support the throughput metric (BATADV_ATTR_THROUGHPUT) in:

* package/gluon-mesh-batman-adv/src/respondd.c
* package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c
* package/gluon-radv-filterd/src/gluon-radv-filterd.c (ok, this came after your change)

Right now, all these things depend on BATADV_ATTR_TQ (which is not available with BATMAN_V). Maybe this was planned by you but I just wanted to mention it here since the author of this issue report assumes that these things should be supported.

Maybe you or @txt-file have the patches to get the support for BATADV_ATTR_THROUGHPUT somewhere. At least Zwickau told me in the past that they were using BATMAN_V with your Chemnitz-Umland firmware and it also seems to me that I can see their neighbor information on your map https://map.chemnitz.freifunk.net/#!v:m;n:6466b3bec204

ecsv, 2019-05-21:
Basic steps (just take this as a rough idea - not the complete solution) which someone has to do are following:

    diff --git a/package/gluon-mesh-batman-adv/src/respondd.c b/package/gluon-mesh-batman-adv/src/respondd.c
    index 810dfc59..5cc3ea04 100644
    --- a/package/gluon-mesh-batman-adv/src/respondd.c
    +++ b/package/gluon-mesh-batman-adv/src/respondd.c
    @@ -627,7 +627,6 @@ static struct json_object * ifnames2addrs(struct json_object *interfaces) {
     static const enum batadv_nl_attrs parse_orig_list_mandatory[] = {
     	BATADV_ATTR_ORIG_ADDRESS,
     	BATADV_ATTR_NEIGH_ADDRESS,
    -	BATADV_ATTR_TQ,
     	BATADV_ATTR_HARD_IFINDEX,
     	BATADV_ATTR_LAST_SEEN_MSECS,
     };
    @@ -638,6 +637,7 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
     	struct nlmsghdr *nlh = nlmsg_hdr(msg);
     	struct batadv_nlquery_opts *query_opts = arg;
     	struct genlmsghdr *ghdr;
    +	uint32_t throughput;
     	uint8_t *orig;
     	uint8_t *dest;
     	uint8_t tq;
    @@ -666,9 +666,12 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
     				      BATADV_ARRAY_SIZE(parse_orig_list_mandatory)))
     		return NL_OK;
     
    +	if (!attrs[BATADV_ATTR_TQ] &&
    +	    !attrs[BATADV_ATTR_THROUGHPUT])
    +		return NL_OK;
    +
     	orig = nla_data(attrs[BATADV_ATTR_ORIG_ADDRESS]);
     	dest = nla_data(attrs[BATADV_ATTR_NEIGH_ADDRESS]);
    -	tq = nla_get_u8(attrs[BATADV_ATTR_TQ]);
     	hardif = nla_get_u32(attrs[BATADV_ATTR_HARD_IFINDEX]);
     	lastseen = nla_get_u32(attrs[BATADV_ATTR_LAST_SEEN_MSECS]);
     
    @@ -692,7 +695,16 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
     		json_object_object_add(opts->interfaces, ifname, interface);
     	}
     
    -	json_object_object_add(obj, "tq", json_object_new_int(tq));
    +	if (attrs[BATADV_ATTR_TQ]) {
    +		tq = nla_get_u8(attrs[BATADV_ATTR_TQ]);
    +		json_object_object_add(obj, "tq", json_object_new_int(tq));
    +	}
    +
    +	if (attrs[BATADV_ATTR_THROUGHPUT]) {
    +		throughput = nla_get_u32(attrs[BATADV_ATTR_THROUGHPUT]);
    +		json_object_object_add(obj, "throughput", json_object_new_int(throughput));
    +	}
    +
     	json_object_object_add(obj, "lastseen", json_object_new_double(lastseen / 1000.));
     	json_object_object_add(obj, "best", json_object_new_boolean(attrs[BATADV_ATTR_FLAG_BEST]));
     	json_object_object_add(interface, mac1, obj);
    diff --git a/package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c b/package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c
    index 9c2be2f7..086126ad 100644
    --- a/package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c
    +++ b/package/gluon-status-page-mesh-batman-adv/src/neighbours-batadv.c
    @@ -17,7 +17,6 @@ struct neigh_netlink_opts {
     static const enum batadv_nl_attrs parse_orig_list_mandatory[] = {
       BATADV_ATTR_ORIG_ADDRESS,
       BATADV_ATTR_NEIGH_ADDRESS,
    -  BATADV_ATTR_TQ,
       BATADV_ATTR_HARD_IFINDEX,
       BATADV_ATTR_LAST_SEEN_MSECS,
     };
    @@ -28,6 +27,7 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
       struct nlmsghdr *nlh = nlmsg_hdr(msg);
       struct batadv_nlquery_opts *query_opts = arg;
       struct genlmsghdr *ghdr;
    +  uint32_t throughput;
       uint8_t *orig;
       uint8_t *dest;
       uint8_t tq;
    @@ -54,9 +54,12 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
     	                        BATADV_ARRAY_SIZE(parse_orig_list_mandatory)))
         return NL_OK;
     
    +  if (!attrs[BATADV_ATTR_TQ] &&
    +      !attrs[BATADV_ATTR_THROUGHPUT])
    +    return NL_OK;
    +
       orig = nla_data(attrs[BATADV_ATTR_ORIG_ADDRESS]);
       dest = nla_data(attrs[BATADV_ATTR_NEIGH_ADDRESS]);
    -  tq = nla_get_u8(attrs[BATADV_ATTR_TQ]);
       hardif = nla_get_u32(attrs[BATADV_ATTR_HARD_IFINDEX]);
     
       if (memcmp(orig, dest, 6) != 0)
    @@ -73,7 +76,16 @@ static int parse_orig_list_netlink_cb(struct nl_msg *msg, void *arg)
       if (!neigh)
         return NL_OK;
     
    -  json_object_object_add(neigh, "tq", json_object_new_int(tq * 100 / 255));
    +  if (attrs[BATADV_ATTR_TQ]) {
    +    tq = nla_get_u8(attrs[BATADV_ATTR_TQ]);
    +    json_object_object_add(neigh, "tq", json_object_new_int(tq * 100 / 255));
    +  }
    +
    +  if (attrs[BATADV_ATTR_THROUGHPUT]) {
    +    throughput = nla_get_u32(attrs[BATADV_ATTR_THROUGHPUT]);
    +    json_object_object_add(neigh, "throughput", json_object_new_int(throughput));
    +  }
    +
       json_object_object_add(neigh, "ifname", json_object_new_string(ifname));
       json_object_object_add(neigh, "best", json_object_new_boolean(attrs[BATADV_ATTR_FLAG_BEST]));
 

The changes in gluon-radv-filterd.c are left as an exercise to the reader. You basically have to change all occurrences of tq/max_tq in the datastructures to also store larger values than uint16_t (and name the variables accordingly), find a good hysteresis function/value for the throughput metric and adjust parse_originator like shown above.

There are more changes required to get support for the throughput value in various gather daemons and frontends (status page, maps, ...)

ambassador86, 2019-05-24:
Hello @krombel @ecsv ,
i know that the status page support is missing. I am glad that now other communities started playing around with BATMAN_V. Our team in Zwickau hacked the status page support a little bit to get some lines on the map. Unfortunately they implemented it in a way that only supports BATMAN_V, The changes can be found in our site repo: https://gitlab.com/FreifunkChemnitz/site-ffc/tree/zwickau-devel/patches

krombel, 2019-05-24:
Are those patches really recommended by you? I e.g. saw [this](https://gitlab.com/FreifunkChemnitz/site-ffc/blob/zwickau-devel/patches/batman-v-wired-tpo.patch) which makes me assume that it is not fully implemented on your side either...

ambassador86, 2019-05-26:
Hey @krombel,
these patches are good enough if you only have one site and all domains share the same routing algorithm. If you have mixed routing algorithms in your domains the patches will break the status page for BATMAN_IV domains.

awlx, 2019-05-26:
Our experimental firmware (http://firmware.ffmuc.net/experimental/) now applies all your patches. It's looking good so far. And as most people in our chat (https://chat.ffmuc.net) confirmed they are fine to break the old setup in favour of the new ... as we want to migrate everything to BATMAN_V anyway ;).

rotanid, 2019-06-06:
if anyone provides a pull request that doesn't break BATMAN_IV we are happy to take it.
otherwise this won't be fixed before 2019.1 release sadly

christf, 2019-08-25:
we could scrap the idea of querying the mesh protocols for their neighbours but query using multicast. This would be mesh protocol agnostic and fairly easy to do. It would also lighten the maintenance burden on the respondd modules.

However it would not mean the same thing as today. Now mesh neighbours are neighbours who are known to the mesh protocol. When implementing the above, neighbours will be known to the link layer. So there is a tradeoff.

closed 2026-09-27 by maurerle