freifunk-gluon/gluon #1727: Enhancement: Display Wifi Channels on Status Page

christf, 2019-05-21:
When working in the field it is helpful if all channels in use by the device are displayed on the status page to ease build local meshes.

Using a map to extract the information may not always be possible depending on the uplink working at that time. Keeping the information de-centrally matches the goals of the project.

rotanid, 2019-05-22:
do you agree, #644 is enough and this is a duplicate?

Adorfer, 2019-05-26:
@rotanid it depends if all available respondd-data displayed on the status page. 
If this is not the case, then it has to be added to the statuspage after it's integrated in the respondd-info.

rotanid, 2019-05-26:
ok, but then this issues is blocked by the other one nevertheless.

CodeFetch, 2019-05-27:
@rotanid Can you add this in #1415 and close this issue?
BTW this is also a part that should be modularized. #1415 is blocked by #1158...

rotanid, 2019-05-27:
thanks, done.

closed 2019-05-27 by rotanid