freifunk-gluon/gluon #1728: BATMAN_V: Throughput on wired interfaces is not measured correctly

krombel, 2019-05-24:
### Bug report

**What is the problem?**
When there are two devices which are connected via e.g. vxlan and wifi the wifi connection is preferred.
batctl always responds "1.0" as speed while wifi has higher values (which is the reason for the preference

**What is the expected behaviour?**
vxlan shows a throughput near to the link capacity. At least sth bigger than "1.0"

**Gluon Version:**
v2018.2.1

**Custom patches:**
@ambassador86 pasted a link on #1726 to some patches from Freifunk Chemnitz which contains a patch which overwrites that value for batman-adv:
https://gitlab.com/FreifunkChemnitz/site-ffc/blob/b89e6568/patches/batman-v-wired-tpo.patch

This makes me assume that this is not an issue for our community (Freifunk München) but for BATMAN_V in general.

rotanid, 2019-06-06:
this looks like an upstream issue, but maybe it's better with batman-adv 2019.2 ? there's a PR for this, maybe you can test that, otherwise try to report it upstream please

mweinelt, 2019-06-06:
If not we can set throughput manually on MoW/MoL interfaces to something reasonable like 100 Mbit/s.

T-X, 2019-07-28:
One issue identified by @NeoRaider seems to be that the vxlan driver currently does not inherit/propagate the throughput of the underlying interface. A patch for the vxlan driver is work in progress (by @NeoRaider).

However I just noticed, that even with vxlan=false in the site.conf the BATMAN_V throughput is not determined correctly:

<pre><code>root@freifunk060464a439b1:/# batctl n
[B.A.T.M.A.N. adv openwrt-2018.1-8, MainIF/MAC: primary0/5e:75:6a:4b:97:ab (bat0/06:04:64:a4:39:b1 BATMAN_V)]
IF             Neighbor              last-seen
a6:73:e9:f5:86:48    0.380s (        1.0) [    br-wan]
f6:9b:e5:70:62:28    0.220s (        1.0) [    br-wan]
</code></pre>

Only using the eth1 (the only interface part of br-wan) directly seems to work:

<pre><code>root@freifunk060464a439b1:/# batctl if del br-wan
root@freifunk060464a439b1:/# brctl delif br-wan eth1
root@freifunk060464a439b1:/# batctl if add eth1
root@freifunk060464a439b1:/# batctl n
[B.A.T.M.A.N. adv openwrt-2018.1-8, MainIF/MAC: primary0/5e:75:6a:4b:97:ab (bat0/06:04:64:a4:39:b1 BATMAN_V)]
IF             Neighbor              last-seen
a6:73:e9:f5:86:48    0.370s (     1000.0) [      eth1]
f6:9b:e5:70:62:28    0.420s (     1000.0) [      eth1]
</code></pre>

So it seems that with bridge interfaces we have a similar issue. The easiest, upstreamable solution for this I can currently think of is submitting a patch for the Linux bridge to inherit/propagate the minimum throughput of all its slave interfaces.

---

Also, current ethtool output:

<pre><code>root@freifunk060464a439b1:/# ethtool br-wan
Settings for br-wan:
        Link detected: yes
root@freifunk060464a439b1:/# ethtool eth1
Settings for eth1:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: umbg
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
</pre></code>

A-Kasper, 2019-07-29:
> So it seems that with bridge interfaces we have a similar issue. The easiest, upstreamable solution for this I can currently think of is submitting a patch for the Linux bridge to inherit/propagate the minimum throughput of all its slave interfaces.

I'm not sure if this would work. What happens, if you've got a 2,4 and a 5 GHz device bridged, and there is no other 5 GHz device in range, or with very slow connection. My first guess would be that this change would result in also bad 2,4 GHz result... 
Also I think that the fastes link would be important.

ambassador86, 2019-07-29:
From our experience in Chemnitz it is not the best idea to create a bridge over wireless interfaces (in the sense of adding this bridge later to the mesh). Bridges usually do not expose any link speed. If an interface is added to the batman mesh the speed will be used if it is exposed by the underlying interface. We almost never get reliable speed information on dual band machines.

NeoRaider, 2019-11-09:
@A-Kasper We never use bridges for wireless devices.

@T-X I think propagating the maximum port speed would make more sense than the minimum. This would match what would happen if the bridge was replaced with a hardware switch: The link speed determined between the device and the switch can be higher than the actual usable throughput, but never lower. I wonder if we could also try to get rid of bridges in our setups by default...

I have a patch for VXLAN link info propagation, which I never submitted upstream. I'll also look into the bridge link info handling. Is there anything else preventing proper link speed handling for wired meshing?

NeoRaider, 2019-11-09:
http://patchwork.ozlabs.org/project/netdev/list/?series=141871

Adorfer, 2019-11-11:
As we have Powerline wired-mesh setups (with largely fluctuating speed over the course of day due to bad phone chargers in th TN), a static link speed evaluation on wired connections should be avoided. As long as there is just one Uplink, it does not matter a lot, but with multiple VPN links in largely various speeds like dsl16M VS Docsis400 it should.

NeoRaider, 2019-11-11:
@Adorfer That is a discussion for batadv development, not Gluon.

In any case, basing a metric on an inaccurate throughput metric is still better than treating all links the same regardless of throughput IMO (@T-X Does BATMAN V still combine this metric with packet loss?)

rotanid, 2026-11-03:
@grische @krombel is this issue still valid or have we solved it in the meantime with something?

grische, 2026-11-03:
No, the problem is still there. 

More details can be found here https://github.com/freifunk-gluon/gluon/pull/3736#discussion_r3082622185

I have a complex script that might solve this, but it needs a lot more thought, as this can only measure the local throughput to the next hop and not to the target:
https://github.com/grische/gluon/commit/b063e35481b28833e89b66063366d14290092f89

T-X also has a few more ideas, but nothing that really solves the problem.