freifunk-gluon/gluon #1730: Support for Linksys WRT3200ACM

smoe, 2019-05-27:
Hello, I just did 
```
--- a/targets/mvebu-cortexa9
+++ b/targets/mvebu-cortexa9
@@ -1,2 +1,5 @@
 device linksys-wrt1200ac linksys-wrt1200ac
 factory .img
+
+device linksys-wrt3200acm linksys-wrt3200acm
+factory .img
```
and 2018.2.1 runs fine on my machine. You may consider to formally add it in analogy to the WRT1200AC (https://github.com/freifunk-gluon/gluon/issues/956). Is there something I shall test for you experts out there? It does not mesh via 11s (https://github.com/kaloz/mwlwifi/issues/127), which obviously would be great. And super-great it would be if MU-MIMO would work, especially since for a few installations I found that the direction of the meshing router is another direction than where most local users are.

CodeFetch, 2019-05-27:
@smoe I've one of those devices, too. The WLAN chipset does not yet support 802.11s and likely never will. As I've read somewhere 802.11s is considered for newer Marvell chipsets only. Thus it is not interesting for Gluon.

smoe, 2019-05-27:
Hi @CodeFetch , thank you for your swift reply. It is enbraced by a "if $(BROKEN)" that did not show in the diff.

The information that 11s and MU-MIMO are not implemented would be great to find (find more easily. should it be stated somewhere) on the OpenWRT pages. I admit to have decided in favour of the machine when I read that this would be a machine all designed to function with the Open Source firmwares (see respective promo sprech on https://www.linksys.com/de/p/P-WRT3200ACM/). Would not have bought it if I had know about the current lack of support by OpenWRT (and by gluon with it).

I have still some hope that MU-MIMO and 802.11s eventually are finding their support in OpenWRT. To have that device shown as $BROKEN seems like a constructive approach towards it. My interest in these technologies is a better support of venues with many visitors / wide-open places. I have yet no exact understanding about how useful the device now is in such environments, even without the MU in MU-MIMO.

CodeFetch, 2019-05-27:
@smoe In my tests MU-MIMO with the ordinary antennas (like WRT3200ACM) did only increase throughput within a range of about 10 meters. I think you need either multiple directional antennas or cross-polarized antennas to make use of it over a larger range. Regarding 802.11s I doubt that it ever gets supported by the 8887 or 8964.  

If Linksys would have been really interested in building a well-OpenWrt-supported router they'd have used ath9/10k chipsets with an ARM CPU. They have donated routers to the OpenWrt community as a thank you in a state where they were very flawed.

CodeFetch, 2019-09-27:
@rotanid Can you close this please? (Wontfix)

smoe, 2019-09-27:
Can close that, too. Anybody with a need for a WRT3200ACM out there?

closed 2019-09-27 by smoe

CodeFetch, 2019-09-27:
@smoe That's an expensive device. I recommend you selling it on eBay.

smoe, 2019-09-27:
Thank you, @CodeFetch, I should probably not give up, yet. Would you have a pointer to an instruction how to get the original firmware back? I had flashed twice, with two different versions of our firmware, so I think the easy way is gone.