freifunk-gluon/gluon #1731: Add user with restricted capabilities

CodeFetch, 2019-05-29:
Due to questions of liability it would be nice to have a user account with restricted sudo capabilities for remote maintenance.

Permissions that come to my mind are:

- Rebooting
- Manual sysupgrade (still checking signature)
- Restarting WLAN
- Diagnostics

Are there any implementational obstacles?

Adorfer, 2019-06-26:
in order to get a more clear picture, since "diagnostics" can mean a lot and certainly we don't want this user to (e.g) tcpdump promisc on the wan_if or the private_wifi (if present): 

What are the devices such a user should not have access? i guess we will run in a rather lengtly collection.... i am undecided if to start from allow all or deny all. 
What should be readable, what even writable?

cawi2001, 2019-06-27:
Hello all,

We don’t want to have a backdoor or a user who is able to log on the running Router. With that Account it’s possible to get into the local Network. It should be a decission of the owner of the Router, if someone is allowed to log on or Not. 

Freifunk is not comercial  , so that we have to take care about all the Router in the Network.

Kind regards 

Carsten

mweinelt, 2019-06-27:
Hey @cawi2001,

I do think that it currently already is the choice of the user to allow or deny remote access, so I don't understand what point you wanted to add to the discussion.

CodeFetch, 2019-06-27:
@cawi2001 Firmware updates are a backdoor, but a user with restricted capabilities can be arbitrarily restricted and thus it is not a backdoor. It's just a key to a door to an apartment while the root user is the janitor and has the master key to all apartments. Linux systems are multi-user systems conceptually and do have the security requirements to do this if they are configured properly.

At the moment many people add their public key to gain access as a root user while they don't really need root access. That's what I'd like to change. It's possible to e.g. deny the user to analyze WAN-traffic, but not when they have root access.

My approach is to generally deny everything and only allow very specific commands.

Adorfer, 2019-06-27:
> only allow very specific commands

to be precises (or to state the obvious): we are talking ab remote shell (dropbear/cli) users, correct?

I feel like a similar discussion about restricted LUCI-logins. but the gluon-status page is out of discussion here? 
<strike>or in other words: what about putting additional features "with login" on the status page? Ah, not possible, not way to have ssl for the moment, so no propper auth method without risking of session hijack.</strike>
https://github.com/openwrt/luci/issues/623

if you look at the dev&proc-tree: What do you suggest?

mweinelt, 2022-03-26:
There has been no activity on this issue in a while and I don't see anyone working to achieve the features mention in the top post.

Closing as stale.

closed 2022-03-26 by mweinelt