freifunk-gluon/gluon #1732: RFC: Switch to usign

CodeFetch, 2019-05-29:
OpenWrt devs are planning to implement sysupgrade signature verification using usign.
We could use it instead of ecdsaverify.

mweinelt, 2019-06-02:
> OpenWrt devs are planning to implement sysupgrade signature verification using usign.

Can you link the source of this information?

CodeFetch, 2019-06-02:
http://lists.infradead.org/pipermail/openwrt-devel/2019-April/016763.html

There was a follow-up discussion, but I can't find it at the moment.

NeoRaider, 2019-06-15:
I don't think OpenWrt's image signatures will be useful for Gluon - we need signed autoupdater manifests that include metainfo like upgrade branch and date. Through the manifests, image authenticity is verified as well.

Also:
- libecdsautil is linked into the autoupdater, which is more efficient than verifying the signatures in a separate process (RAM-/CPU-wise)
- On 4MB devices, we have already dropped usign together with opkg, so we don't waste space for two different signature utilities
- Switching the signature utility without a very good reason would needlessly create work for communities

I'm closing this issue - please reopen if there is more to discuss.

closed 2019-06-15 by NeoRaider

CodeFetch, 2019-06-15:
I thought it was useful for #1731 for allowing a user with restricted permissions to manually do a sysupgrade with a signed image, but your arguments convinced me.