freifunk-gluon/gluon #1734: More automated testing on real hardware or QEMU for gluon

SmithChart, 2019-06-01:
During the Freifunk Meetup during GPN19 [0] we talked about automated testing of gluon on real hardware or QEMU.

I have briefly introduced my CI system. Here are some follow ups on this.
* Blog [1] [2]
* Test repo [3] (Tests are here [3a])
* Hardware is abstracted using labgrid [4]

Currently this setup runs using a TL-WR 841 under my desk and is currently not publicly available. I would happily share my work and knowledge.

In the discussion we agreed that we would like to continue this discussion and that it would be great to have such an infrastructure for gluon development and or other communities.

For the linux kernel there is a distributed testing infrastructure: Kernel-Ci [5].
Maybe we can learn from their system and adapt this to gluon.

This issue is meant to start a discussion on this topic.
I can suggest that we may have a meeting during the Chaos Communication Camp this summer or maybe have a meeting in mumble.

Cheers
Chrissi^

[0]: https://entropia.de/GPN19:Freifunk_Meetup
[1]: https://stratum0.org/blog/posts/2017/08/16/freifunk-ci/
[2]: https://stratum0.org/blog/posts/2018/11/22/freifunk-ci-2/
[3]: https://gitli.stratum0.org/ffbs/ffbs-ci
[3a]: https://gitli.stratum0.org/ffbs/ffbs-ci/blob/master/test_uboot_strategy.py
[4]: https://labgrid.readthedocs.io/
[5]: https://kernelci.org/

rotanid, 2019-06-01:
Hi,
thanks four your ideas and work on this topic!
Regarding meetings i'd like to remind everyone that we agreed last year to not have Gluon meetings at events, as these meetings are not inclusive (time, distance, money).
There's a planned Gluon Developer meeting in mumble in the next weeks.
Regards

mweinelt, 2019-06-07:
@lemoer is currently in process of setting up a ci pipeline using jenkins.

A docker build environment is being prepared in #1738.

CodeFetch, 2019-06-11:
There is also a test suite from Qualcomm somewhere to use real hardware, but I think as Gluon focuses on the software development and not e.g. debugging WiFi drivers, [lemoer's approach](https://github.com/lemoer/pynet) is the right way to start.

mweinelt, 2022-03-26:
I have written a QEmu test driver based on what NixOS was using like a year ago. It does work, but the test harness around it would need to be written. A way to define test scripts, start one test after another.

https://github.com/freifunk-gluon/gluon/blob/qemu-serial-testdriver/tests/driver.py