freifunk-gluon/gluon #1754: gluon-web-wifi-config: libiwinfo does not like phy-names

CodeFetch, 2019-06-17:
In [wifi-config.lua](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-wifi-config/luasrc/lib/gluon/config-mode/model/admin/wifi-config.lua) the txpower_list-function uses the phy-names (e.g. phy0) which it derives from the gluon-util.

Currently libiwinfo only handles the name of the VIF properly. 

Thus the TX-Offset is not being respected at all.

You can see the same behavior by executing `iwinfo phy0 info` which shows "unknown" in nearly all fields compared to e.g. `iwinfo client0 info` which shows the correct HWID, TX power offset etc.

This should be fixed upstream.

Adorfer, 2019-06-29:
i am not sure if i understand the working of this script correct. 
Are the radios initialized to the final country and HT-mode turned on and settled (seconds!) before making the query for available txpowerlist?
If radio is turned off there will probably no valid list. 

and if you like dig deeper into the upstream-problems: Often there are lists to enourmous values (like up to "30" on old NSM2, but on some units with such long lists (WR4900) you may set the parameter to 30, but you never get the "*" in the list beyound 23 (just as an example). 

And off course we have the complex topic of offsets of antenna gains.

in other words: The hole topic of setting txpowers is flawed on many levels that i personally would consider this as "beyound repair" and would drop this function, at least in the Web-based-config mode. 
(Whoever want's to touch the topic should please do so via cli)

rotanid, 2019-06-29:
i would probably agree on removing it from config mode...

CodeFetch, 2019-06-29:
@rotanid Ubiquiti devices sometimes have an external power amplifier chip. The Ubiquiti firmware thus sets an offset depending on the subvendor device-ID. You can find the offsets here: https://git.openwrt.org/?p=project/iwinfo.git;a=blob_plain;f=hardware.txt;hb=HEAD

This means that Ubiquiti devices sometimes send with a far higher transmit power than iw shows.

The subvendor device ID can be read from flash on Ubiquiti devices. This is also how the model detection in OpenWrt works. I've tried to add a patch for adding the missing model declarations to OpenWrt, but it was declined as ar71xx is out-of-life. 
https://github.com/openwrt/openwrt/pull/2144/files

But it is possible, that I add such a script to Gluon. This script could set the antenna gain depending on the model and the TX power offset which it both derives from the subvendor device ID.

My suggestion is: I remove the iwinfo script as long as it does not work and introduce a script for retrieving the antenna gain and power offset from Ubiquiti devices in Gluon.

@Adorfer I just need the measurement as we discussed on the Forum from you. Then I know whether the antenna gain is included in ART. It does not even need to be super-accurate. I already have a script with the correct antenna gains. After that we have full knowledge about the real transmission power of these devices.

NeoRaider, 2020-05-29:
@CodeFetch How do we proceed with this issue? Having some way to determine the antenna gain / power offset more accurately would also help with #94.

CodeFetch, 2020-05-30:
I've a Gluon package somewhere on my desktop which can set the correct power offsets for Ubiquiti devices, but the result of my discussion with some OpenWrt devs was that the correct way is to fix it in mac80211 and introduce a DTS parameter for the antenna gain, PA gain, and something else I forgot about. There is still an out-of-tree patch for iw from Felix Fietkau which needs to be upstreamed and was blocked, because nobody from the regdb team felt responsible. https://patchwork.kernel.org/patch/2854533/