freifunk-gluon/gluon #1756: deprecated messages of batman-adv

bobcanthelpyou, 2019-06-17:
Deprecated messages of batman_adv via `dmesg` with the latest Gluon master.
```
[   10.164627] batman_adv: B.A.T.M.A.N. advanced openwrt-2019.2-1 (compatibility version 15) loaded
[   19.781858] batman_adv: [Deprecated]: sh (pid 1201) Use of sysfs file "mesh_iface".
[   19.781858] Use batadv genl family instead
[   22.481837] batman_adv: [Deprecated]: sh (pid 1660) Use of sysfs file "gw_mode".
[   22.481837] Use batadv genl family instead
[   22.634668] batman_adv: [Deprecated]: sh (pid 1660) Use of sysfs file "hop_penalty".
[   22.634668] Use batadv genl family instead
[   22.780563] batman_adv: [Deprecated]: sh (pid 1660) Use of sysfs file "multicast_mode".
[   22.780563] Use batadv genl family instead
[   22.893830] batman_adv: [Deprecated]: sh (pid 1660) Use of sysfs file "orig_interval".
[   22.893830] Use batadv genl family instead
[   26.258930] batman_adv: [Deprecated]: sh (pid 1989) Use of sysfs file "mesh_iface".
[   26.258930] Use batadv genl family instead
```

Might be introduced by https://github.com/open-mesh-mirror/batman-adv/commit/5bbd4721bfda46839addc9089135450d0e35f459

https://github.com/open-mesh-mirror/batman-adv/blob/master/Documentation/ABI/obsolete/sysfs-class-net-batman-adv
https://github.com/open-mesh-mirror/batman-adv/blob/master/Documentation/ABI/obsolete/sysfs-class-net-mesh

mweinelt, 2019-06-17:
This will probably be resolved through https://github.com/freifunk-gluon/gluon/pull/1752.

bobcanthelpyou, 2019-06-18:
Upgraded to https://github.com/freifunk-gluon/gluon/commit/3d3617ed2c85c24be1b8c2de876045381d9adf0e (v2018.2-138-g3d3617ed)

```
[   22.724641] batman_adv: B.A.T.M.A.N. advanced openwrt-2019.2-1 (compatibility version 15) loaded
[   34.145626] batman_adv: [Deprecated]: sh (pid 1377) Use of sysfs file "mesh_iface".
[   34.145626] Use batadv genl family instead
[   37.501006] batman_adv: [Deprecated]: sh (pid 1847) Use of sysfs file "gw_mode".
[   37.501006] Use batadv genl family instead
[   37.688045] batman_adv: [Deprecated]: sh (pid 1847) Use of sysfs file "hop_penalty".
[   37.688045] Use batadv genl family instead
[   37.789215] batman_adv: [Deprecated]: sh (pid 1847) Use of sysfs file "multicast_mode".
[   37.789215] Use batadv genl family instead
[   37.814747] batman_adv: [Deprecated]: sh (pid 1847) Use of sysfs file "orig_interval".
[   37.814747] Use batadv genl family instead
[   41.564276] batman_adv: [Deprecated]: sh (pid 2201) Use of sysfs file "mesh_iface".
[   41.564276] Use batadv genl family instead
```

mweinelt, 2019-06-18:
Yeah, on second thought … it stills needs to talk to batadv-legacy, so it will still use the sysfs interface for that.

Adorfer, 2019-06-18:
Do you have reference how to update our scripts/packages currenty working on the note, looking for hop-pentalty etc?

bobcanthelpyou, 2019-06-18:
Does the documentation update help? https://github.com/open-mesh-mirror/batman-adv/commit/82d480bbd07911199bc7c2a5d0c0b17707b2bb0e

`grep -r "/sys/class/net" *|grep batman_adv` might make some of the obsolete usage "visible"

Not sure how to handle these as there is batman-adv-legacy, too.

mweinelt, 2019-06-18:
At some point batctl will move everything to netlink and then it won't be usable with batadv-legacy unless some work is done to make it so. There is really nothing to worry about with these messages for Gluon users at this time.

Adorfer, 2019-06-19:
these messages in syslog are really nasty (2 lines per call relevant batctl command). 
This spams the syslog and makes (perhaps) important/relevant syslog-messages expire faster than neccesary. 

(here > 90% of the syslog are like the following.)

```
Wed Jun 19 17:30:01 2019 kern.warn kernel: [66461.808013] batman_adv: [Deprecated]: batctl (pid 11862) Use of sysfs file "iface_status".
Wed Jun 19 17:30:01 2019 kern.warn kernel: [66461.808013] Use batadv genl family instead

```

NeoRaider, 2019-06-19:
As long as the new batctl still supports the old sysfs files as a fallback, switching shouldn't be an issue.

NeoRaider, 2019-06-19:
@Adorfer How many of these messages are left after the merge of #1752? batctl should not cause any such warning anymore.

mweinelt, 2019-06-19:
```
root@64289-rocket-fuel:~# logread | grep -A1 Deprecated
Tue Jun 18 16:08:20 2019 kern.warn kernel: [   22.503074] batman_adv: [Deprecated]: sh (pid 2310) Use of sysfs file "mesh_iface".
Tue Jun 18 16:08:20 2019 kern.warn kernel: [   22.503074] Use batadv genl family instead
--
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.633877] batman_adv: [Deprecated]: sh (pid 2559) Use of sysfs file "gw_mode".
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.633877] Use batadv genl family instead
--
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.839075] batman_adv: [Deprecated]: sh (pid 2559) Use of sysfs file "hop_penalty".
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.839075] Use batadv genl family instead
--
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.994441] batman_adv: [Deprecated]: sh (pid 2559) Use of sysfs file "multicast_mode".
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.994441] Use batadv genl family instead
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.994588] batman_adv: [Deprecated]: sh (pid 2559) Use of sysfs file "orig_interval".
Tue Jun 18 16:08:22 2019 kern.warn kernel: [   23.994588] Use batadv genl family instead
--
Tue Jun 18 16:08:23 2019 kern.warn kernel: [   24.758153] batman_adv: [Deprecated]: sh (pid 2765) Use of sysfs file "mesh_iface".
Tue Jun 18 16:08:23 2019 kern.warn kernel: [   24.758153] Use batadv genl family instead
--
Tue Jun 18 16:08:25 2019 kern.warn kernel: [   26.513364] batman_adv: [Deprecated]: sh (pid 3150) Use of sysfs file "mesh_iface".
Tue Jun 18 16:08:25 2019 kern.warn kernel: [   26.513364] Use batadv genl family instead
Tue Jun 18 16:08:25 2019 kern.warn kernel: [   26.526097] batman_adv: [Deprecated]: sh (pid 3150) Use of sysfs file "mesh_iface".
Tue Jun 18 16:08:25 2019 kern.warn kernel: [   26.526097] Use batadv genl family instead
root@64289-rocket-fuel:~# batctl -v
batctl openwrt-2019.2-0 [batman-adv: openwrt-2019.2-1]
```

NeoRaider, 2019-06-19:
I'm considering just reverting to batadv 2019.1, because we broke our (unwritten, forgotten) policy of not backporting such updates in Gluon without a good reason.

rotanid, 2019-06-19:
> I'm considering just reverting to batadv 2019.1, because we broke our (unwritten, forgotten) policy of not backporting such updates in Gluon without a good reason.

ACK

Adorfer, 2019-06-19:
@NeoRaider 
 >  How many of these messages are left after the merge of #1752?

still a lot!

[image]

mweinelt, 2019-06-19:
> I'm considering just reverting to batadv 2019.1, because we broke our (unwritten, forgotten) policy of not backporting such updates in Gluon without a good reason.

If this is an unwritten and forgotten rule please write it down somewhere.

Regarding the kernlog messages, they don't bother me and shouldn't bother anyone else, really.

Adorfer, 2019-06-20:
it bothers me that the logread-buffer is limited. and if it's flooded with >80% batman-adv-kernelwarnings i do lose time horizon (due to FIFO) when logging into a node in order to assess a potential situation.

commit c1a77339 (2019-06-20): Revert "batman-adv: update to current openwrt-routing / batman-adv v2019.2"

This reverts commit 8dd13cbbca649967137f6eb392267dec444dfc4e.

Fixes #1756
closed 2019-06-20 by NeoRaider