freifunk-gluon/gluon #1757: tracking bugfix release v2018.2.2

rotanid, 2019-06-18:
only few things are missing before we can tag a new bugfix release v2018.2.2
this ticket keeps track of them

- release notes
- packages bump to 524ae77e9bb20002252af19b63e6607155b3af4d
- OpenWrt 18.06 bump as soon as kernel 4.9.182 is upstream
- #1736

belzebub40k, 2019-06-19:
FTR: kernel 4.9.182 is broken for brcm2708 targets. Could be fixed by reverting commit [staging: vc04_services: prevent integer overflow in create_pagelist()](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/staging/vc04_services/interface/vchiq_arm?h=linux-4.9.y&id=cf07331c8827c9e9e0b4274c9b60204c18592241)

https://github.com/freifunk-mwu/gluon/commit/75dfe8870092420e547ee556d2e35f5cdf563384

mweinelt, 2019-06-19:
The revert is also queued for the next v4.9 stable release, so yes we could temporarily use a downstream revert to get this going.

https://git.kernel.org/pub/scm/linux/kernel/git/stable/stable-queue.git/commit/?id=44aa7eeb48d304381c6145a44d660fd3761a2033

NeoRaider, 2019-06-20:
I took care of that revert.

2tata, 2019-06-23:
I recognize that wifi meshing functionality between gluon-v2017.1.8-7 (e968a225be0efafbd2d4638d0bd530e2d4e841f3) and gluon-v2018.2.1-13 (eed810aac1b0f6795622907b0de7dbc0fbfadc9d) seems to be broken.

the node [TataWilh210-SO](https://map.ffnw.de/#!/en/map/98ded0889a7c) was able to see [ffnw-galaxy2](https://map.ffnw.de/#!/en/map/10feed568d22) with nearly 80-90% but now its lower that 10% same for [TataWilh510-NW](https://map.ffnw.de/#!/en/map/98ded0727ce8) to [OldenburgRaiffeisenLappan5](https://map.ffnw.de/#!/en/map/802aa86e503e) but before the link wars almost 60-70% and now its gone.

A more extreme case both [OldenburgRaiffeisenMosle5](https://map.ffnw.de/#!/en/map/802aa834090f) and [OldenburgRaiffeisenLappan5](https://map.ffnw.de/#!/en/map/802aa86e503e) were able to see [Oldenburg-Lappan-kismet](https://map.ffnw.de/#!/en/map/802aa8d5ceb5) with almost 100% bidirectional. Now its 96% - 0% the zero (ore 1%) in one directory is striking. Shall I open an issue for that or is there one which fits to this anomaly?

Adorfer, 2019-06-23:
@2tata: what is the link to "bugfix-release 18.2.2"?
I didn't check all mentioned nodes, but the ones  guess this is more the "CPE210 attenuator/amp discussion"
https://github.com/freifunk-gluon/gluon/issues/796 (where it's still unclear to me which of the both options is better or which is less bad)
plus "basic_rate"/"supported" rate considerations https://github.com/freifunk-gluon/gluon/pull/1540 where we try to drop support for rates below 12000 in order to safe aritime, even if this results in loss of links (which had been not really usable anyways.)

2tata, 2019-06-23:
> @2tata: what is the link to "bugfix-release 18.2.2"?

It seems something buggy in v2018.2.x which does not appear before.

> I didn't check all mentioned nodes, but the ones guess this is more the "CPE210 attenuator/amp discussion"
> #796 (where it's still unclear to me which of the both options is better or which is less bad)

Ok, but this affected ubiquity devices as well  like:

> A more extreme case both [OldenburgRaiffeisenMosle5(gluon-v2018.2.1-13)](https://map.ffnw.de/#!/en/map/802aa834090f) and [OldenburgRaiffeisenLappan5(gluon-v2018.2.1-13)](https://map.ffnw.de/#!/en/map/802aa86e503e) were able to see [Oldenburg-Lappan-kismet(gluon-v2017.1.8-7)](https://map.ffnw.de/#!/en/map/802aa8d5ceb5) with almost 100% bidirectional. Now its 96% - 0% the zero (or 1%) in one directory is striking. Shall I open an issue for that or is there one which fits to this anomaly?

> plus "basic_rate"/"supported" rate considerations #1540 where we try to drop support for rates below 12000 in order to safe aritime, even if this results in loss of links (which had been not really usable anyways.)

ok, but this not is not really related to my described anomaly isn't it?

Ok, I'll open an issue for that.

bobcanthelpyou, 2019-06-24:
Did a `git diff v2018.2.x..master docs` and it might be useful to add the following commits to the docs of the v2018.2.x branch (if i didn't forgot anything else):

deb020d3ef850c93549c7036ece5e3039339c2f3
ff7aeafad2a5bcae1b965e86cfb128dd24a78daa
cae79eb07426b2a9f87b4d0a4d5a07ecebfd6558
170c494f409df37459d9aa9d5af8bc675b7f9d4a
e8b37b2d7c80f5fbc81efb98753fbf0ce5104d16
85494380632f54676ab95b21f1366c45a6f8b487

by the way de82e58cf202b8cd77f5ce181d7adf8c9a9b176a is missing in master

mweinelt, 2019-06-24:
Thankss everyone, I just backported the docs changes and released v2018.2.2.

https://gluon.readthedocs.io/en/v2018.2.x/releases/v2018.2.2.html

closed 2019-06-24 by mweinelt