freifunk-gluon/gluon #1758: FR: make respondd-listenaddress configurable by site.conf

Adorfer, 2019-06-18:
According to https://github.com/freifunk-gluon/gluon/pull/1030
current master (2019.x candidate) changed respondd listen address from ff02 to additionally(?) ff02::1/2

- i understand that this change is needed for babel networks
- during testing in a network with nodes between 2016.2.x, 2018.2.x and master we did not find a single adress which lets all nodes reply
- for current existing batman-adv-networks this may be a breaking change
- in existing networks there are nodes with "autoupdater turned off" and users not reachable/responding to "please update your node" requests. 
- sending requests to multple address is possible, but may burden additional background traffic for the network
- for now some nodes "without vpn" behind (vpn uplink) of older gluon nodes seem not reply to the hopglass-server at all (my be hopglas issue, has to be investigated)

my suggestion: 
please make the listener-address/mcastgroup for respondd configurable by site.conf

belzebub40k, 2019-06-18:
Simply join both addresses ff05::2:1001 and ff02::2:1001. This is what we are doing to scrape both old and new nodes.

If your using Yanic as Meshviewer Backend this is pretty easy.

````
[[respondd.interfaces]]
ifname = "br0"
multicast_address = "ff05::2:1001"
[[respondd.interfaces]]
ifname = "br0"
multicast_address = "ff02::2:1001"
````

Adorfer, 2019-06-18:
@belzebub40k  thank you for stating the obvious. 
I assume that you did not read my point 4 ("background traffic") and 5 ("new nodes behind old nodes").

belzebub40k, 2019-06-18:
Sorry, missed that. No need to be rude.

Adorfer, 2019-06-18:
"change the setup of the mapserver" is not a valid answer to "please make the address configurable".

belzebub40k, 2019-06-18:
True but this is what was requested in the v2017.1 release notes.

> respondd now listens on ff05::2:1001 in addition to ff02::2:1001 for mesh-wide operation (#984)
> 
> Eventually, ff02::2:1001 will be available for exchanging information between neighbouring nodes only; map servers should be moved to ff05::2:1001.

https://gluon.readthedocs.io/en/latest/releases/v2017.1.html#internals

Adorfer, 2019-06-18:
and therefore i ask to make it configurable, since point 2 in my request:
We are running a mapserver for more than a dozend of different L2-domains of different communities. And there are some communities where "by default" autoupdater is (or was) off and they will stick with their 2016.2.x  nodes foreever. (luckly they are on 11s already, but that's a different issue). 

The point is: We are not living in an ideal world, i simply ask to make this configurable. 

What i can do in my "own domain" on my own mapserver: i stated in the initial posting that am quite aware about that. I am just asking to make it configurable. 

But if we agree that "those networks without old nodes shall suffer": agree, but then say it this clear. and do not try to explain things to me already know (see initial statement).

belzebub40k, 2019-06-18:
I only tried to provide a quick solution as it took my some time to fix our map server. I wasn't saying your request is not valid and I would leave that decision to the gluon maintainers.

Adorfer, 2019-06-18:
o.k understood. Thank you for explaining to everybody in verbose what i wrote on point 4 in the initial statement.

petabyteboy, 2019-06-18:
I could not make ff05::2 work with hopglass server. I suspent this is because it is not a link-local address and additional setup is required?

mweinelt, 2019-06-18:
You'll probably need to bind your socket to the link local address of the interface you want send the packet out of.

NeoRaider, 2019-06-18:
We removed the deprecated address to fix #1701, and for that reason, we won't allow adding it back.

Regarding hopglass, it might have the same issue that gluon-neighbour-info had before a5614a5019db4625afc2211e0271e51eb03c3cd3 - it is necessary to set IPV6_MULTICAST_IF using setsockopt for non-link-local multicast groups. Please open an issue against hopclass if you suspect this is the case.

closed 2019-06-18 by NeoRaider

T-X, 2019-06-18:
Another option to select the outgoing interface for IPv6 multicast is to set a route like:

``$ ip -6 route add ff05::2/128 dev <out-iface> table local``

At least that's the first issue I usually come across with routable IPv6 multicast and multiple network interfaces and when the default route is not where I want the IPv6 multicast to go out on.

(but hopglass probably still needs to set a hop-limit/ttl > 1 via setsockopt() for routed/babel networks)

Adorfer, 2019-06-18:
it's pitty that this will kick all 2016.x nodes from the map, until they update or i send requests to both addresses systematically. 
(l2-bridges/routers of different domains in the same area as described in #1701 should be avoided anyway). So my request would still be to keep it configurarable in respondd, but if it's not possible, i will just patch it locally as long as there are 2016.x nodes in the network.