freifunk-gluon/gluon #1766: FritzBox 4040 bootloader image broken [regression]

RalfJung, 2019-06-24:
### Bug report

**What is the problem?**

A user is reporting the following problem:

> Nach dem flash nach der einschlägigien Anleitung https://fritz-tools.readthedocs.io/de/latest/ quitiert die Fitzbox mit einem wiederholten doppelten blinken der Info LED in rot ihren Dienst.
> Eine IP um ein den Config Mode zukommen vergibt sie auch nicht. 

Translation: trying to boot the FritzBox with the Gluon bootloader fails, the info LED just double-blinks in red.

The same problem has also been reported in another community at https://forum.freifunk.net/t/installation-der-firmware-auf-fritzbox-4040-schlaegt-fehl/20624.

**What is the expected behaviour?**

Booting the FritzBox with the image should work.

**Gluon Version:**

2018.2.1

The bootloader from 2018.2 works fine, so as a work-around our users are now installing a previous firmware version and then doing a sysupgrade.

**Site Configuration:**

https://git.hacksaar.de/FreifunkSaar/gluon-site/blob/6b83c15caaabfe1159ea912882d05696681e5a32/site.conf

**Custom patches:**

None

mweinelt, 2019-06-24:
Thanks for your report.

Can you provide a serial log of the flash procedure? The theory is, that if we see a Red LED flashing, we will likely see a related error message in the serial log.

RalfJung, 2019-06-24:
I don't even own the device, sorry. I just forwarded this on behalf of a user in our community, and because I saw similar a report from another community.

belzebub40k, 2019-06-24:
Between v2018.2 and v2018.2.1 there was one commit for uboot-4040.

> [91d3b87353](https://github.com/openwrt/openwrt/commit/91d3b87353) uboot-fritz4040: fix crash caused by interaction with gcc 7.1+

mweinelt, 2019-06-24:
I've just successfully tested the flash procedure for a firmware based on eed810aac1b0f6795622907b0de7dbc0fbfadc9d (v2018.2.1-7-geed810aa
)

```
Eva_AVM >.............................[tcp_check_ack] <wrong ack> 2049 0
.[tcp_check_ack] <wrong ack> 2251 0
................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Eva_AVM >
flash ..........................................................................<Reboot Device>
```

The Info-LED is continously *red* while the flash procedure is running.

mweinelt, 2019-06-24:
Master build based on 7d386a66f36c947f1c0770f64ab71f22cac1b21a (v2018.2-157-g7d386a66).

https://firmware.darmstadt.freifunk.net/images/1.5~20190621/other/gluon-ffda-1.5~20190621-avm-fritz-box-4040-bootloader.bin

```
Eva_AVM >[autodetect_Handler] urlader Version error
[autodetect_Handler] urlader Version error
..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Eva_AVM >
flash ..........................................................................<Reboot Device>
```

I can't reproduce this with recent Gluon versions.

mweinelt, 2019-06-24:
The FRITZ!Box 4040 I'm testing with advertises it's bootloader as
```
(AVM) EVA Revision: 1.3243
(C) Copyright 2005 AVM Date: Dec 13 2016 Time: 16:13:32 (0) 3 0x0-0x240D
```

mweinelt, 2019-06-24:
```
[FLASH:] MACRONIX Uniform-Flash 32MB 256 Bytes WriteBuffer
[FLASH:](Eraseregion [0] 512 sectors a 64kB) 
[SYSTEM:] CortexA9 
<create jffs2 from 0x1F00000 len 0x100000 jffs2_size 16>
Eva_AVM >
<ERROR: FIRMWARE_ILLEGAL_CONFIG>
```
https://mgmt.saar.freifunk.net/firmware/experimental/other/gluon-ffsaar-1.7.3-avm-fritz-box-4040-bootloader.bin

RalfJung, 2019-06-24:
Ah right, I could have thought earlier of pointing you to our firmware images. ;)

[Here is the diff](https://git.hacksaar.de/FreifunkSaar/gluon-site/compare/1.7.2...1.7.3) of our site config and build script between the two versions. We did change the build invocations a bit, but the change looks all right to me and the fact that another community also has this problem makes me believe that's not it.

I will hopefully soon build a version based on Gluon 2018.2.2, which might be closer to the v2018.2.1-7-geed810aa you were testing.

RalfJung, 2019-06-25:
We now have a firmware based on Gluon 2018.2.2, the FritzBox bootloader is at https://mgmt.saar.freifunk.net/firmware/1.7.4~rc1/other/gluon-ffsaar-1.7.4~rc1-avm-fritz-box-4040-bootloader.bin.

mweinelt, 2019-06-25:
> We now have a firmware based on Gluon 2018.2.2, the FritzBox bootloader is at https://mgmt.saar.freifunk.net/firmware/1.7.4~rc1/other/gluon-ffsaar-1.7.4~rc1-avm-fritz-box-4040-bootloader.bin.

```
[FLASH:] MACRONIX Uniform-Flash 32MB 256 Bytes WriteBuffer
[FLASH:](Eraseregion [0] 512 sectors a 64kB) 
[SYSTEM:] CortexA9 
<create jffs2 from 0x1F00000 len 0x100000 jffs2_size 16>
Eva_AVM >
<ERROR: FIRMWARE_ILLEGAL_CONFIG>
```

RalfJung, 2019-06-25:
Well that's strange. None of the commits you mentioned you built yourself exactly coincides with the ones used to build these FW, but you got one on the 2018.2 branch that is between 2018.1.2 and 2018.2.2...

mweinelt, 2019-06-25:
So what we built for our latest stable firmware was actually this https://github.com/blocktrron/gluon/commits/ffda-1.6.1.

I just tested last nights master build and that works. Going to test v2018.2.2 next.

RalfJung, 2019-06-25:
Maybe it's something about the build environment? Compiler versions and so on? We are using the following docker file to generate ours:
```
FROM debian:stretch

RUN apt-get update
RUN apt-get install -y build-essential git subversion python2.7 gawk unzip libncurses5-dev zlib1g-dev libssl-dev wget cmake pkg-config curl ca-certificates bsdmainutils
RUN cd /tmp && wget https://git.universe-factory.net/libuecc/snapshot/libuecc-6.zip && unzip libuecc-6.zip && cd libuecc-6 && mkdir build && cd build && cmake -D CMAKE_BUILD_TYPE=RELEASE .. && make && make install && cd /tmp && rm -rf libuecc-6.zip libuecc-6 && ldconfig
RUN cd /tmp && wget https://github.com/tcatm/ecdsautils/archive/ab4eda463b4cdbb58136cec171a686fd11694c4e.zip && unzip ab4eda463b4cdbb58136cec171a686fd11694c4e.zip && cd ecdsautils-ab4eda463b4cdbb58136cec171a686fd11694c4e && mkdir build && cd build && cmake -D CMAKE_BUILD_TYPE=RELEASE .. && make && make install && cd /tmp && rm -rf ab4eda463b4cdbb58136cec171a686fd11694c4e.zip ecdsautils-ab4eda463b4cdbb58136cec171a686fd11694c4e

RUN useradd ci
USER ci
```
It's been a while since we last re-generated it though, we are using versions of everything from 2018-04-27. But I doubt Debian stable has moved that much since then.

mweinelt, 2019-06-25:
I'm currently compiling our images on a Debian Buster machine. Maybe try the Dockerfile given in `contrib/Dockerfile` instead?

RalfJung, 2019-06-25:
I'd rather not build on a not-yet-stable distribution. But thanks for pointing out that Dockerfile, we might switch to it once buster is released!

mweinelt, 2019-06-25:
Buster is already in full-freeze, so I wouldn't worry too much.

 https://release.debian.org/buster/freeze_policy.html

RalfJung, 2019-06-25:
I'm aware of the Debian freeze process, but still I prefer to wait for a release.

RalfJung, 2019-06-25:
So I guess the working hypothesis is that something between Gluon 2018.2.1 and 2018.2.2 introduced a regression only when using GCC 7 (or some other part of the toolchain that changed between stretch and buster)?

mweinelt, 2019-06-25:
I pulled in this change at the end of january 2019 when updating the OpenWrt reference.
```
91d3b87353 uboot-fritz4040: fix crash caused by interaction with gcc 7.1+
```

RalfJung, 2019-06-25:
Actually stretch is on GCC 6.3. So maybe that patch broke older compilers?

mweinelt, 2019-06-25:
Just built v2018.2.2 using the Gluon Docker container and the resulting image works.

rotanid, 2019-06-25:
next up: someone tries the same dockerfile but with stretch ;)

mweinelt, 2019-06-26:
Yes, I can confirm that building on Debian Stretch with GCC 6.3 results in `<ERROR: FIRMWARE_ILLEGAL_CONFIG>`.

mweinelt, 2019-06-26:
If someone else wants to reproduce this, I used this environment:

```docker
FROM debian:stretch-slim

RUN apt update && apt install -y --no-install-recommends \
    ca-certificates \
    file \
    git \
    subversion \
    python \
    build-essential \
    gawk \
    unzip \
    libncurses5-dev \
    zlib1g-dev \
    libssl-dev \
    libelf-dev \
    wget \
    time \
    lua-check \
  && rm -rf /var/lib/apt/lists/*

RUN useradd -d /gluon gluon
USER gluon

VOLUME /gluon
WORKDIR /gluon
```

Adorfer, 2019-06-26:
just for my understanding (and perhaps until it's fixed here or upstream):
which versions of gcc are now considering as OK for building FB4040-bootloader?

kevin-olbrich, 2019-06-26:
Maybe OT:
For my understanding: Until I saw this issue, I always thought that OpenWRT builds it's toolchain for every target - for example:

[image]

This issue will lead to a lot of issues because stretch uses GCC 6.3 while Fedora 30 is at v9.1.

If the system GCC is used, whats the reason for building GCC then? Compiling the compiler?

NeoRaider, 2019-06-26:
The system GCC can't directly affect the bootloader, as the bootloader is built using the target GCC. The system GCC could affect the build in two ways:
- Miscompilation of target GCC
- Miscompilation of host utilities (for example firmware image build tools that run during the build)

rotanid, 2019-06-28:
i can't reproduce this issue, it's working for us.
after having no problems with our regular builds on Debian Stretch, i took an empty folder and built completely from scratch and without any patches, only the Gluon tag v2018.2.2.
Even with this build there was no issue with flashing it to a fresh 4040, it worked on the first try.

MPW1412, 2019-07-04:
Our broken images here at Freifunk Münsterland where build on a Hetzner Cloud machine (CX51). So the OS was Ubuntu 18.04. The server was deleted after the build was finished unfortunately.

The confirmed working images were build on Ubuntu 16.04 though.

rotanid, 2019-07-04:
So someone needs to do a new build completely from scratch and without custom patches on another hosting platform than Hetzner but with Ubuntu 18.04.
With full debug logging like described in https://github.com/freifunk-gluon/gluon/wiki/Troubleshooting so we can search the logs.

rotanid, 2019-07-04:
my own build of v2018.2.2 on Ubuntu 18.04 (docker image) works. so it's not only the OS.

@NeoRaider suggested on IRC that there is something wrong with the padding in some cases because of issues with https://github.com/chunkeey/FritzBox-4040-UBOOT/blob/master/fritz/fritzcreator.sh

RalfJung, 2019-07-04:
Our confirmed working and confirmed broken images were both built in the same Debian Stretch Docker container (linked above). The only difference is the Gluon version.

I can do another build of that and provide logs if that's helpful, but it would not be Ubuntu.

rotanid, 2019-07-04:
if you can reproduce broken images, that's great.
let's wait with further testing for an updated fritzcreator

RalfJung, 2019-07-04:
Our 1.7.3 and 1.7.4~rc1 were both broken, so it looks fairly reproducible. Someone could test [1.7.4](https://mgmt.saar.freifunk.net/firmware/1.7.4/) to confirm.

NeoRaider, 2019-07-20:
I've pushed a potential fix to the branch `wip/fritz4040`, please test.

rotanid, 2019-08-24:
@RalfJung your builds 1.7.4~rc2 and 1.7.4 work for me. 1.7.3 doesn't. 1.7.2 works. can't find 1.7.2~rc1 anymore.
have you tested neoraiders branch or what have you done between 1.7.4~rc1 and 1.7.4~rc2? any changes to the build environment, software or hardware?

RalfJung, 2019-08-24:
> your builds 1.7.4~rc2 and 1.7.4 work for me. 1.7.3 doesn't.

That would be [this diff](https://git.hacksaar.de/FreifunkSaar/gluon-site/compare/1.7.3...1.7.4). Maybe 2018.2.2 helped?

> can't find 1.7.2~rc1 anymore.

There are more firmwares at https://archive.saar.freifunk.net/firmware/

rotanid, 2019-08-24:
@RalfJung 
> have you tested neoraiders branch or what have you done between `1.7.4~rc1` and `1.7.4~rc2`? any changes to the build environment, software or hardware?
so no changes to the build environment, only code?

RalfJung, 2019-08-25:
Those were the only changes, yes.

rotanid, 2019-08-31:
> I've pushed a potential fix to the branch `wip/fritz4040`, please test.

ping @NeoRaider :)
as discusses a while ago on IRC, there were no problems with images built on your branch

SmithChart, 2019-09-04:
> I've pushed a potential fix to the branch `wip/fritz4040`, please test.

We had a community-member at our meeting today who wanted to use a Fritzbox 4040 with our (current) v2018.2.2 firmware. We also had the "blinking-red-LED"-issue.

I have tested @NeoRaider 's Patch 7e39df6aabe0d807c44ef9e04ee030ff9b001ff0 on top of our current tree. (Our current state is here: https://gitli.stratum0.org/ffbs/ffbs-gluon/commits/v2018.2.2.1-ffbs )

With this patch we were able to successfully flash and configure the router.

commit fdf552e5 (2019-09-04): uboot-fritz4040: add fixes for bootloader image generation

This seems to unbreak bootloader image generation with certain shells /
build setups.

Fixes #1766
commit ad8a51db (2019-09-04): uboot-fritz4040: add fixes for bootloader image generation

This seems to unbreak bootloader image generation with certain shells /
build setups.

Fixes #1766

(cherry picked from commit fdf552e55ead4424172a9f7b691e5c8bb19ad877)
closed 2019-09-04 by NeoRaider