freifunk-gluon/gluon #1767: Support primary MAC address selection in OpenWrt

adrianschmutzler, 2019-06-24:
I currently work on a Pull Request for OpenWrt which aims at storing the devices' primary MAC addresses (the "sticker" address) in the device tree:
https://github.com/openwrt/openwrt/pull/2159

If this is merged, retrieving the primary MAC address will be as simple as calling `get_mac_primary`.

I currently have included all devices there which I know the correct MAC address myself. These are:
ath79/ar71xx: 841v8 to v12, 842v2, 1043v1 to v5, Ubiquiti XM Outdoor, Ubiquiti AC Mesh/Lite, WDR3600/4300, CPE210v1 to v3, WA901NDv2, 740v2, 741v2/v4, MR3020v1, Archer C7 v2/v4/v5, C25v1, C60v1/v2, GL-AR150
ipq40xx: FritzBox 4040
mpc85xx: WDR4900v1
ipq806x: C2600

Obviously, to be successful this needs a lot of other devices taken care of. I would be glad if someone here could provide me with data for additional devices.
This could be as easy as stating
"on Archer C58v1 the sticker MAC address is found on eth0, while eth1 is +1"
(arbitrary data for this example, C58v1 is actually needed, too)

If you specify an ethernet device or wireless device like this, please be sure to include the MAC address increment (+1/-1/0) of the other one. This will make it easier to identify the correct one in the DTS files. For Wifi, please specify whether it is 2.4 GHz or 5 GHz, not only phyX.

Note that this will only work for DTS-based targets, so gluon will only benefit from it after it switched to ath79.
However, for helping me with sticker addresses it is not relevant whether the device is ar71xx, ath79, something different or not even gluon-based.

Adorfer, 2019-06-28:
is it ok if can state that "model x  gives after 'sysupgrade -n' a nodeid/nicename corresponding to the sticker-mac"?
if yes then i can state: 

D-Link DIR505: OK (MAC not on product, only on giftbox)
Netgear R6120: OK
TP-Link ArcherC25v1: fixed by #1771 
TP-Link ArcherC5v1: OK
TP-Link ArcherC5v4: OK
TP-Link CPE210v1.0: OK
TP-Link CPE210v1.1: OK
TP-Link MR3040v1: OK
TP-Link WDR3600v1: OK
TP-Link WDR4911v1: OK
TP-Link WR1043NDv1: OK
TP-Link WR841v10: OK
TP-Link WR841v9: OK
TP-Link WR941v5(CN)/v4(EU): NOK? (NonTL bootloader)
UBNT UniFi-AC-MESH: OK
UBNT-NSM2 (SWX-M2N): OK
UBNT-NSML2 (SWX-M2L): OK
UBNT-PicostatioinM2: ? (Sticker in removable cover, no match)

adrianschmutzler, 2019-07-01:
@Adorfer Thanks for checking this.

From @CodeFetch I was informed about the Gluon script handling primary MAC address assignment in Gluon:
https://github.com/openwrt/openwrt/pull/2159#issuecomment-505044231
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/luasrc/lib/gluon/upgrade/010-primary-mac

I now just assumed that those are correct and transferred all of them to my OpenWrt pull request.

Thus, despite some inconsistencies for about 5 devices I will have to investigate, individual lookup is only relevant if you really find devices that are set up incorrectly in Gluon or you can add devices with OpenWrt not part of Gluon master yet.
Consequently, it would be very interesting whether 941v4/v5 is really wrong in Gluon.

For ubnt-xm (Picostation), I have access to devices myself and can verify that this is correctly set.

adrianschmutzler, 2019-07-01:
> TP-Link WDR4911v1: OK

WDR4900v1 ?

CodeFetch, 2019-07-01:
@adrianschmutzler  TP-Link WR941v5(CN) is a Chinese version of the WR841N v8 as far as I know. @Adorfer Have you checked the  WR941ND v4 or did you assume the WR941v5(CN) is a clone of this?

Adorfer, 2019-07-01:
please do not care about that 941cn-devices. ig got this from a bulk of upcycled to 16/64MB with modified bootloader (breed)

adrianschmutzler, 2019-07-02:
If someone owns any of listed devices, I would be interested in the output of the following command:
```find /proc/device-tree/ -name "*mac-address*"```

_No patching necessary, just with current master or stable (please tell OpenWrt version)_

ath79 (no ar71xx!):
- any ubnt,xm
- any ubnt,xw
- ubnt,airrouter
- ubnt,unifi
- avm,fritz300e
- glinet,gl-ar300m
- avm,fritz4020

ramips:
- dir-860l-b1
- gl-mt300a
- gl-mt300n
- gl-mt300n-v2
- gl-mt750
- rt-ac51u
- vocore2
- wt3020
- zbt-wg3526

adrianschmutzler, 2019-07-03:
In OpenWrt PR, GL-AR300M is reported to have eth0 as primary MAC address:
https://github.com/openwrt/openwrt/pull/2159#issuecomment-507695732
https://github.com/openwrt/openwrt/pull/2159#issuecomment-507907279

In Gluon it defaults to phy0 if I'm not mistaken (but flash source address is different). Maybe that's incorrect in Gluon?

adrianschmutzler, 2019-09-26:
Okay, so label MAC support in OpenWrt is added.

In the release after openwrt-19.07, this should provide an option to reduce the number of devices that require setting the primary MAC address in Gluon.

Just to document it here:
There is a new function "get_mac_label" in /lib/functions/system.sh, that will provide the label MAC address if set in OpenWrt or an empty string if not.
Setting the MAC address in OpenWrt can be done via label_mac_device in DTS or via ucidef_set_label_macaddr in board.d/02_network.
The latter will just write the correct MAC address into system.@system[0].label_macaddr

So, Gluon could exploit that by using the function get_mac_label for retrieving the primary mac address and thus benefit from the values provided in OpenWrt.
Despite, Gluon could also exploit the UCI setting and write primary MAC addresses missing in OpenWrt directly there, so no additional mechanism is used when retrieving them.
That way, 010_primary_mac in Gluon could be changed into a uci-default script just setting system.@system[0].label_macaddr for devices not having an OpenWrt-based label MAC address.

Exploiting the latter, one could even implement this mechanism already with openwrt-19.07, where just the two patches providing the function get_mac_label would have to be backported:
https://github.com/openwrt/openwrt/commit/0340718863040919ecaba74b2173ed52ebc194ce
https://github.com/openwrt/openwrt/commit/469e347f19ce9eefdc16f421b8e1f18ed60c310c
This won't provide any addresses from OpenWrt, but would enable to already modify 010_primary_mac to use system.@system[0].label_macaddr

This is just meant as a description on how you _may_ proceed, so feel free to ignore it if you do not like the approach. Since this PR was about adding support to OpenWrt and this is done now for most of Gluon's devices (except ar71xx), I will close this issue.

closed 2019-09-26 by adrianschmutzler