freifunk-gluon/gluon #1768: ath9k devices run out of memory above certain client count

mweinelt, 2019-06-25:
### Bug report

**What is the problem?**

Devices with ath9k radios seem to be quickly running out of memory as soon as they reach a certain client count.

Running `wifi` causes the memory to be freed again 

Example: Archer C7 v2 looses ~68 MB of normally free RAM in roughly 20 minutes, then OOM reboots, multiple times

https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?orgId=1&var-node=f4f26d707c53&refresh=15m

[image]
[image]
[image]

Other affected nodes are:
- 1043N v3
  https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?orgId=1&var-node=f4f26d3e9cac&refresh=15m
- CPE210 v1.1
  https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?orgId=1&var-node=8416f96a33dc&refresh=15m

**What is the expected behaviour?**

Memory usage should increase at a reasonable rate per client

**Gluon Version:**

v2018.2.1-7-geed810aa

**Site Configuration:**

https://git.darmstadt.ccc.de/ffda/site/tree/v2018.2.x

**Custom patches:**

none

blocktrron, 2019-06-25:
We were experiencing this issue since last year, so probably everything based on LEDE 17.01 onwards is affected.

awlx, 2019-06-25:
Could it be related to this? 
https://forum.openwrt.org/t/page-table-leak-on-tp-link-td-w8980-with-openwrt-18-06-2/39356

CodeFetch, 2019-06-26:
A log would be great. This should really not happen...

mweinelt, 2019-06-26:
So far we have not seen anything meaningful in dmesg/logread. Errors/warnings in there are only symptomatic, as they happen late when little memory is left.

We have not found any issue in userspace and therefore continued looking into the kernel space.

I've asked tackin on IRC to provide use with the outputs of `/proc/slabinfo` while a router is leaking memory, which he will likely provide tomorrow.

mweinelt, 2019-06-26:
We have currently only seen this kind of memory leak on devices featuring ath9k radios, but that might be totally circumstantial.

I'd be totally happy if people could check their high-client APs for frequent out-of-memory reboots. I'm especially interested in devices that are not ar71xx/ath79 based.

mweinelt, 2019-06-26:
We currently assume IPQ40xx (dual ath10k) not to be affected, since we have this sample data:

FRITZ!Box 4040
0.11.8+tackin / gluon-v2018.2.1-3-g015b0d68+
https://maps.tackin.de/#!/en/map/f0b014618973
https://draco.freifunk-trier.starletp9.de:3000/d/q8Ud1fJiz/single-nodes?orgId=1&var-node_name=54309_Fritz4040_DigiLay_Test&from=1561088751091&to=1561341537117

[image]
[image]
[image]

CodeFetch, 2019-06-27:
Great, so this is likely an ath9k bug. The ath9k changes are sparse, but it does happen often that there are memory leaks.
It's important to know the first Gluon version with this issue. Then we can easily find the commit. The biggest change in ath9k recently was the introduction of airtime fairness. I remember there were many leaks which were fixed one after another, but maybe the haven't found all of them.

This is surely a leak:
[image]

CodeFetch, 2019-06-27:
The interesting thing is that this happened as the client count decreased:
https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?orgId=1&var-node=f4f26d3e9cac&from=1561567558793&to=1561580164838

awlx, 2019-06-27:
I think Ubiquiti also had some major memory leaks with ath9k and airtime fairness ... I just don't find the release notes for it anymore. Because this could give as hint in which kernel version it was fixed :/.

Or they fixed it for themselves and didn't upstream it :/.

CodeFetch, 2019-06-27:
echo 0 > /sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags

CodeFetch, 2019-06-27:
@nbd168 I know you're very busy, but I believe to remember that you commited a fix of a memory leak introduced with airtime fairness support in ath9k.

Can you give us a hint on what to do?

BTW do you go to Paris?

awlx, 2019-06-27:
> echo 0 > /sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags

Thx, I hope I get some people to try this :).

mweinelt, 2019-06-27:
We've disabled airtime fairness on device mentioned in https://github.com/freifunk-gluon/gluon/issues/1768#issuecomment-506291460 and https://github.com/freifunk-gluon/gluon/issues/1768#issuecomment-506291951. Let's wait and see.

tackin, 2019-06-27:
Like to add one:
TP-Link Archer C7 v2  on 0.11.8+tackin / gluon-v2018.2.1-3-g015b0d68+
[image]
[image]

tackin, 2019-06-27:
> echo 0 > /sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags
ArcherC7_v2 says:
cat /sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags
cat: can't open '/sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags': No such file or directory

Am I wrong or is this disabled by default in the releases 2018.2.x ?

CodeFetch, 2019-06-27:
@tackin I'm on v2018.2.1 and have this entry.

tackin, 2019-06-27:
@CodeFetch ergo: not the reason for the memory leak, as we don't seem to have this entry.

CodeFetch, 2019-06-27:
@tackin That doesn't mean it's not activated. Just that you can't disable it. Airtime fariness is  activated since its introduction by default and a very nice feature BTW.

mweinelt, 2019-06-27:
On an Archer C7 V2 the ath9k radio could as well be phy1, so check `/sys/kernel/debug/ieee80211/phy1/ath9k/airtime_flags` instead.

CodeFetch, 2019-06-27:
I've checked our network. We don't have this issue but we have at most 35 clients on ath9k devices. It seems that this issue only happens when there are many, many clients...

Thus I'd recommend someone to create a lot of STA VIFs with an ath9k device and stress another ath9k device with that. ath9k devices can create more than 2000 STA VIFs as far as I understand.

tackin, 2019-06-27:
> 
> 
> On an Archer C7 V2 the ath9k radio could as well be phy1, so check `/sys/kernel/debug/ieee80211/phy1/ath9k/airtime_flags` instead. But please don't do it on the device you are currentyl collecting slabinfo from! :)
@mweinelt:
# cat /sys/kernel/debug/ieee80211/phy1/ath9k/airtime_flags
7

Soooory! Thought I tried both.

A-Kasper, 2019-06-27:
We have some Ubiquiti Bullets with many clients and Reboots. If you provide exact steps I would provide logs/monitor and test whatever you like. just need detailed instructions. I could also give direct access to these nodes, because they belong to our Verein and are placed in Streetlights

http://map.freifunk-bochum.de:3000/d/000000004/node-byid?orgId=1&var-nodeid=68725106af6e&from=now%2Fd&to=now

CodeFetch, 2019-06-27:
Here is a simple ash script to create STA VIFs for simulating WLAN clients. It should be used on a clean OpenWrt. I haven't really tested it, because I've other things to do atm.
With "scriptname down" you can remove the interfaces again.

```
#!/bin/sh

ssid=Freifunk

vifs=20
radio=phy0
prefix=test

if [ $1 = "down" ]; then
	i=0
	while [ $i -lt $vifs ]; do        
        	ip link set $prefix$i down
	        iw dev $prefix$i del
	        i=$(( $i + 1 ))
	done
	exit 1
fi

i=0
while [ $i -lt $vifs ]; do
	mac=DE:AD:01:$(hexdump -n3 -e '/1 ":%02X"' /dev/urandom|sed s/^://g)
	iw phy $radio interface add $prefix$i type station
	ip link set $prefix$i address $mac
	ip link set $prefix$i up
	iw $prefix$i connect $ssid
	i=$(( $i + 1 ))
done
```

kpanic23, 2019-06-27:
WR841N v9 with 64MB RAM in the city centre's pedestrian area. lots of cafes, bars, restaurants.
Memory usage peaks before every reboot.

[image]

CodeFetch, 2019-06-27:
For those who want to check their network here's a very simple python3-script to extract 10 nodes with the highest client count from a meshviewer.json-file:
```
import json

print([[n['hostname'],n['clients'],n['clients_wifi24'],n['clients_wifi5']] for n in sorted(json.load(open('meshviewer.json'))['nodes'], key=lambda n: n['clients'])[::-1][:10]])
```

tackin, 2019-06-28:
An extract of several /proc/slabinfo dumps during RAM-leaking:

kmalloc-4096         110    133   4352    7    8 : tunables    0    0    0 : slabdata     19     19      0
kmalloc-4096         134    147   4352    7    8 : tunables    0    0    0 : slabdata     21     21      0
kmalloc-4096         134    245   4352    7    8 : tunables    0    0    0 : slabdata     35     35      0
kmalloc-4096         139    147   4352    7    8 : tunables    0    0    0 : slabdata     21     21      0
kmalloc-4096        1700   1764   4352    7    8 : tunables    0    0    0 : slabdata    252    252      0
kmalloc-4096        1967   1995   4352    7    8 : tunables    0    0    0 : slabdata    285    285      0
kmalloc-4096        4146   4151   4352    7    8 : tunables    0    0    0 : slabdata    593    593      0
kmalloc-4096        5118   5222   4352    7    8 : tunables    0    0    0 : slabdata    746    746      0
kmalloc-4096        5337   5355   4352    7    8 : tunables    0    0    0 : slabdata    765    765      0
kmalloc-4096        5532   5551   4352    7    8 : tunables    0    0    0 : slabdata    793    793      0
kmalloc-512          273    280    768   10    2 : tunables    0    0    0 : slabdata     28     28      0
kmalloc-512          295    320    768   10    2 : tunables    0    0    0 : slabdata     32     32      0
kmalloc-512          296    320    768   10    2 : tunables    0    0    0 : slabdata     32     32      0
kmalloc-512          335    400    768   10    2 : tunables    0    0    0 : slabdata     40     40      0
kmalloc-512         2138   2150    768   10    2 : tunables    0    0    0 : slabdata    215    215      0
kmalloc-512         2401   2420    768   10    2 : tunables    0    0    0 : slabdata    242    242      0
kmalloc-512         2496   2500    768   10    2 : tunables    0    0    0 : slabdata    250    250      0
kmalloc-512         3611   3620    768   10    2 : tunables    0    0    0 : slabdata    362    362      0
kmalloc-512         3687   3700    768   10    2 : tunables    0    0    0 : slabdata    370    370      0
kmalloc-512         3847   3850    768   10    2 : tunables    0    0    0 : slabdata    385    385      0

skbuff_head_cache    517    644    288   14    1 : tunables    0    0    0 : slabdata     46     46      0
skbuff_head_cache    519    644    288   14    1 : tunables    0    0    0 : slabdata     46     46      0
skbuff_head_cache    522    644    288   14    1 : tunables    0    0    0 : slabdata     46     46      0
skbuff_head_cache    522    798    288   14    1 : tunables    0    0    0 : slabdata     57     57      0
skbuff_head_cache   5887   6104    288   14    1 : tunables    0    0    0 : slabdata    436    436      0
skbuff_head_cache   6275   6356    288   14    1 : tunables    0    0    0 : slabdata    454    454      0
skbuff_head_cache   8442   8456    288   14    1 : tunables    0    0    0 : slabdata    604    604      0
skbuff_head_cache  11574  11676    288   14    1 : tunables    0    0    0 : slabdata    834    834      0
skbuff_head_cache  12080  12236    288   14    1 : tunables    0    0    0 : slabdata    874    874      0
skbuff_head_cache  13035  13048    288   14    1 : tunables    0    0    0 : slabdata    932    932      0

tackin, 2019-06-28:
btw:
I can confirm LAN-Clients never cause this problem. I have a ArcherC7v2 on gluon-v2018.2.1+ with more than 70 LAN-Clients peaks each day, that never leaked.

CodeFetch, 2019-06-28:
@tackin Thank you very much. I need to see the whole output. If that is the whole output then the cache names are missing and it is quite useless. Nonetheless I can't see anything suspicious here.

BTW we need one output from the device when it's working fine and one when it is leaking.

tackin, 2019-06-28:
@CodeFetch : I edited the the previous comment.

CodeFetch, 2019-06-28:
@tackin Great! Now I understand why you posted the extract. So there seem to be skbs aggregating. I'll have a look what can cause this...

CodeFetch, 2019-06-28:
It could be possible that the ath9k driver can't send them out fast enough, because the driver's TX-queue is not triggered by mac80211 often enough. (retry packets aggregating)

There has been a fix for this by nbd168: https://github.com/torvalds/linux/commit/2b4a66980217332d91ab1785e1750857d6d52bc8#diff-ccb93195360636a322b558d121709687

But that's only a guess...

tackin, 2019-06-28:
I'm not in the details of the programming but my thinking is: 40+ clients may stuff the air down to a minumum throughput and make the driver buffer to increase more and more.

On my Fritzbox4040 I had only 1MB/s Traffic in the uplink but was not able to stream a mini video anymore when 40 other clients are around. They all did not produce much traffic (1 MB) or couldn't anymore because airtime is stuffed.
But on a Fritzbox4040 we don't see that leaking even with 70 Clients.

CodeFetch, 2019-06-28:
@tackin 
I haven't found anything that was fixed that could lead to this behavior. Thus either the issue still persists with the kernel master or it was fixed incidentally. The only way to track this issue down is to use kmemleak.

TomSiener, 2019-06-28:
Setting /sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags from 7 to 0 and restarting wifi does not affect the oom behaviour on an CPE210v2 with 2017.1.8

CodeFetch, 2019-06-28:
@TomSiener Turning off airtime fairness does not need a WiFi restart. Is it still set to 0 afterwards?

TomSiener, 2019-06-28:
yes. the value 0 is remaining after "wifi"

awlx, 2019-06-28:
We rolled out Gluon v2018.2.2 over the night let's see if this changes something.
Previously affected nodes:
https://map.ffmuc.net/#!/en/map/98ded0a79164
https://map.ffmuc.net/#!/en/map/60e327edfb5a

mweinelt, 2019-06-28:
Please try Gluon master, merge this commit https://github.com/freifunk-gluon/gluon/commit/65d1f4ea8f82 and build with `GLUON_DEBUG=true`.

tackin, 2019-06-28:
@awlx  My RAM-leak-test-node is on gluon 2018.2.2 as well. ... so upgrading  will not help.

mweinelt, 2019-06-28:
According to @T-X kmemleak will do us no good:

> 17:06 <T_X> hm, I don't expect CONFIG_DEBUG_KMEMLEAK to reveal anything here. the "wifi" command is able to free the memory, so there must be some references left
> 17:07 <T_X> KMEMLEAK can only detect memory leaks which have no more references

nbd168, 2019-06-28:
Maybe it's something like what I fixed in OpenWrt commit 0c285bd081da55bd63da18f7596e7107a46bb798

CodeFetch, 2019-06-28:
@mweinelt Right, I overestimated what kmemleak can do :(... I expected it to detect stale sk_buffs, but it is just really stupid...

@nbd168 As this issue seems to only happen on ath9k devices I suspect the TX-retry-queue. I can't find a case in ath9k similar to your commit. It does not really explain why this happens with so many clients (maybe collisions). Why is there no limit for the TX queues' sizes in ath9k? ATH_MAX_SW_RETRIES is set to 30 by default. Is it possible that it just fills the memory?

CodeFetch, 2019-06-28:
@tackin @awlx  @TomSiener Can you please try an image built with CONFIG_SLUB_DEBUG=y?

Then execute echo 1 > /sys/kernel/slab/kmalloc-4096/trace and post the dmesg output before and when the memory increases.

tackin, 2019-06-28:
Errored on my build for 2018.2.2 with:
Configuration failed:
 * unable to set 'CONFIG_SLUB_DEBUG=y'
Makefile:126: die Regel für Ziel „config“ scheiterte
make: *** [config] Fehler 1

tackin, 2019-06-28:
sorry ...
Try to build 2018.2.2 and add in file /gloun/targets/generic

config 'CONFIG_SLUB_DEBUG=y'

Configuration failed:
 * unable to set 'CONFIG_SLUB_DEBUG=y'
Makefile:126: die Regel für Ziel „config“ scheiterte

rubo77, 2019-06-29:
Maybe CONFIG_SLUB_DEBUG_ON ?

NeoRaider, 2019-06-29:
```
config 'CONFIG_KERNEL_SLABINFO=y'
```
(see `openwrt/config/Config-kernel.in`)

tackin, 2019-06-29:
@NeoRaider @rubo77 @CodeFetch
I already build a firmware with config 'CONFIG_KERNEL_SLABINFO=y' and posted the output of /proc/slabinfo

> 
> Then execute echo 1 > /sys/kernel/slab/kmalloc-4096/trace and post the dmesg output before and when the memory increases.

So do I understand you right, that I can/should do this on that build without adding somthing els to the build?

tackin, 2019-06-29:
Need to say, I build the slabinfo-build with:
diff --git a/targets/generic b/targets/generic
index 517603a6..a05296a3 100644
--- a/targets/generic
+++ b/targets/generic
@@ -51,6 +51,7 @@ if [ "$GLUON_DEBUG" = 1 ]; then
        config '# CONFIG_USE_SSTRIP is not set'
 fi
 
+config 'CONFIG_KERNEL_SLABINFO=y'
 
 packages '-odhcpd-ipv6only' '-ppp' '-ppp-mod-pppoe' '-wpad-mini'
 packages 'gluon-core' 'ip6tables'

but without the make OPTION  GLUON_DEBUG=1 
Is this important?

CodeFetch, 2019-06-29:
@NeoRaider I'd like to see a SLUB trace. For that CONFIG_SLUB_DEBUG needs to be set.

@tackin SLUB != SLAB
I don't know why CONFIG_SLUB_DEBUG fails to build. Maybe because the symbol is missing in the ar71xx-config. But as you can [see here](https://github.com/openwrt/openwrt/blob/master/target/linux/generic/config-4.9) this entry exists.  I'll try later, but I'm very busy atm.

CodeFetch, 2019-06-29:
@tackin @awlx @TomSiener 

I've looked at the dependencies of CONFIG_SLUB_DEBUG and there should be no reason why it fails. Please build with V=s and this commit https://github.com/freifunk-gluon/gluon/commit/bbde72247b364488254223e23fe5e9470e8a7fb4.

@tackin Without GLUON_DEBUG the symbols might be stripped which gives unusable output.

NeoRaider, 2019-06-29:
@CodeFetch You are mixing up OpenWrt and kernel .config. CONFIG_SLUB_DEBUG is a kernel config symbol, you can not directly affect those in the target config.

Fortunately, OpenWrt provides CONFIG_KERNEL_SLABINFO, which also enables CONFIG_KERNEL_SLUB_DEBUG (which can't be set directly, as it is a hidden config symbol):
```
config KERNEL_SLABINFO
        select KERNEL_SLUB_DEBUG
        select KERNEL_SLUB_DEBUG_ON
        bool "Enable /proc slab debug info"
```

CodeFetch, 2019-06-29:
@NeoRaider Oh damned... You're so right. Thank you.

@tackin Can you please provide the dmesg output before and while the memory increases? echo 1 > /sys/kernel/slab/kmalloc-4096/trace 
and GLUON_DEBUG needs to be set to 1 or the output will be useless.

tackin, 2019-06-29:
@CodeFetch New build is in progress. I will probably have your requested data with the first oom-reboot tomorrow afternoon.

tackin, 2019-06-30:
I'm so sorry people. In the night I bricked the WDR4300-Test-node with my flashing (wget an uncomplete Image and sysupgraded). This morning I replaced the WDR4300 with a older WR1043.
Each time I try to do 
```
cat  /sys/kernel/slab/kmalloc-4096/trace
0
echo 1 > /sys/kernel/slab/kmalloc-4096/trace

```
This kills the router and need to powercyle.

After all I'm also not able to get the WDR4300 in recovery-mode to tftp a gluon-factory-image as in the howto https://openwrt.org/toh/tp-link/tl-wdr4300.
For the PC it looks like the node-swichtports are "dead" most times.

CodeFetch, 2019-06-30:
@tackin The need to power-cycle sounds like it is still working, but overwhelmed with the debugging. That's a good sign. If you could connect it using the serial console, we'd likely see some useful messages.

DL3DCF, 2019-06-30:
I set `/sys/kernel/debug/ieee80211/phy0/ath9k/airtime_flags` to `0` on a few testnodes with very frequent OOM events since updated to 2018.2.1. 

It did a great impact on the uptime of two Ubiquiti NSM2 loco with xm hardware. After suffering OOMs about every hour, both nodes now have an uptime of 4 days. Right now both have >40 clients and are still healthy.

On a TP-Link Archer C7 v5 the setting on `phy1` (`phy0` does not have `airtime_flags`) also noticeably improved the stability. Only a single OOM event occurred afterwards, probably caused by the ath10k device. The prior uptime was about 50 minutes with 40-50 clients.

What does the "airtime fairness" do and how big is it's impact on throughput?

tackin, 2019-06-30:
@CodeFetch 
The device is about 20km away from my home and has no Serial-connector.  I also have no serial-UART.
A friend is back form sailing next sunnday. If he has time he can help with uart and soldering a serial.
The WR1043v1 is this one: https://draco.freifunk-trier.starletp9.de:3000/d/q8Ud1fJiz/single-nodes?orgId=1&var-node_name=54309_Test_1043_xx3
with very less free RAM atm. Not sure if you might see anything important. OOM occured with in seconds, not minutes like on the 4300.

CodeFetch, 2019-06-30:
@DL3DCF Airtime fairness is a great feature, but the kernel version 4.9 that we currently use does only support it, because nbd patched it. Thus it might be more stable with a newer kernel version. It measures the airtime it uses to send data to the clients and distributes the times equally among the stations. This is useful, because a station with a very low data rate/bad signal could take up most of the airtime, while transferring very few data. When you have many clients there likely is one that has a bad signal and all the other clients suffer from it. With airtime fairness it's like: Your signal sucks. That's your fault. I don't spend all my airtime on you, because I would let others suffer who can receive a 100 times more data in that same time frame.

CodeFetch, 2019-06-30:
@tackin Did it reboot on its own or did you need to power-cycle it?

[83 more replies]

closed 2020-08-28 by rotanid