freifunk-gluon/gluon #1771: [tp-link-archer-c25-v1] nodeid/primary mac not matching device label

Adorfer, 2019-06-27:
**What is the problem?**

Affected device:  TP-LINK Archer C25 v1

nodeid after flashing via "sysupgrade -n" (or firstboot) 
50c7bf25482f (mac:  50:c7:bf:25:48:2f)

**What is the expected behaviour?**

primary mac (->nodeid) should correspond to sticker label mac: 
50c7bf254830 (mac:  50:c7:bf:25:48:30)

**Gluon Version:**
gluon-v2018.2.2 (800c3b2bacdf3175b153413adc15ccdf802de2c8)

**Site Configuration:**

site.conf/site.mk/modules/buildlog: 
see: https://images.ffdus.de/brokenl2tp.key/site/

pictures 

[image]

[image]

rotanid, 2019-06-27:
please check with plain openwrt 18.06.3 and probably you can then report it upstream at openwrt

mweinelt, 2019-06-27:
@Adorfer Can you post the output of `ip link`? Is the sticker address visible on any interface?

Adorfer, 2019-06-27:
> please check with plain openwrt 18.06.3

What should i look for on OpenWRT?

mweinelt, 2019-06-27:
@Adorfer `cat /tmp/sysinfo/board_name`

Adorfer, 2019-06-27:
archer-c25-v1 (as stated in the issue)

mweinelt, 2019-06-27:
Ok, please try out https://github.com/freifunk-gluon/gluon/commit/10d05d268c60aa9c0b0db7eff88bddc20366fac7, it should provide a fix.

CodeFetch, 2019-06-27:
@rotanid It's not an upstream issue as OpenWrt does not care about which interface's MAC address is on the label at the moment. Adrian Schmutzler is working on it for ath79 as described in https://github.com/freifunk-gluon/gluon/issues/1767

Adorfer, 2019-06-27:
> Ok, please try out [10d05d2](https://github.com/freifunk-gluon/gluon/commit/10d05d268c60aa9c0b0db7eff88bddc20366fac7), it should provide a fix.

works like a treat! Thanks.

`root@dus-50c7bf254830`

closing.

closed 2019-06-27 by Adorfer

adrianschmutzler, 2019-07-01:
FYI: Note that C25 v1 networking was incorrectly set up as "two-port" device in OpenWrt. I fixes this only recently (master, 19.07, 18.06; both ath79 and ar71xx):
https://github.com/openwrt/openwrt/commit/cf2aa873ea1787d9206feac0391f748093c4809b

I do not know how Gluon sets up networking internally, but maybe the MAC address issue discussed here is a side-effect either of the incorrect initial setup or of my patch if it has been pulled already.

Also note that my OpenWrt-PR for primary-mac-address will not address ar71xx at all.

codedust, 2019-07-01:
For reference, here's a configuration dump of a C25:
```
root@64560-bw-ap6:~# cat /tmp/sysinfo/board_name
archer-c25-v1
root@64560-bw-ap6:~# cat /lib/gluon/release 
1.7~20190629
root@64560-bw-ap6:~# cat /lib/gluon/gluon-version 
v2018.2-168-g2e6c6f24
root@64560-bw-ap6:~# echo `cat /lib/gluon/core/sysconfig/primary_mac`
50:c7:bf:3a:6b:b1
root@64560-bw-ap6:~# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether ea:91:af:70:78:ef brd ff:ff:ff:ff:ff:ff
3: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel master br-wan state DOWN qlen 1000
    link/ether 50:c7:bf:3a:6b:b2 brd ff:ff:ff:ff:ff:ff
4: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 8e:3b:7e:c6:51:9c brd ff:ff:ff:ff:ff:ff
7: br-wan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether 8e:3b:7e:c6:51:98 brd ff:ff:ff:ff:ff:ff
8: local-port@local-node: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP qlen 1000
    link/ether 50:c7:bf:3a:6b:b1 brd ff:ff:ff:ff:ff:ff
9: local-node@local-port: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether da:ff:07:00:ff:ff brd ff:ff:ff:ff:ff:ff
10: br-client: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 50:c7:bf:3a:6b:b1 brd ff:ff:ff:ff:ff:ff
11: primary0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UNKNOWN qlen 1000
    link/ether 8e:3b:7e:c6:51:9b brd ff:ff:ff:ff:ff:ff
12: bat0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UNKNOWN qlen 1000
    link/ether 50:c7:bf:3a:6b:b1 brd ff:ff:ff:ff:ff:ff
14: vx_mesh_lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1430 qdisc noqueue master bat0 state UNKNOWN qlen 1000
    link/ether 8e:3b:7e:c6:51:9c brd ff:ff:ff:ff:ff:ff
17: mesh0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UP qlen 1000
    link/ether 8e:3b:7e:c6:51:99 brd ff:ff:ff:ff:ff:ff
18: mesh1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UP qlen 1000
    link/ether 8e:3b:7e:c6:51:9d brd ff:ff:ff:ff:ff:ff
19: client0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP qlen 1000
    link/ether 8e:3b:7e:c6:51:98 brd ff:ff:ff:ff:ff:ff
20: client1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP qlen 1000
    link/ether 8e:3b:7e:c6:51:9c brd ff:ff:ff:ff:ff:ff
```

adrianschmutzler, 2019-07-01:
@codedust What's the address on the label?

codedust, 2019-07-01:
@adrianschmutzler I'm not quite sure what you mean...

Adorfer, 2019-07-01:
what is the mac on your c25's bottom housing sticker?

commit 10d05d26 (2019-06-27): gluon-core: fix primary mac selection on tp-link archer c25 v1

Fixes #1771
commit 74bb239d (2019-08-26): gluon-core: fix primary mac selection on tp-link archer c25 v1

Fixes #1771