freifunk-gluon/gluon #1772: v2018.2.4 tracking issue

mweinelt, 2019-06-29:
Hi, we might need to do another minor release in the v2018.2.x series. There was a regression in the TCP SACK fixes, although I'm not sure if we are affected by it with the services we publicly expose TCP sockets on.

- [ ] Update past 4.9.184 and 4.14.131 to fix a regression of the TCP SACK fixes that unnecessarily punishes applications with low socket send buffers.
  ```
  commit caa51edc7e9606418611e68de624efbd0042adf5
  Author: Eric Dumazet <edumazet@google.com>
  Date:   Fri Jun 21 06:09:55 2019 -0700
  
      tcp: refine memory limit test in tcp_fragment()
      
      commit b6653b3629e5b88202be3c9abc44713973f5c4b4 upstream.
      
      tcp_fragment() might be called for skbs in the write queue.
      
      Memory limits might have been exceeded because tcp_sendmsg() only
      checks limits at full skb (64KB) boundaries.
      
      Therefore, we need to make sure tcp_fragment() wont punish applications
      that might have setup very low SO_SNDBUF values.
      
      Fixes: f070ef2ac667 ("tcp: tcp_fragment() should apply sane memory limits")
      Signed-off-by: Eric Dumazet <edumazet@google.com>
      Reported-by: Christoph Paasch <cpaasch@apple.com>
      Tested-by: Christoph Paasch <cpaasch@apple.com>
      Signed-off-by: David S. Miller <davem@davemloft.net>
      Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
  ```

rubo77, 2019-06-29:
The next would be v2018.2.3

rotanid, 2019-07-02:
as the OpenWrt 18.06 branch contains the mentioned fixes and a release 18.06.4 has been tagged, we could bump OpenWrt to receive the patch and resolve this issue.
@mweinelt mentioned on IRC though that he is not sure if Gluon is affected by the issue at all.

rotanid, 2019-08-27:
the v2018.2.x branch has just been updated with the backports @mweinelt and i decided worth backporting from master branch.

Adorfer, 2019-08-27:
i am not sure where this regression comes from, i had to add "kmod-sched" to the site.mk manually to have a working build since this month's commits on v2018.2.x (was working in july 2019)

https://github.com/freifunk-gluon/gluon/commit/aa2c0dc3f9819de9deb902a0c36f098f49619c2a#commitcomment-34849879

NeoRaider, 2019-08-27:
@Adorfer Please open a new ticket with the following info:
- site repo
- build target
- build log

rotanid, 2019-08-31:
> @Adorfer Please open a new ticket with the following info:
> 
>     * site repo
> 
>     * build target
> 
>     * build log

@Adorfer can you please do this? we don't want releases with new bugs, but we really need more information ...

Adorfer, 2019-08-31:
i have no idea how to report, because it's not linked to gluon itself, it's more an issue with the packages. So it's off-topic here.

rotanid, 2019-09-08:
i created a PR with a release notes draft: #1812

rotanid, 2019-09-14:
closing this as v2018.2.3 has been released.

closed 2019-09-14 by rotanid