freifunk-gluon/gluon #1777: Setting authorized_keys results in unauthorized access

yanosz, 2019-07-02:
### Bug report

**What is the problem?**
Setting `authorized_keys = { 'ssh-rsa AAAA ...}` and using gluon-authorized-keys results in ssh-access for root without any authentication. 

**What is the expected behavior?**
SSH-Access requires authentication, with the specified keys being accepted.

**Gluon Version:**
2018.2.x , master

**Site Configuration:**
https://github.com/FieldTracks/gluon-site-config 

**Custom patches:**
gluon-wlan-encrypt-psk (c.f. https://github.com/yanosz/gluon/commit/3aec1963dba2bb218a00472847731546d4db4f83) - I don't think it matters.

kevin-olbrich, 2019-07-02:
At least for my nodes, this is not reproducible.

Did you forget to include "lock-password" in your site.mk? (Seems to be missing in the repo).
Without that package, the nodes allow root access without any password (tested with skip setup images in 2017.x).

ecsv, 2019-07-02:
So you can now argue in many ways but I would guess that yanosz expected following dependency in gluon-authorized-keys:

    diff --git a/package/gluon-authorized-keys/Makefile b/package/gluon-authorized-keys/Makefile
    index 2995107a..02bfaf15 100644
    --- a/package/gluon-authorized-keys/Makefile
    +++ b/package/gluon-authorized-keys/Makefile
    @@ -7,7 +7,7 @@ include ../gluon.mk
     
     define Package/gluon-authorized-keys
       TITLE:=Fill /etc/dropbear/authorized_keys from site.conf
    -  DEPENDS:=+gluon-core
    +  DEPENDS:=+gluon-core +gluon-lock-password
     endef
     
     $(eval $(call BuildPackageGluon,gluon-authorized-keys))

But it is not there and it is not documented anywhere that you should enable ``gluon-lock-password`` when you enable ``gluon-authorized-keys``.

Most people just have it implicitly enabled because they have (gluon-)setup-mode enabled.

yanosz, 2019-07-02:
Thanks for the feedback - that's fast :). 

gluon-lock-password is new to me. I checked the docs twice and and asked on irc.  Searching for "lock-password" in the docs has no match either.

@ecsv Manually applying the patch fixes the issue. Introducing a dependency like this looks very reasonable to me. Thanks!  You're correct about the missing config mode - my configuration is rather unique, I guess.

Please let me know, when this patch is merged - I'd like to do a re-test, then.

mweinelt, 2019-07-02:
@ecsv Care to create a pull request?

mweinelt, 2019-07-02:
Also backported to v2018.2.x in 2280cb3a09ea4eb2b4c68154f60fd7d7cc9dee80.

closed 2019-07-02 by mweinelt

yanosz, 2019-07-02:
Confirmed - both https://github.com/freifunk-gluon/gluon/commit/2280cb3a09ea4eb2b4c68154f60fd7d7cc9dee80 and https://github.com/freifunk-gluon/gluon/commit/2280cb3a09ea4eb2b4c68154f60fd7d7cc9dee80 result in fixed installations.

Adorfer, 2019-07-06:
> results in ssh-access for root without any authentication.

can't reproduce in 2018.2.2. 
pleas sum up the situation for old fw. What is the race condition to stubmle into this.

rotanid, 2019-07-06:
depends on certain packages:
"using gluon-authorized-keys" - without using gluon-lock-password

Adorfer, 2019-07-06:
and "gluon-lock-password" is only installed automatically with which packages in the site.mk enabled?
Or has it to be included "explicitly" in the site.mk, otherwise the builds are vulnerable?

NeoRaider, 2019-07-06:
@Adorfer Builds that include the config mode are fine.

yanosz, 2019-07-06:
To make sure that you're not effected, SSH to your node and execute:
```bash
if [ "$(opkg list-installed | grep gluon-lock-password)" == "" ]; then echo "You are AFFECTED";else echo "You are not affected";fi
```

commit 2eb15bac (2019-07-02): gluon-authorized-keys: Force installation of gluon-lock-password

The gluon-authorized-keys is usually installed to use SSH keys to
authenticate a user against the device. To make this useful, it is also
required to disable passwordless SSH access to the device.

This new dependency is only required when the user doesn't have
gluon-setup-mode enabled already.

Fixes: #1777
Reported-by: yanosz <github@yanosz.net>
Fixes: a753fa79e30f ("gluon-authorized-keys: add keys from site.conf")
commit 2280cb3a (2019-07-02): gluon-authorized-keys: Force installation of gluon-lock-password

The gluon-authorized-keys is usually installed to use SSH keys to
authenticate a user against the device. To make this useful, it is also
required to disable passwordless SSH access to the device.

This new dependency is only required when the user doesn't have
gluon-setup-mode enabled already.

Fixes: #1777
Reported-by: yanosz <github@yanosz.net>
Fixes: a753fa79e30f ("gluon-authorized-keys: add keys from site.conf")

(cherry picked from commit 2eb15bac0e974385b6b49d3c9e74f623d3d63c3f)