freifunk-gluon/gluon #1783: RFC: respondd: Determine tunnel interfaces by checking for tun_flags and Pointopoint flag

christf, 2019-07-06:
After a discussion in #wireguard, it seems like respondd is not correctly determining whether a  mesh interfaces is a tunnel or not. When considering the pointtopoint flag on each interface, it would consider wireguard interfaces as well.

 I cannot think of any gotchas, what are your thoughts?

NeoRaider, 2019-07-09:
It seems preferable to me to actually check the interface type string using netlink.

Currently we check whether an interface is a TUN/TAP interface or not; I assume this will not work for L2TP either.

christf, 2019-08-25:
I am assuming you are referring to the unsighned short ifi_type in struct ifinfomsg? For this I have not been able to find a definition of its contents.

According to the docs there is also a rtattr attached called IFLA_LINK that contains the "link type". The kernel header says:
```
  /* IFLA_LINK.
     For usual devices it is equal ifi_index.
     If it is a "virtual interface" (f.e. tunnel), ifi_link
     can point to real physical interface (f.e. for bandwidth calculations),
     or maybe 0, what means, that real media is unknown (usual
     for IPIP tunnels, when route to endpoint is allowed to change)
   */
``` 
That would require to figure out whether the ifindex is equal to the IFLA_LINK value and if it isn'T then deduce that the current interface must be a tunnel of some sort. Imho this seems to be our best shot.

NeoRaider, 2020-05-10:
@christf  I was refering to the `IFLA_INFO_KIND` field (nested in `IFLA_LINKINFO`), which contains the link kind as a string (like "vxlan").

NeoRaider, 2020-05-10:
You can also look at the `iproute2` code. The kind is the first word in the third line printed by `ip -d link` for virtual interfaces.

NeoRaider, 2020-06-01:
Hmm, this issue is much more cumbersome than I anticipated. There are way too many ways to ask the kernel what "type" an interface is, and every driver implements a different subset:

- `ifi_itype` just tells us whether an interface uses Ethernet headers or something else (so it has the same value for all interface types we support for meshing). The same information can be found in `/sys/class/net/*/type`
- IFLA_LINK is unset for most interface types (but there is `/sys/class/net/*/iflink`, which contains this information, but it is just set to the ifindex for tunnel interfaces, and also not netns-aware; all in all, useless)
- IFLA_INFO_KIND is not set by all drivers; no way to detect wireless or l2tp interfaces; ok for wireguard AFAICT from the kernel code; ok for TUN/TAP
- ethtool driver information: Not implemented for l2tp or wireguard, no good solution for wireless; ok for TUN/TAP
- `/sys/class/net/*/uevent` DEVTYPE: This one is looking good, shows WLAN, Wireguard and L2TP accurately - for TUN/TAP we'll need to keep the `tun_flags` check

There is also the option to query the various subsystems of the kernel whether a specific interface belongs to a subsystem (e.g. ask Wireguard whether X is a wg interface, ask L2TP whether X is L2TP), but with the goal in mind to keep us sane and the code small, this doesn't seem like a viable solution either.

commit 3fda210f (2020-08-08): gluon-mesh-{batman-adv,babel}: respondd: use libgluonutil to determine interface type

Also make babel match batman-adv and only emit the wireless/tunnel/other
fields when they are non-empty.

Fixes: #1783
closed 2020-08-08 by rotanid