freifunk-gluon/gluon #1801: Enhance Autoupdater with Beta Repo

A-Kasper, 2019-08-24:
Whenever we realease a new firmware some people are doing some kind of beta testing. installing a non release channel image is complcated with wget etc.

I would like to suggest to enhance the autoupdater with another commandline option. something like autoupdater --chanel test
with this the channel named test should be used instead of the default channel in config. all without changing the release channel.

This shouldn't be tooo complex, but would help more people to test new images

kpanic23, 2019-08-24:
This feature already exists.
Just run `autoupdater -b $BRANCHNAME`

rotanid, 2019-08-24:
> This feature already exists.
> Just run `autoupdater -b $BRANCHNAME`

this is it.

closed 2019-08-24 by rotanid

Adorfer, 2019-08-25:
I would suggest to add this option explicitly in 
https://gluon.readthedocs.io/en/v2018.2.x/features/autoupdater.html#command-line

mweinelt, 2019-08-25:
Pull request welcome.