freifunk-gluon/gluon #1806: Start 'haveged' in config mode

oszilloskop, 2019-08-31:
I'm currently testing the latest Babel firmware. This firmware was built with the current master in combination with the Babel-PR. 

During my tests I noticed that in some cases booting into config mode does not work correctly anymore. After a `'firstboot'` command somehow my routers get stuck during the following boot process and I don't get to the 192.168.1.1 config mode website. (I used 'firstboot' to create the same behavior like after a factory flash and/or to simulate a sysupgrade without taking over the parameters.)

Same behavior on an AVM4020, an UBNT ERX and a DIR-505.

I connected a serial console to a UBNT ERX and noticed that the pausing boot process is resumed when `'haveged'` is started manually. After the successful first initial boot there were no further problems during the following boot processes. (Until the next 'firstboot'.)

Following a serial log of a **bad** ERX boot process into config mode after 'firstboot' (including manually start of 'haveged'):

```
3: System Boot system code via Flash. 
## Booting image at bfd40000 ...      
   Image Name:   MIPS OpenWrt Linux-4.14.137         
   Image Type:   MIPS Linux Kernel Image (lzma compressed)          
   Data Size:    1969575 Bytes =  1.9 MB             
   Load Address: 80001000             
   Entry Point:  80001000             
...............................   Verifying Checksum ... OK         
   Uncompressing Kernel Image ... OK  
No initrd              
## Transferring control to Linux (at address 80001000) ...          
## Giving linux memsize in MB, 256    
        
Starting kernel ...    
        
[    0.000000] Linux version 4.14.137 (jenkins@build) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7835+24-89808e211c)) #0 SMP Sun Jun 23 13:10:01 2019             
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3 
[    0.000000] bootconsole [early0] enabled          
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)             
[    0.000000] MIPS: machine is UBNT-ERX             
[    0.000000] Determined physical RAM map:          
[    0.000000]  memory: 10000000 @ 00000000 (usable) 
[    0.000000] Initrd not found or empty - disabling initrd         
[    0.000000] VPE topology {2,2} total 4            
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.     
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes 
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:           
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]              
[    0.000000]   HighMem  empty       
[    0.000000] Movable zone start for each node      
[    0.000000] Early memory node ranges              
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]             
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]    
[    0.000000] random: get_random_bytes called from start_kernel+0x94/0x4b0 with crng_init=0      
[    0.000000] percpu: Embedded 14 pages/cpu s26256 r8192 d22896 u57344            
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024        
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2  
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)  
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)     
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)       
[    0.000000] Writing ErrCtl register=00020070      
[    0.000000] Readback ErrCtl register=00020070     
[    0.000000] Memory: 253108K/262144K available (4652K kernel code, 242K rwdata, 1032K rodata, 268K init, 254K bss, 9036K reserved, 0K cma-reserved, 0K highmem)            
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1          
[    0.000000] Hierarchical RCU implementation.      
[    0.000000] NR_IRQS: 256           
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns 
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns            
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns           
[    0.015493] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)             
[    0.087805] pid_max: default: 32768 minimum: 301  
[    0.097112] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)         
[    0.110132] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)    
[    0.125792] Hierarchical SRCU implementation.     
[    0.135196] smp: Bringing up secondary CPUs ...   
[    0.157031] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.     
[    0.157040] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes 
[    0.157051] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 375 lines cut ...]
[   12.815489] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.     
[   12.858959] xt_time: kernel timezone is -0000     
[   12.873245] kmodloader: done loading kernel modules from /etc/modules.d/*       
[   14.428479] urandom_read: 5 callbacks suppressed  
[   14.428491] random: jshn: uninitialized urandom read (4 bytes read)             
[   15.601668] random: crng init done 
[   18.300889] mtk_soc_eth 1e100000.ethernet: PPE started           
[   18.317545] br-setup: port 1(eth0.1) entered blocking state      
[   18.328736] br-setup: port 1(eth0.1) entered disabled state      
[   18.340451] device eth0.1 entered promiscuous mode
[   18.350044] device eth0 entered promiscuous mode  
[   18.361351] br-setup: port 1(eth0.1) entered blocking state      
[   18.372474] br-setup: port 1(eth0.1) entered forwarding state    
[   18.384327] IPv6: ADDRCONF(NETDEV_UP): br-setup: link is not ready              
[   19.312400] IPv6: ADDRCONF(NETDEV_CHANGE): br-setup: link becomes ready
```         

site.mk: 
https://github.com/FreifunkMD/site-ffmd/blob/babel/site.mk

UBNT-ERX image:
http://firmware.md.freifunk.net/experimental-babel-v0.0.74-experimental_babel-0829/sysupgrade/gluon-ffmd-babel-v0.0.74-experimental_babel-0829-ubnt-erx-sysupgrade.tar

AVM4020 image:
http://firmware.md.freifunk.net/experimental-babel-v0.0.74-experimental_babel-0829/sysupgrade/gluon-ffmd-babel-v0.0.74-experimental_babel-0829-avm-fritz-box-4020-sysupgrade.bin

NeoRaider, 2019-09-01:
Please check using `ps` which command/initscript is hanging.

christf, 2019-09-01:
how would this command be identified in the output of ps?

oszilloskop, 2019-09-01:
It looks like `'wg keygen'` is waiting for entropy.
`'wg keygen'` is called by the script '404-wireguard-generate-key'.

Each call of `'ps'` results in the following output:
```
root@(none):/# ps
  PID USER       VSZ STAT COMMAND
    1 root      1472 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 IW   [kworker/0:0]
    4 root         0 IW<  [kworker/0:0H]
    5 root         0 IW   [kworker/u8:0]
    6 root         0 IW<  [mm_percpu_wq]
    7 root         0 SW   [ksoftirqd/0]
    8 root         0 IW   [rcu_sched]
    9 root         0 IW   [rcu_bh]
   10 root         0 SW   [migration/0]
   11 root         0 SW   [cpuhp/0]
   12 root         0 SW   [cpuhp/1]
   13 root         0 SW   [migration/1]
   14 root         0 SW   [ksoftirqd/1]
   15 root         0 IW   [kworker/1:0]
   16 root         0 IW<  [kworker/1:0H]
   17 root         0 SW   [cpuhp/2]
   18 root         0 SW   [migration/2]
   19 root         0 SW   [ksoftirqd/2]
   20 root         0 IW   [kworker/2:0]
   21 root         0 IW<  [kworker/2:0H]
   22 root         0 SW   [cpuhp/3]
   23 root         0 SW   [migration/3]
   24 root         0 SW   [ksoftirqd/3]
   25 root         0 IW   [kworker/3:0]
   26 root         0 IW<  [kworker/3:0H]
   27 root         0 IW   [kworker/u8:1]
   31 root         0 IW   [kworker/u8:2]
   40 root         0 IW   [kworker/u8:3]
   48 root         0 IW   [kworker/u8:4]
   67 root         0 IW   [kworker/0:1]
  138 root         0 IW   [kworker/1:1]
  139 root         0 SW   [oom_reaper]
  140 root         0 IW<  [writeback]
  142 root         0 SW   [kcompactd0]
  143 root         0 IW<  [crypto]
  145 root         0 IW<  [kblockd]
  168 root         0 IW<  [watchdogd]
  192 root         0 SW   [kswapd0]
  251 root         0 IW<  [pencrypt]
  253 root         0 IW<  [pdecrypt]
  294 root         0 IW   [kworker/2:1]
  301 root         0 IW   [kworker/3:1]
  355 root         0 IW<  [ipv6_addrconf]
  366 root         0 SW   [ubi_bgt0d]
  372 root         0 IW<  [kworker/1:1H]
  373 root         0 IW<  [kworker/0:1H]
  374 root         0 IW<  [kworker/2:1H]
  375 root         0 IW<  [kworker/3:1H]
  389 root         0 IW   [kworker/0:2]
  390 root         0 IW   [kworker/0:3]
  391 root         0 IW   [kworker/0:4]
  434 root         0 SW   [ubifs_bgt0_1]
  502 root      1188 S    /sbin/ubusd
  503 root      1204 S    /bin/ash --login
  510 root      1272 S    /bin/sh /etc/rc.common /lib/gluon/setup-mode/rc.d/S1
  765 root      1272 S    /bin/sh /etc/rc.common /lib/gluon/setup-mode/rc.d/S1
  767 root      1208 S    /bin/sh /usr/bin/gluon-reconfigure
  804 root      1204 S    /bin/sh ./404-wireguard-generate-key
  806 root      1036 S    wg genkey
  818 root      1204 R    ps
```

oszilloskop, 2019-09-01:
During my tests I noticed that the boot process sometimes continues after a few minutes. The time varies a lot (a few minutes to ∞).

christf, 2019-09-01:
yeah it certainly looks that way. Interestingly, haveged is part of the image, it is apparently not started (in time?) on firstboot.

NeoRaider, 2019-09-01:
The wireguard package is the problem: The key must not be generated in the upgrade script, but only in an init script when wireguard is actually started. This can either be handled in the wireguard initscript itself, or in a separate init script that runs before wireguard's.

christf, 2019-09-01:
how then display the secret key at the end of the configuration process? Is wireguard even initialized in setup mode?

Interestingly, this does not happen when using fastd which uses the very same mechanism (using update script to generate the key)

NeoRaider, 2019-09-01:
@christf No update script is generating the fastd key, it is set to `generate` there. The actual key generation takes place during the `/etc/init.d/fastd show_key` call used to display the key on the reboot page.

christf, 2019-09-01:
this is fixed in the latest version.

closed 2019-09-01 by christf