freifunk-gluon/gluon #1809: gluon-web-wifi-config: fetching information returns Internal Server Error

blocktrron, 2019-09-04:
### Bug report

**What is the problem?**
WiFi config is not displayed when accessing the WiFi tab in advanced setup mode.
Instead only a box with the following content is displayed:

```
Failed to execute dispatcher target for entry '/admin/wifi-config'.
The called action terminated with an exception:
/usr/lib/lua/simple-uci.lua:22: attempt to index local 'c' (a nil value)
```

Viewing the recent commits for lua-simple-uci, whis commit could be interesting: https://github.com/freifunk-gluon/packages/commit/cf0619a0f9e1133652f8a7dbd663bcb6bcd64756#diff-951cae4df37f09f01930d67db0316037

**What is the expected behaviour?**
WiFi config should be visible.

**Gluon Version:**
recent master

**Site Configuration:**
https://git.darmstadt.ccc.de/ffda/site

blocktrron, 2019-09-04:
After changing the Line https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-wifi-config/luasrc/lib/gluon/config-mode/model/admin/wifi-config.lua#L139 from

`ht.default = uci.get('gluon', 'wireless', 'outdoor_' .. radio .. '_htmode') or 'default'`

to

`ht.default = uci:get('gluon', 'wireless', 'outdoor_' .. radio .. '_htmode') or 'default'`

the  WiFi configuration is displayed as expected again.

@NeoRaider was that the only issue or does lua-simple-uci have a bug here?

NeoRaider, 2019-09-05:
The bug is in wifi-config, simple-uci is working as expected.

closed 2019-09-06 by NeoRaider

blocktrron, 2019-09-06:
@NeoRaider thanks for the fix!

commit b2f5f71c (2019-09-06): gluon-web-wifi-config: fix incorrect UCI call

Fixes #1809

Fixes: bf5524915937 ("gluon-core: add outdoor support for 5 ghz radios")