freifunk-gluon/gluon #1823: Missing error message when domain names are assigned twice

Noodlesalat, 2019-09-21:
### Missing error message when domain names are choosen twice in different domain.conf files

**What is the problem?**
When trying to build gluon with the following domain config no direct error message is added to the logs.

dom4.conf:
```
[...]
domain_names = {
    **dom4** = 'Main-Kinzig-Kreis'
},
[...]
```
dom5.conf
```
[...]
domain_names = {
    **dom4** = 'Wartburgkreis',
},
[...]
```
build output:
```
make[4]: Entering directory '/mnt/Nudelsalat/site-fffd/gluon/package/gluon-site'
rm -f /mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/.built
touch /mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/.built_check
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='site.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/site.json'
rm -rf /mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains
mkdir -p /mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom5.json' ]
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='domains/dom5.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom5.json'
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom0.json' ]
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='domains/dom0.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom0.json'
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom3.json' ]
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='domains/dom3.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom3.json'
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom1.json' ]
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='domains/dom1.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom1.json'
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/legacy.json' ]
GLUON_SITEDIR='/mnt/Nudelsalat/site-fffd' GLUON_SITE_CONFIG='domains/legacy.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/legacy.json'
[ ! -e '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/domains/dom4.json' ]
Makefile:83: recipe for target '/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/.built' failed
make[4]: *** [/mnt/Nudelsalat/site-fffd/gluon/openwrt/build_dir/target-mips_24kc_musl/gluon-site/.built] Error 1
make[4]: Leaving directory '/mnt/Nudelsalat/site-fffd/gluon/package/gluon-site'
time: package/feeds/gluon_base/gluon-site/compile#0.02#0.01#0.16
package/Makefile:111: recipe for target 'package/feeds/gluon_base/gluon-site/compile' failed
make[3]: *** [package/feeds/gluon_base/gluon-site/compile] Error 2
make[3]: Leaving directory '/mnt/Nudelsalat/site-fffd/gluon/openwrt'
package/Makefile:107: recipe for target '/mnt/Nudelsalat/site-fffd/gluon/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[2]: *** [/mnt/Nudelsalat/site-fffd/gluon/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/mnt/Nudelsalat/site-fffd/gluon/openwrt'
/mnt/Nudelsalat/site-fffd/gluon/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make[1]: *** [world] Error 2
make[1]: Leaving directory '/mnt/Nudelsalat/site-fffd/gluon/openwrt'
Makefile:150: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```

**What is the expected behaviour?**
An explicit error message should be provided in order to support debugging.

**Gluon Version:**
v2018.2.3-16-g36b71dab
[https://git.maglab.space/fffd/gluon/commit/36b71dabb733d36ef9a2be3c1afa219efec71f30](https://git.maglab.space/fffd/gluon/commit/36b71dabb733d36ef9a2be3c1afa219efec71f30)

**Site Configuration:**
[https://git.maglab.space/fffd/site-fffd](https://git.maglab.space/fffd/site-fffd)

**Custom patches:**
- outdoor mode
- prevent building perl in parallel
- some fixes cherry-picked from gluon master
(tell me if a test with plain gluon is necessary)

mweinelt, 2019-09-21:
This was implemented in https://github.com/freifunk-gluon/gluon/commit/c208fc4fd9c94b765413795dfc1f995a0139e5c8.

```
$ git branch --contains c208fc4fd9c94b765413795dfc1f995a0139e5c8
master
```

Unfortunately only on master and therefore in v2019.1 and later. You can try and cherry-pick that commit on top of v2018.2.3, it could work.

mweinelt, 2019-11-08:
Tested on v2018.2.3-3-g34e60e2e. Cannot reproduce, therefore closing.

```
GLUON_SITEDIR='/gluon/site' GLUON_SITE_CONFIG='domains/dom1.conf' lua -e 'print(require("cjson").encode(assert(dofile("../../scripts/site_config.lua"))))' > '/gluon/openwrt/build_dir/target-x86_64_musl/gluon-site/domains/dom1.json'
ln: failed to create symbolic link '/gluon/openwrt/build_dir/target-x86_64_musl/gluon-site/domains/ffda_da_110.json': File exists
Failed to alias domain 'dom1' as 'ffda_da_110', name already taken by domain 'dom0'.
make[4]: *** [Makefile:124: /gluon/openwrt/build_dir/target-x86_64_musl/gluon-site/.built] Error 1
make[4]: Leaving directory '/gluon/package/gluon-site'
time: package/feeds/gluon_base/gluon-site/compile#0.17#0.18#0.34
make[3]: *** [package/Makefile:113: package/feeds/gluon_base/gluon-site/compile] Error 2
make[3]: Leaving directory '/gluon/openwrt'
make[2]: *** [package/Makefile:107: /gluon/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/gluon/openwrt'
make[1]: *** [/gluon/openwrt/include/toplevel.mk:218: world] Error 2
make[1]: Leaving directory '/gluon/openwrt'
make: *** [Makefile:152: all] Error 2
```

closed 2019-11-08 by mweinelt