freifunk-gluon/gluon #1824: re-enable batman-adv multicast optimizations

rotanid, 2019-09-24:
this issue is here to track that we want to re-enable batman-adv multicast optimizations after batman-adv has been updated to v2019.2 or later.

so this is blocked by #1791 

see also commits 9b1eb40fe7becd8f89c73c67772a94455502ac38 and 302a7951894b8a26080c2732c3855328caea18d7

mweinelt, 2019-12-04:
@T-X How do you want to handle this? Get it into v2019.2 (End of 2019) or wait until 2020.1 (Mid 2020)?

T-X, 2019-12-13:
@lemoer reported some issues with energy saving on Android phones somewhere. Which sounded like a potential blocker. @lemoer, could you (or the person from your community) reiterate here and post the according links? (or even better, create an extra issue here and reference it in this ticket?)

mweinelt, 2019-12-13:
https://github.com/freifunk-gluon/gluon/issues/1832

Not really our issue as I experience it with Android 10 in any WLAN network, not just those based on Gluon.

lemoer, 2019-12-14:
You are right. Neverthelesss I would suggest, that we disable multicast optimization till this issue is solved.

Imho broken user experience is not what we desire.
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

rotanid, 2019-12-18:
currently it IS disabled in standard Gluon, therefor we would only need to delay this issue.

But, even if something is fixed upstream, it will be years until every android phone got the fix.
so, do we never want to do it because of this??

lemoer, 2019-12-19:
Well, yes. You are right. Multicast optimizations in batman are still disabled.

But some preparational commits were already merged, which split the mutlicast domains between client devices (wifi, ...) and the bat device on the gluon nodes. Since some clients do not answer to MLD, the bridge does not forward the multicast packets to this clients and they become unreachable. This commits already raised the issue even before multicast optimizations in batman are enabled on any side.
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

mweinelt, 2021-05-09:
Reenabled in #2206.

closed 2021-05-09 by mweinelt