freifunk-gluon/gluon #1826: Private WiFI should support WPA3

mweinelt, 2019-09-25:
OpenWrt 19.07 has support for WPA3 and we should upgrade `web-private-wifi` to configure the private wifi in WPA2/WPA3 mixed mode.

- Upside is more secure connections for devices that support it.
- Downside is that we would need to bundle a cryptographic library.
  - This can therefore only be made available to devices with `>=8MB` of flash.

blocktrron, 2019-09-25:
Be my guest: https://github.com/blocktrron/gluon/tree/wip-wpa3

skorpy2009, 2019-09-30:
>Downside is that we would need to bundle a cryptographic library.
>* This can therefore only be made available to devices with >=8MB of flash.

A solution to this could be to just drop those devices with not enough flash.

rotanid, 2019-09-30:
> > Downside is that we would need to bundle a cryptographic library.
> > 
> > * This can therefore only be made available to devices with >=8MB of flash.
> 
> A solution to this could be to just drop those devices with not enough flash.

why? they are marked as deprecated and we can simply not add such features to them.

PR #1876 (2020-03-25): gluon-private-wifi: add support for WPA3 encryption
This commit adds WPA3 encryption for Gluon's Private WiFi feature.

It allows node operators to select between three encryption modes:
 - WPA2 (old standard - no MFP)
 - WPA2/WPA3 mixed-mode (optional MFP)
 - WPA3 (mandatory MFP)

MFP is supported by all drivers which are currently fully supported by Gluon (rt2800, mt76, ath9k, ath10k).

As we need to include a OpenSSL library for WPA3, this PR also includes a mechanic to define a featureset. This way, packages can be included at compile-time as well as the featureset retrieved at run-time to [... cut]
closed 2020-03-25 by blocktrron