freifunk-gluon/gluon #1832: Android does not respond to multicast request queries in power saving mode

lemoer, 2019-09-28:
We figured out that Android does not respond to multicast listener queries when it is in sleeping mode.

When the device wakes up and it's multicast entry in the router has expired, it takes up to 20 seconds till the router sends the next multicast listener query and the device responds. Only after this IPv6 neighbour discovery is possible again and the device has restored its IPv6 connectivity.

T-X, 2019-09-28:
This sounds like an Android bug. Either an IGMP/MLD query should wake up the device, so that it could receive it. Or it should send an unsolicited IGMP/MLD report upon waking up.

I'm not familiar with the Android ecosystem. @lemoer, could you file a bug for Android and reference it here?

lemoer, 2019-09-28:
https://issuetracker.google.com/issues/141785896

lemoer, 2019-12-14:
Currently I do not plan to reopen this issue by myself:

1. I do not really know how to answer their question of reproduction of the issue.
2. I do not have enough time to investigate here.

So best would be someone takes this issue in his responsibility or we revert multicast optimization for now.
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

CodeFetch, 2020-02-04:
It takes years for the buggy android devices to vanish...

T-X, 2020-02-16:
I updated the Si6 Networks ipv6toolkit and wrote a script around it to more easilly reproduce and evaluate the issues with MLD in an isolated setup:

* https://github.com/T-X/icmpv6-test
* https://github.com/T-X/ipv6toolkit/tree/fixes%2Bmldq6-tool

And I opened a new issues on the Android bugtracker here with additional information on how to reproduce the issue and added the results from the script:

* https://issuetracker.google.com/issues/149630944

T-X, 2020-02-16:
What I found quite interesting when playing with the ICMPv6 test script was:

* Everything looks fine when the phone is connected to a charger
* When on battery, Echo Requests-Replies still work fine
* When on battery, Neighbor Advertisements to Neighbor Solicitations seem rate-limited?
* When on battery, MLD Queries are filtered completely?
* When on battery, when sending MLD Queries via unicast (not quite standard compliant) there are at least some responses, similar to NDISC (now heavily rate-limited instead of filtered completely?)

Especially because ICMPv6 Echo Requests always work it currently seems to me that they are actively filtering based on the ICMPv6 type. And the people who implemented that might have overlooked the importance of MLD.

T-X, 2020-03-10:
I wrote a small tool: mldpoker. It will send direct, unicasted MLD Queries asking for MLD reports to hosts which just (re-)appeared, probably from sleep mode. Can people experiencing this issue check if it makes a difference for them?

MLD Poker source code:
* https://github.com/T-X/mldpoker

OpenWrt package (rebased on master branch of Gluon package repository):
* https://github.com/T-X/packages/tree/mldpoker/net/mldpoker

T-X, 2020-03-11:
Changelog v2:
* OpenWrt package: Adding UCI config which excludes bat0

lemoer, 2020-03-11:
Oh, nice. Thanks. That's a good idea to solve this also for legacy clients!

We will test it and come back to you.

T-X, 2020-06-15:
Just a brief update here: The mldpoker so far does not work. Sending MLD queries with a unicast link-layer destination is not sufficient.

What seems to work (but is not implemented in mldpoker yet) is sending an ICMPv6 Echo Request. They are always received and responded to by the slumbering Android device and wakes it up for a short amount of time. However the timing for the MLD Query is a bit tricky then. It needs to send shortly after receiving the Echo Reply from the Android device.

Sending an MLD Query before the Echo Reply does not work and sending it > 0.2 seconds after the Echo Reply does not work either. Also the Echo Reply is received about 0.2 to 0.7 seconds after the Echo Request, as the device needs a moment to wake up.

So a bit more complexity in mldpoker is needed (or I might just implement it in the bridge right away, I'll need to think about what might be easier to start with).

CodeFetch, 2020-06-19:
@T-X Actually I guess the correct way is to implement this in Batman together with the multicast optimizations directly... What do you mean with implementing it in the bridge? This sounds like something that would not be accepted upstream.

T-X, 2020-07-05:
@CodeFetch: It makes more sense in the bridge as it's not a batman-adv specific issue. It's a general issue with IGMP/MLD snooping bridges/switches together with Android devices. So for OpenWrt it'd be useful, too, and there I would submit it as well.

Whether upstream Linux will accept it, is another question. But I made it a kernel build option, disabled by default, which should lower the barrier for entrance there.

T-X, 2020-07-05:
@lemoer and others, would be great if you could test #2076 on a variety of devices. With an HTC U11 running Android 9 this seems to work and run reliably for me.

T-X, 2020-07-05:
PS: And the best way to confirm/test I'd recommend:

* On your Android phone, set Settings->Battery->Sleep to the minimum (15 seconds)
* Reduce /etc/config/network, config interface 'lan' -> option robustness from 9 to 1; that way any failure of the Android device to respond would result in connectivity issues and should be noticeable the fastest
* Keep the Android phone on battery, turn off the screen, wait a while (> 60 seconds)
* turn it on again, try to access sites via IPv6

To get the previous behaviour, to compare with, execute the following on the router: `$ echo 0 > /sys/class/net/client0/brport/multicast_wakeupcall`. Use `echo 2 > /sys/class/net/client0/brport/multicast_wakeupcall` to reenable. This can be done on the fly.

You can also install ip-bridge and run `$ bridge mdb | grep client0` and check for a multicast address matching your Android device. With this patch this entry should be present as long as the Android device is connected. Even if it goes to sleep. Before this PR it vanished from this table frequently.

closed 2020-10-08 by mweinelt

Adorfer, 2020-12-01:
FYI: https://issuetracker.google.com/issues/149630944 (thanks @awlx)