freifunk-gluon/gluon #1834: Ubiquiti UniFi AC LR missing fitting line in manifest in v2018.2.x and newer

rotanid, 2019-09-29:
Gluon v2018.2.x and newer on a Ubiquiti UniFi AC LR does not find a fitting line in autoupdater manifests.
It worked with Gluon v2018.1.x

Output-Information with Gluon v2018.1.x (LEDE 17.01.x)
```
# lua -e 'print(require("platform_info").get_image_name())'
ubiquiti-unifi-ac-lite
# cat /tmp/sysinfo/board_name
unifiac-lite
# cat /tmp/sysinfo/model
Ubiquiti UniFi-AC-LITE
```
Output-Information with Gluon v2018.2.x (OpenWrt 18.06.x)
```
# lua -e 'print(require("platform_info").get_image_name())'
ubiquiti-unifi-ac-lite-mesh
# cat /tmp/sysinfo/board_name
unifiac-lite
# cat /tmp/sysinfo/model
Ubiquiti UniFi-AC-LITE/MESH
```

ubnt_unifi_ac_board_detect() does not know the magic of the Ubiquiti UniFi AC LR, yet.

With help from @mweinelt i found out the magic is `e527` by calling:
`dd if=eeprom iflag=skip_bytes bs=2 skip=12 count=1 2> /dev/null | hexdump -v -e "1/1 \"%02x\""`

The PR #1332 and its merged commit 6d005b9803b394e09deb2063cae422c5054c04d8 is wrong.

A quickfix in Gluon would be to replace the erroneously added alias `ubiquiti-unifi-ac-lr` with the alias `ubiquiti-unifi-ac-lite-mesh`

Model detection should be contributed in upstream OpenWrt nevertheless.

rotanid, 2019-10-03:
i fixed the device alias in master branch, cherry-picked this to v2019.1.x branch and backported the fix to the v2018.2.x branch.

rotanid, 2019-10-04:
let's keep this issue open until upstream model detection is done

mweinelt, 2019-11-06:
@rotanid Gluon master includes 8cf67728d3b41c47ee5e322bb8c2908696380f24 now, so we need another update to reenable autoupdates afaict.

commit 94ca071e (2019-10-03): ar71xx-generic: fix device alias for Ubiquiti UniFi AC LR

fixes #1834
fixes #1332
commit 82244336 (2019-10-03): ar71xx-generic: fix device alias for Ubiquiti UniFi AC LR

fixes #1834
fixes #1332
commit 34e60e2e (2019-10-03): ar71xx-generic: fix device alias for Ubiquiti UniFi AC LR

fixes #1834
fixes #1332
PR #1857 (2019-11-08): ar71xx-generic: refactor image profiles / add alias for UniFi AC-LR
 ar71xx-generic: consolidate Ubiquiti UniFi AC images

OpenWrt generated images for the following devices are identical:

 - UniFi AC Lite / UniFi AC Mesh / UniFi AC LR
 - UniFi AC Pro / UniFi AC Mesh Pro

Gluon currently outputs devices as a separate image. However, as the
model detection is done at run-time, a symlink to one image would have
the same effect.

To save some space, create aliases for the devices instead of dedicated
profiles.

-------

 ar71xx-generic: add alias for UniFi AC LR

As upstream now has model detect [... cut]
closed 2019-11-08 by blocktrron