freifunk-gluon/gluon #1837: add Support for TP-Link Archer C60 v2

Marssl78, 2019-10-02:
Hi there!

I think, adding Archer C60 V2 would be easy, as it is supported in OpenWRT since 18.06.0. Further, it seems "functionally promising", because (other than V1) its migration from ar71xx to ath79 is completed, so might (also other than V1) run in 5-GHz-Band without OOM-issues...

What I already did for testing:
I built a firmware with added profile for V2 and ran the testings as required. Primary MAC didn't match, so I built again with changed package for selecting MAC of interface 'eth0' as primary MAC, now primary MAC is matching the one labelled at the bottom of the case.

LEDs are working correctly, firmware is flashable through OEM WebInterface (will test TFTP-Method in addition this evening). Meshing over LAN / WLAN 2.4 / WLAN 5 and MeshVPN (fastd) is working, node shows up on map and clients get access to inet as expected.

I will let follow filled-out form of testings. Meanwhilest the device can be seen on our map:

[https://bgl.map.ffgl.eu/#!v:m;n:d80d176dd712](url)

rotanid, 2019-10-03:
thank you for your work, we are looking forward to a pull request with the necessary changes and the filled-out checklist.

feel free to ask if you need any help/information!

Marssl78, 2019-10-04:
Just opened PR #1840 :-)

Testing checklist will go on tonight... ;-)

rotanid, 2019-10-06:
merged: b823b80356f63f4b52bec572cdafe30cf328e5f1

closed 2019-10-06 by rotanid