freifunk-gluon/gluon #1839: add Support for TP-Link RE450v3

larsan, 2019-10-03:
We recently got word that the image for the RE450 could not be flashed to a newly acquired RE450 and it turns out that a third hardware revision was released, While the current builds only support V1 and V2, the List of supported devices does not differenciate between them.

achterin, 2019-12-03:
> While the current builds only support V1 and V2, the list of supported devices does not differentiate between them.

it doesnt differentiate because this is not true. gluon supports v1, only openwrt supports both. in my opinion v2 will never get fully supported by gluon (only with broken flag like re355) as its only available with 64mb ram, which is not enough to run both radios stable.

NeoRaider, 2019-12-07:
Docs fixed in ed2479bd40613010a44a519c357a785a1c46b792 and f44a6342b46e1a9afc2760627f3070fe3e30fc9f.

blocktrron, 2020-04-10:
Closing this, as hardware support is not a Gluon issue. If somebody ported the device or it was added to upstream OpenWrt it can be added to Gluon.

closed 2020-04-10 by blocktrron